espressif/esp-idf #14897: Writing flash data with PSRAM XIP enabled causes MMAP flash read errors

swilson, 2024-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

mmap flash reads should return correct data

### What is the actual behavior?

mmap read returns garbage randomly. Turning off PSRAM XIP (CONFIG_SPIRAM_FETCH_INSTRUCTIONS & CONFIG_SPIRAM_RODATA) seems to avoid the issue. Perhaps because when PSRAM XIP is enabled (aka SPI_FLASH_CACHE_NO_DISABLE) the regular flash APIs use mutex locking, but the MMAP APIs disable the cache? 

### Steps to reproduce.

[main.c.txt](https://github.com/user-attachments/files/17781948/main.c.txt)

### Debug Logs.

```plain
I (1556) example: esp_partition_read deadbeefdeadbeefdeadbeefdeadbeef
I (1557) example: mmap read: ESP-IDF Partition Memory Map Example
I (1659) example: mmap read: ESP-IDF Partition Memory Map Example
I (1759) example: mmap read: ESP-IDF Partition Memory Map Example
I (1859) example: mmap read: ESP-IDF Partition Memory Map Example
I (1959) example: mmap read: ESP-IDF Partition Memory Map Example
I (2059) example: mmap read: ESP-IDF Partition Memory Map Example
I (2138) example: esp_partition_read deadbeefdeadbeefdeadbeefdeadbeef
I (2159) example: mmap read: �������������������������������������
```

### More Information.

[partitions_example.csv](https://github.com/user-attachments/files/17781919/partitions_example.csv)
[sdkconfig.txt](https://github.com/user-attachments/files/17781945/sdkconfig.txt)

swilson, 2024-11-15:
This bug was initially discovered when a call to esp_ota_set_boot_partition() (which uses MMAP flash reads) failed while another thread was writing to to a different partition.

ginkgm, 2024-12-01:
Hi @swilson ,

Thanks for your report. This issue is confirmed. 

The cause of the issue is from the flash models: when an erasing/writing is in progress, no read is allowed.

In other modes, the cache is disabled and therefore no read (including from mmap region) will happen simultaneously. However in XIP PSRAM mode, the cache should be kept enabled so that execution from PSRAM can keep going. The problem here is about the regions mmaped to flash. They cannot return correct data but the behavior is not forbidden.

Workaround that you can do for now is to avoid the reading from mmap region and flash erasing/writing concurrently. 
- on the task that reads: Take a mutex before the mmap and release it after unmap
- on the task that writes: Wrap the erasing/writing code with the mutex.

We will provide a mutex in the mmap API - when any one of mmap against flash starts, all flash write/erase will be blocked, until the last flash region is unmaped. This mutex will be used in Espressif APIs automatically. For users,

- If they want to avoid flash erase/write during the mmap (Usually when it's a temporary call to read something from flash before unmap), they can make use of the mutex via a flag in mmap API to avoid the problem.
- If they want to have a opening mmap that may have flash operation in the middle, they will have to avoid reading from mmap region during flash operations by themselves. Maybe we will provide panic mechanism or some APIs to take the mutex when actually reading from the region. But this is something for the future.