espressif/esp-idf #14898: ESP32使用blufi配网，当调用esp_blufi_controller_deinit后再次调用esp_blufi_controller_init会发生coredump

lipan465, 2024-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp-idf-v5.2.2

### Power Supply used.

USB

### What is the expected behavior?

ESP32使用blufi配网，当调用esp_blufi_controller_deinit后再次调用esp_blufi_controller_init会发生coredump

### What is the actual behavior?

ESP32使用blufi配网，当调用esp_blufi_controller_deinit后再次调用esp_blufi_controller_init会发生coredump

### Steps to reproduce.

ESP32使用blufi配网，当调用esp_blufi_controller_deinit后再次调用esp_blufi_controller_init会发生coredump;
出问题的代码如下

> 

> `static esp_err_t esp_bt_controller_rom_mem_release(esp_bt_mode_t mode)
{
    bool update = true;
    intptr_t mem_start=(intptr_t) NULL, mem_end=(intptr_t) NULL;

    if (btdm_controller_status != ESP_BT_CONTROLLER_STATUS_IDLE) {
        return ESP_ERR_INVALID_STATE;
    }

    //already released
    if (!(mode & btdm_dram_available_region[0].mode)) {
        return ESP_ERR_INVALID_STATE;
    }`

当已经调用过released就会返回失败，然后
esp_err_t esp_blufi_controller_init() {
    esp_err_t ret = ESP_OK;
#if CONFIG_IDF_TARGET_ESP32
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
#endif
这里就会coredump；请优化；

反观esp32C3的实现就没有问题
`esp_err_t esp_bt_controller_rom_mem_release(esp_bt_mode_t mode)
{
    esp_err_t ret = ESP_OK;

    if (btdm_controller_status != ESP_BT_CONTROLLER_STATUS_IDLE) {
        return ESP_ERR_INVALID_STATE;
    }

    bt_area_t rom_btdm_data = {
        .start = (intptr_t) ets_rom_layout_p->data_start_btdm,
        .end   = (intptr_t) ets_rom_layout_p->data_end_btdm,
        .name  = "ROM btdm data",
    };
    bt_area_t rom_btdm_bss = {
        .start = (intptr_t)ets_rom_layout_p->bss_start_btdm,
        .end   = (intptr_t)ets_rom_layout_p->bss_end_btdm,
        .name  = "ROM btdm BSS",
    };
    bt_area_t rom_btdm_inter_data = {
        .start = (intptr_t) ets_rom_layout_p->data_start_interface_btdm,
        .end   = (intptr_t) ets_rom_layout_p->data_end_interface_btdm,
        .name  = "ROM interface btdm data",
    };
    bt_area_t rom_btdm_inter_bss = {
        .start = (intptr_t)ets_rom_layout_p->bss_start_interface_btdm,
        .end   = (intptr_t)ets_rom_layout_p->bss_end_interface_btdm,
        .name  = "ROM interface btdm BSS",
    };

    if (btdm_controller_status != ESP_BT_CONTROLLER_STATUS_IDLE) {
        ret = ESP_ERR_INVALID_STATE;
    }

    if (mode & ESP_BT_MODE_BLE) {
        /* Free BTDM memory used by the ROM */
        if (ret == ESP_OK) {
            ret = esp_bt_mem_release_areas(&rom_btdm_data, &rom_btdm_bss);
        }

        if (ret == ESP_OK) {
            ret = esp_bt_mem_release_areas(&rom_btdm_inter_data, &rom_btdm_inter_bss);
        }
    }

    return ret;
}` 他这个如果if (mode & ESP_BT_MODE_BLE) 为false的时候直接返回OK

### Debug Logs.

```plain
1
```

### More Information.

1

esp-zhp, 2024-11-19:
1-问题出现原因
实际上 esp_bt_controller_mem_release 调用一次就好了，后面再调用该 API 的时候会返回错误，
返回错误的原因是该 API 已经被调用过了，相关的内存已经被释放。

2-解决办法
保证下面 API 只会被调用一次
```
#if CONFIG_IDF_TARGET_ESP32
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
#endif
```

你可以将该 API 从 esp_blufi_controller_init 函数移出来。

esp-zhp, 2024-11-22:
fixed in mater 7addb57ccd53576b40d3cfeb24c185036053f4b4
[image]

Alvin1Zhang, 2024-12-23:
Thanks for sharing, fix on release/5.2 is available at https://github.com/espressif/esp-idf/commit/aabe255f8955a28662493a326f6dadb88442837d, thanks.

closed 2024-12-23 by Alvin1Zhang

esp-zhp, 2025-02-12:
•	master: 3fc6c93936077cb1659e1f0e0268e62cf6423e9d
•	v5.4: 5f93ec3b11b6115475c34de57093b3672d594e8f
•	v5.3: f40aa9c587a8e570dfde2e6330382dcd170d5a5d (pending merge into release branch)
•	v5.2: bf50c0c197af30990026c8f8286298d2aa5a3c99 (pending merge into release branch)
•	v5.1: b1657d9dd4d0e48ed25e02cb8fe8413f479a2a84
•	v5.0: cc00e9f2fc4f7e8fbaff27851b4a8b45fa483501