espressif/esp-idf #14901: Install Error on MacBook

latheefth, 2024-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-4291-g7a305c0284

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I am new to ESP and trying to install the IDF to start working on it.
I cloned git clone --recursive https://github.com/espressif/esp-idf.git
and then ./install.sh esp32 but get the below error. 
Error --> ERROR: Failed to download, and retry count has expired
[image]

### What is the actual behavior?

To be able to use IDF

### Steps to reproduce.

I cloned git clone --recursive https://github.com/espressif/esp-idf.git
and then ./install.sh esp32 but get the below error. 
Error --> ERROR: Failed to download, and retry count has expired

### Build or installation Logs.

```plain
latheefth@Abduls-MBP esp-idf % ./install.sh esp32
Detecting the Python interpreter
Checking "python3" ...
Python 3.13.0
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Selected targets are: esp32c2, esp32s3, esp32c6, esp32p4, esp32c3, esp32, esp32s2, esp32c5, esp32h2, esp32c61
Current system platform: macos
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Installing xtensa-esp-elf@esp-14.2.0_20240906
Using GitHub assets mirror for URL: https://github.com/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz => https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz
Downloading https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz
Destination: /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp
98%WARNING: file size mismatch for /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp, expected 182793460, got 180173727
WARNING: Download failure: <urlopen error retrieval incomplete: got only 180173727 out of 182793460 bytes>
WARNING: Failed to download https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz to /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp
Downloading https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz
Destination: /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp
98%WARNING: file size mismatch for /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp, expected 182793460, got 180224867
WARNING: Download failure: <urlopen error retrieval incomplete: got only 180224867 out of 182793460 bytes>
WARNING: Failed to download https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz to /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp
Downloading https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz
Destination: /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp
97%WARNING: file size mismatch for /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp, expected 182793460, got 178382928
WARNING: Download failure: <urlopen error retrieval incomplete: got only 178382928 out of 182793460 bytes>
WARNING: Failed to download https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz to /Users/latheefth/.espressif/dist/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz.tmp
ERROR: Failed to download, and retry count has expired
Please make sure you have a working Internet connection.
```

latheefth, 2024-11-16:
I have Sonoma os installed on Mac pro15 using open core patcher.

peterdragun, 2024-11-18:
Hi @latheefth, can you please try to verify that if you download the file directly it will download the whole tar as expected? You can use `wget` with the following command:
```sh
wget https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz
```
and check the file size with e.g. `ls -l xtensa-esp-elf-14.2.0_20240906-x86_64-apple-darwin.tar.xz`
I also recommend checking whether you have enough space left on the disk for this file.
I am also on Mac OS (but 15.1; arm), and I was able to download this file (and also the arm version) without any issues. The expected size matches the one that was downloaded.

latheefth, 2024-11-18:
Please find the log as below, I don't find any issue with the wget 
[image]

peterdragun, 2024-11-19:
Can you please verify that you can still see the issue if you run the install script? Maybe it was some issue with the server which has been resolved already. If that does not help, please try not using the dl espressif mirror but using the GitHub assets directly. (unset `IDF_GITHUB_ASSETS` env var)

dobairoland, 2024-11-20:
@latheefth We are sorry for the inconvenience. The only workaround I can offer now is to download the missing files with `wget` and put them into the `~/.espressif/dist` directory. After that `install.sh` will be able to install them.

latheefth, 2024-11-20:
OK..but I am not able to find list directory below. Shall I  create dist directory and then do wget for each missing files? Please confirm.
[image]

peterdragun, 2024-11-20:
@latheefth `dist` directory is not part of the ESP-IDF source git directory but is located with other installed tools in your home directory in `~/.espressif/dist` as mentioned by Roland.

jjhax, 2026-10-16:
Exactly the same issue on M1. `eim` wont install either because of issues with Python. We're just left running around in circles.

I don't mean to sound rude, but I can't understand how there's so many unfixed issues that nobody is solving at espressif. It's shocking. Are you using AI to write your codebase?!

dobairoland, 2026-10-16:
Apologies that this takes so long. First, we had issues with the reproduction. Then we were counting on EIM replacing the ESP-IDF-side install scripts. That got delayed several times. I try to adjust the linked PR in a way so it could be accepted and backport it.

closed 2026-10-24 by igrr