espressif/esp-idf #14902: Please provide a simple a working and simple example of the new I2C API

juantxorena, 2024-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Since version v5.2 there's a new API for I2C devices, and the old one is deprecated. There's only an example in the repo, but it's IMHO overly complicated and not very applicable to a lot of people.

See the old example:
https://github.com/espressif/esp-idf/tree/release/v5.1/examples/peripherals/i2c/i2c_simple

It is a simple device where it's configured, and it's shown how to read from a register and write to a register.  Very few lines of code, and enough to get a gist of the API.

And now compare this to the new example, for reading and writing from an EEPROM:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_eeprom

it's a mess with a ton of type definitions, apparently for no reason:
https://github.com/espressif/esp-idf/blob/7a305c0284b7af7cd8b8f12b48f72e2685d9a363/examples/peripherals/i2c/i2c_eeprom/components/i2c_eeprom/i2c_eeprom.h#L27
memory allocations where in the old one none was needed:
https://github.com/espressif/esp-idf/blob/7a305c0284b7af7cd8b8f12b48f72e2685d9a363/examples/peripherals/i2c/i2c_eeprom/components/i2c_eeprom/i2c_eeprom.c#L28-L40
managing a buffer manually:
https://github.com/espressif/esp-idf/blob/7a305c0284b7af7cd8b8f12b48f72e2685d9a363/examples/peripherals/i2c/i2c_eeprom/main/i2c_eeprom_main.c#L56-L60
etc. It just looks overly verbose and complicated.

So please provide a simple example for the new I2C API where it's shown how to:
- Configure the master and the device
- Read from a register
- Write to a register

suda-morris, 2024-11-18:
This example function should illustrate the "i2c write to a register" well: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/i2c_tools/main/cmd_i2ctools.c#L206

This example function should illustrate the "i2c read from a register" well: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/i2c_tools/main/cmd_i2ctools.c#L152

suda-morris, 2024-11-19:
linked in https://github.com/espressif/esp-idf/pull/14315

vinicius-de-assis, 2024-11-26:
> linked in #14315

@suda-morris Hi there! I noticed that my pull request hasn't received any feedback since August. As this is my first time submitting a pull request to the esp-idf repo, I wanted to check if there's anything I need to address or improve. Any guidance would be greatly appreciated. Thank you!

closed 2024-12-05 by espressif-bot