espressif/esp-idf #14903: Ethernet crossover connection between 2 ESP32s3 w5500

bogdan-voitovych, 2024-11-17:
Could you say if it possible. I tried to set on both static IP and run PING on both sides. Currently, ping not working on both sides. 
Thanks.

kostaond, 2024-11-18:
This is not an issue. Please use forum for asking questions next time. Anyway, what do you mean by "Ethernet crossover connection"? Do you mean bridging at L2? If so, here is example: https://github.com/espressif/esp-idf/tree/master/examples/network/bridge.

bogdan-voitovych, 2024-11-18:
The idea is to connect two ESP32s3 directly without a router through Ethernet using two W5500 SPI modules.

Current status:

- 2 ESP with W5500 modules are connected
- Used CrossOver cable
- RJ45 Green and Yellow LEDs are ON on both sides. 

On one side I use a basic ethernet example with some changes in code and receive that log:
I (8356) esp_netif_handlers: eth0 ip: 169.254.189.155, mask: 255.255.0.0, gw: 0.0.0.0
I (8356) eth_example: Ethernet Got IP Address
I (8356) eth_example: ~~~~~~~~~~~
I (8356) eth_example: ETHIP:169.254.189.155
I (8366) eth_example: ETHMASK:255.255.0.0
I (8366) eth_example: ETHGW:0.0.0.0
I (8376) eth_example: ~~~~~~~~~~~

On the other hand, I use the same example with integrated ICMPecho to have a PING possibility. LOG:
I (2388) eth_example: Ethernet Link Up
I (2388) eth_example: Ethernet HW Addr 36:85:18:49:26:20
I (2388) esp_netif_handlers: eth ip: 169.254.189.156, mask: 255.255.0.0, gw: 0.0.0.0
I (2388) eth_example: Ethernet Got IP Address
I (2398) eth_example: ~~~~~~~~~~~
I (2398) eth_example: ETHIP:169.254.189.156
I (2408) eth_example: ETHMASK:255.255.0.0
I (2408) eth_example: ETHGW:0.0.0.0
I (2418) eth_example: ~~~~~~~~~~~
I (4498) main_task: Returned from app_main()
esp> ping 169.254.189.155
esp> From 169.254.189.155 icmp_seq=1 timeout
From 169.254.189.155 icmp_seq=2 timeout

Could somebody help me with that? Is it possible to communicate in such a way?