espressif/esp-idf #14904: Using software-driven Coex Module leads to lost BLE Messages

timoxd7, 2024-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2, v5.1.5, v5.3.1

### Espressif SoC revision.

ESP32

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

Sending Messages using BLE while still listening to ESP NOW Messages will work without loosing a single BLE connection (as it behaved in IDF v4.3.4), no matter if we use the hardware-defined coexistence or the software-defined (software-defined even using [this](https://github.com/espressif/esp-idf/issues/12434#issuecomment-1779133465) while sending messages).

### What is the actual behavior?

After upgrading from v4.3.4 to v5.1.2 we saw sporadic lost Messages sent from one ESP32 to another using NimBLE. After upgrading then to v5.3.1, the issue was even more noticeable, even affecting normal operation of our devices.

After switching from software-defined to hardware-defined coex (`CONFIG_ESP_COEX_SW_COEXIST_ENABLE` is not set), the issue stopped to appear and all BLE Messages were able to be delivered again.

Using Wireshark, we were able to find out that the ESP32 is just not responding or sending messages for a given time period, which will lead to multiple unhandled BLE cycles, loosing multiple messages.

### Steps to reproduce.

Hard to describe, as the issue appears in our own Software. If the issue is not known already, we might create an example project reproducing the issue in an isolated project.

### More Information.

We mainly want to know if the Issue is known already and why the software-defined coexistence got broken from v5.x upwards somehow and got even worse in v5.3.1. We are using ESP NOW as second communication channel, but BLE is used exclusively during normal operation (with ESP NOW only listening for incoming messages, but there are none if nothing gets lost over BLE).

Also, we were not able to exactly find out what are the drawbacks using the hardware-defined coexistence, as it now works kind of good during the first attempts using it. Is there any documentation except the coexistence documentation, stating out more clear what the difference actually is?

esp-lrh, 2024-11-20:
Hi, @timoxd7 

If we disable software coexistence on ESP32, the behavior of hardware will become very simple, we can call it 'Bluetooth priority mode', it means Bluetooth can always use the RF resource if it want. Disable software coexistence does help improve BLE performance, but this may cause unstable wifi performance.

Based on the information you provided, I can't confirm whether this is a known issue. Can you provide more information about BLE messages lost, including:
- Which API are you using to send BLE messages
- The return value of the API and the output log when losing BLE messages
- IDF version or commit id that the issue reproduce
Also, It would be even better if you can build an example project to reproduce the issue.

At last, thanks for your feedback, we will continue to follow up on this issue.

Alvin1Zhang, 2025-03-07:
@timoxd7 Thanks for reporting, would you please help provide more details? Thanks.

closed 2025-05-08 by Alvin1Zhang