espressif/esp-idf #14906: I2C Deadlock

RDobrinov, 2024-11-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1 & 5.4-dev ( master branch )

### Espressif SoC revision.

ESP32-D0WD (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-D0WD (revision v1.0)

### Power Supply used.

USB

### What is the expected behavior?

Do not enter into deadlock.

### What is the actual behavior?

Deadlock occur when no free i2c device left

### Steps to reproduce.

Just request execute i2c_new_master_bus() until no free busses left, and try to release one of them.

### More Information.

When no free buses left [i2c_acquire_bus_handle()](https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/components/esp_driver_i2c/i2c_common.c#L151) do not release s_i2c_platform.mutex.
[ESP_RETURN_ON_FALSE](https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/components/esp_driver_i2c/i2c_common.c#L172) is executed and lock never released

atanisoft, 2024-11-19:
This might be (partially) covered by https://github.com/espressif/esp-idf/pull/14845.

pback34, 2024-11-20:
> This might be (partially) covered by #14845.

it looks like @mythbuster5 committed the same fix as I have for this issue on Nov 6, only hours before me (commit [0a098f4](https://github.com/espressif/esp-idf/commit/0a098f49d4236ecff8b21cf7066fcf572847c454)! Is that correct @mythbuster5 ?

Could there still be deeper issues that remain even if this fixed the apparent issue?

closed 2024-12-03 by espressif-bot