espressif/esp-idf #14910: BLE after `esp_ble_gattc_write_char_descr` does not handle response

matislovas, 2024-11-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-D0WD-V3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Olimex ESP32-POE-ISO-WROVER

### Power Supply used.

USB

### What is the expected behavior?

I am developing BLE client app (bluedroid), base on `gattc_demo.c` example. And ~99/100 times works fine. But recently, testing of the BLE server devices, I came across to situation, where device is unable to connect to esp32 BLE client.

The failing part is the following: after calling `esp_ble_gattc_write_char_descr()` to enable notify or indicate on BLE server device (which is the last step to connect to BLE server device), I am expecting the `ESP_GATTC_WRITE_DESCR_EVT` to be generated, when response from server device is received. But after some testing, I am able to introduce the esp32 client into some state, where it does not generate `ESP_GATTC_WRITE_DESCR_EVT` events.

 I ran a test with BLE sniffer, to collect the packets between client and server devices, and the server device sends, and I see that server device actually sends back the `write_response` in both - successful and unsuccessful scenarios. But in unsuccessful scenario, the esp does not generate `ESP_GATTC_WRITE_DESCR_EVT` and thus, i cannot finish the connection to server device.
 
 This behavior is not persistent, and I need to connect and disconnect the device multiple times, before I move esp into this state. After first unsuccessful, I cannot connect to that server device (failure is at the same place). Only rebooting esp does help. 

### What is the actual behavior?

The `ESP_GATTC_WRITE_DESCR_EVT` is not being generated.

### Steps to reproduce.

1. Connect and disconnect from BLE server device.
2. Observe, if you receive `write_response` packet and if it generates `ESP_GATTC_WRITE_DESCR_EVT` on esp32 side. 

### More Information.

I haven't dug deeper into the bluedroid or controller stacks, but if you will guide on how to provide you with more logs on that front, I would be happy to help.

esp-zhp, 2024-11-21:
Thank you for reaching out and sharing the details of your issue. I have a few questions and suggestions to better understand and troubleshoot the problem:

1. You mentioned:  
   *"But recently, testing of the BLE server devices, I came across to situation, where device is unable to connect to esp32 BLE client."*  
   In the `gatt_client` example, the GATT client initiates a connection to the GATT server, and the client acts as the master role during this process. However, based on your description, it seems the server device is connecting to the GATT client. Could you please confirm which side is the master device in your setup?

2. Additionally, you stated:  
   *"I ran a test with BLE sniffer, to collect the packets between client and server devices, and the server device sends, and I see that server device actually sends back the write_response in both - successful and unsuccessful scenarios. But in unsuccessful scenario, the esp does not generate ESP_GATTC_WRITE_DESCR_EVT and thus, I cannot finish the connection to server device."*  
   Could you clarify why the absence of `ESP_GATTC_WRITE_DESCR_EVT` prevents completing the connection to the server device? The ability to call `esp_ble_gattc_write_char_descr` indicates that the connection is already established. How does this relate to the inability to establish a connection?

3. The absence of `ESP_GATTC_WRITE_DESCR_EVT` after calling `esp_ble_gattc_write_char_descr` does appear to be an issue. To assist further:  
   - Could you provide detailed logs for this scenario?  
   - If possible, share a minimal code snippet that reproduces the issue.  

Additionally, you can enable HCI logs for more detailed debugging. In the `gatt_client` example, add the following code to the `app_main` function:  

   ```c
   /*
    * This code is intended for debugging and prints all HCI data.
    * To enable it, turn on the "BT_HCI_LOG_DEBUG_EN" configuration option.
    * The output HCI data can be parsed using the script:
    * esp-idf/tools/bt/bt_hci_to_btsnoop.py.
    * For detailed instructions, refer to esp-idf/tools/bt/readme.md.
    */

   while (1) {
       extern void bt_hci_log_hci_data_show(void);
       extern void bt_hci_log_hci_adv_show(void);
       bt_hci_log_hci_data_show();
       bt_hci_log_hci_adv_show();
       vTaskDelay(1000 / portTICK_PERIOD_MS);
   }
   ```

This will capture HCI data, which you can parse using the provided script in the ESP-IDF tools. Sharing this data would help us analyze the issue further.  

Looking forward to your response!

matislovas, 2024-11-24:
> In the gatt_client example, the GATT client initiates a connection to the GATT server, and the client acts as the master role during this process. However, based on your description, it seems the [...]

No, my setup works in standard way, as you described. GATT client (esp32) connects to GATT server (in my case - blood pressure monitor). So the esp32 acts as a master and pressure monitor - as slave.

> In the gatt_client example, the GATT client initiates a connection to the GATT server, and the client acts as the master role during this process. However, based on your description, it seems the [...]

Yes, you're right, connection itself is opened, but device "completes" connection only when receiving `ESP_GATTC_WRITE_DESCR_EVT` event without any errors in it (at least, all other 20 GATT server devices, that my application supports works this way). It just seems strange, that in pcap files I see that packet from server device to client, but esp32 does not generate that event. I tried omitting that last step (waiting for the `ESP_GATTC_WRITE_DESCR_EVT` to be generated) and notify/indicate functions on that characteristic actually works - I am able to receive packets from that characteristic, that I haven't received the expected event. I am adding two pcap files from BLE sniffer - [pcaps.zip](https://github.com/user-attachments/files/17892512/pcaps.zip). In `doesnotwork.pcapng` file, the "write descriptor response" packet was received (packet no. 201), but no further communication took place. In `works.pcapng` file, the packet no. is 153, but you can see that after that, the communication with device continues (reading/writing chars).

I will add logs and code snippets shortly.

matislovas, 2024-11-24:
> Additionally, you can enable HCI logs for more detailed debugging. In the gatt_client example, add the following code to the app_main function:

I am not able to include this snippet. IIRC the `bt_hci_log.h` is private header, so linker throws me errors:

```
/home/matas/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/matas/Devel/softdent/esp32/main/src/main.c:33: undefined reference to `bt_hci_log_hci_adv_show'
```

I've added the `CONFIG_BT_HCI_LOG_DEBUG_EN` option to `sdkconfig.defaults`, but it seems that my version of esp-idf (v5.3.1) does not have this config option -  I cannot see it in generated `sdkconfig` file.

matislovas, 2024-11-24:
I can provide logs from my application, with logs enabled in bluedroid.

Working scenario:
```
I (32547) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
I (32577) BT_GATT: GATTC_ConfigureMTU conn_id=3 mtu=247
I (32577) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
I (18:17:49.145) Bluetooth: Service Match found for 46:22:DA:2B:4F:99
I (18:17:49.149) Bluetooth: Get service information from remote device
I (18:17:49.157) Bluetooth: Got tx handler 16
I (18:17:49.161) Bluetooth: Try to register for notification with rx handler 18
I (18:17:49.170) Bluetooth: Notification/Indication write (app_id: 0, conn_id: 0)
I (32617) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
I (18:17:49.214) Bluetooth: Tx descr success! (app_id: 0, conn_id: 0) 
I (18:17:49.215) libemetryhub: Device ("46:22:DA:2B:4F:99", "Viatom BPM BP2") conn status change from "connect_in_progress" to "connected"
I (18:17:49.225) libemetryhub: Device ("46:22:DA:2B:4F:99", "Viatom BPM BP2") should start data process task!
I (18:17:49.235) libemetryhub: Viatom BPM BP2 46:22:DA:2B:4F:99 Setting device time and date
I (32687) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
I (18:17:49.249) libemetryhub: Viatom BPM BP2 46:22:DA:2B:4F:99 Time and date sent to device
I (18:17:49.254) Bluetooth: Write characteristic ok!
I (18:17:49.266) libemetryhub: Started data task for device ("46:22:DA:2B:4F:99", "Viatom BPM BP2")
I (32707) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (32717) BT_GATT: GATT_GetConnectionInfor conn_id=4
I (32727) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (32727) BT_GATT: GATT_GetConnectionInfor conn_id=6
I (32737) BT_GATT: GATT_GetConnectionInfor conn_id=7
I (32747) BT_GATT: GATT_GetConnectionInfor conn_id=8
I (32747) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
```
(The `I (18:17:49.214) Bluetooth: Tx descr success! (app_id: 0, conn_id: 0) ` is printed inside `ESP_GATTC_WRITE_DESCR_EVT` section).

Non-working scenario:
```
I (46876) BT_GATT: GATTC_Discover conn_id=3 disc_type=6
I (46876) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
I (46896) BT_BTM: BTM_CheckAdvData type=0x01
I (46896) BT_BTM: BTM_CheckAdvData type=0x19
I (46896) BT_BTM: BTM_CheckAdvData type=0x03
I (46896) BT_BTM: BTM_CheckAdvData type=0x01
I (46906) BT_BTM: BTM_InqDbRead: bd addr [553b723cc5d9]

I (46916) BT_GATT: GATTC_ConfigureMTU conn_id=3 mtu=247
I (46916) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
I (46926) BT_BTM: BTM_CheckAdvData type=0x01
I (46936) BT_BTM: BTM_CheckAdvData type=0x19
I (46936) BT_BTM: BTM_CheckAdvData type=0x03
I (46936) BT_BTM: BTM_CheckAdvData type=0x01
I (46946) BT_BTM: BTM_InqDbRead: bd addr [e4f823451a79]

I (46946) BT_BTM: BTM_CheckAdvData type=0x01
I (46956) BT_BTM: BTM_CheckAdvData type=0x19
I (46956) BT_BTM: BTM_CheckAdvData type=0x03
I (46966) BT_BTM: BTM_CheckAdvData type=0x01
I (46966) BT_BTM: BTM_InqDbRead: bd addr [477196d52536]

I (46976) BT_BTM: BTM_CheckAdvData type=0x01
I (46976) BT_BTM: BTM_CheckAdvData type=0x19
I (46986) BT_BTM: BTM_CheckAdvData type=0x03
I (46986) BT_BTM: BTM_CheckAdvData type=0x01
I (46996) BT_BTM: BTM_InqDbRead: bd addr [7904ea517297]

I (18:15:00.232) Bluetooth: Service Match found for 46:22:DA:2B:4F:99
I (18:15:00.314) Bluetooth: Get service information from remote device
I (18:15:00.321) Bluetooth: Got tx handler 16
I (18:15:00.326) Bluetooth: Try to register for notification with rx handler 18
I (18:15:00.336) Bluetooth: Notification/Indication write (app_id: 0, conn_id: 0)
I (47036) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4622da2b4f99
I (47666) BT_BTM: BTM_CheckAdvData type=0x01
I (47666) BT_BTM: BTM_CheckAdvData type=0x19
I (47666) BT_BTM: BTM_CheckAdvData type=0x03
I (47666) BT_BTM: BTM_CheckAdvData type=0x01
I (47676) BT_BTM: BTM_InqDbRead: bd addr [3f81ae3c74d7]

I (48066) BT_BTM: BTM_CheckAdvData type=0x01
I (48066) BT_BTM: BTM_CheckAdvData type=0x19
I (48066) BT_BTM: BTM_CheckAdvData type=0x03
I (48066) BT_BTM: BTM_CheckAdvData type=0x01
I (48076) BT_BTM: BTM_InqDbRead: bd addr [fc039fea9385]

I (48176) BT_BTM: BTM_CheckAdvData type=0x01
I (48176) BT_BTM: BTM_CheckAdvData type=0x19
I (48176) BT_BTM: BTM_CheckAdvData type=0x03
I (48186) BT_BTM: BTM_CheckAdvData type=0x01
I (48186) BT_BTM: BTM_InqDbRead: bd addr [4db90bf65ac2]

I (48196) BT_BTM: BTM_InqDbRead: bd addr [4db90bf65ac2]
```

(notice - no `Tx descr success!` print).

esp-zhp, 2024-11-25:
1. The HCI log cannot be used, likely because your version is too outdated. This feature was introduced about six months ago. Please try upgrading to the latest version and test again.

2. It seems this issue does not occur every time, correct? One possible cause could be that the BTC queue is full. You should check the following:  
   - Are there any time-consuming operations being performed within the callback functions of GAP, GATTC, or GATTS? This is not allowed and could lead to issues.

Additionally, from the logs you provided:  
   - In one log, there are no ADV reports after the connection is established.  
   - In the other log, many ADV reports are still being received.  

Could you test again by disabling SCAN after the connection is established to see if the issue persists?

Alvin1Zhang, 2024-12-18:
Thanks for reporting, will close due to inactivity, feel free to reopen if the issue still happens. Thanks.

closed 2024-12-18 by Alvin1Zhang