espressif/esp-idf #14912: ninja: build stopped: subcommand failed.  HELP!!

Shiguma1379, 2024-11-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build project

### What is the actual behavior?

When I build my project it shows "ninja: build stopped: subcommand failed.
 The terminal process "e:\ASOFTWARE\ESP\tools\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1. "

### Steps to reproduce.

I dont konw how to trigger

### Build or installation Logs.

```plain
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2929:43: error: expected declaration specifiers or '...' before 'UBaseType_t'
 2929 |                                           UBaseType_t uxHighestPriorityWaitingTask ) PRIVILEGED_FUNCTION;
      |                                           ^~~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2934:1: error: expected declaration specifiers before 'UBaseType_t'
 2934 | UBaseType_t uxTaskGetTaskNumber( TaskHandle_t xTask ) PRIVILEGED_FUNCTION;
      | ^~~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2940:26: error: expected declaration specifiers or '...' before 'TaskHandle_t'
 2940 | void vTaskSetTaskNumber( TaskHandle_t xTask,
      |                          ^~~~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2941:32: error: type defaults to 'int' in declaration of  
UBaseType_t' [-Werror=implicit-int]
 2941 |                          const UBaseType_t uxHandle ) PRIVILEGED_FUNCTION;
      |                                ^~~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2941:44: error: expected ';', ',' or ')' before 'uxHandle' 2941 |                          const UBaseType_t uxHandle ) PRIVILEGED_FUNCTION;
      |                                            ^~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2951:21: error: expected declaration specifiers or '...' before 'TickType_t'
 2951 | void vTaskStepTick( TickType_t xTicksToJump ) PRIVILEGED_FUNCTION;      
      |                     ^~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2967:1: error: expected declaration specifiers before 'eSleepModeStatus'
 2967 | eSleepModeStatus eTaskConfirmSleepModeStatus( void ) PRIVILEGED_FUNCTION;
      | ^~~~~~~~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2973:1: error: expected declaration specifiers before 'TaskHandle_t'
 2973 | TaskHandle_t pvTaskIncrementMutexHeldCount( void ) PRIVILEGED_FUNCTION; 
      | ^~~~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:2979:36: error: expected declaration specifiers or '...' before 'TimeOut_t'
 2979 | void vTaskInternalSetTimeOutState( TimeOut_t * const pxTimeOut ) PRIVILEGED_FUNCTION;
      |                                    ^~~~~~~~~
In file included from E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/esp_additions/include/freertos/idf_additions.h:21:
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/queue.h:54:1: warning: empty declaration
   54 | struct QueueDefinition; /* Using old naming convention so as not to break kernel aware debuggers. */
      | ^~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/queue.h:55:36: error: storage class specified for parameter 'QueueHandle_t'
[... 554 lines cut ...]
  645 | TaskHandle_t xTaskGetIdleTaskHandleForCPU( BaseType_t xCoreID )
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:/ASOFTWARE/ESP/container/v5.3.1/esp-idf/components/freertos/esp_additions/include/freertos/idf_additions.h:651:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'xTaskGetCurrentTaskHandleForCPU'
  651 | TaskHandle_t xTaskGetCurrentTaskHandleForCPU( BaseType_t xCoreID )      
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:/Dev/ESP/controller/main/main.c:7:1: error: expected '=', ',', ';', 'asm' or  
__attribute__' before '{' token
    7 | {
      | ^
In file included from E:/Dev/ESP/controller/main/main.c:2:
E:/Dev/ESP/controller/Hardware/Button/Button.h:3:6: error: old-style parameter declarations in prototyped function definition
    3 | void Button(void)
      |      ^~~~~~
E:/Dev/ESP/controller/main/main.c:14: error: expected '{' at end of input       
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.

 *  The terminal process "e:\ASOFTWARE\ESP\tools\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1.
```

### More Information.

Win10
Vscode extension
can build example project

Shiguma1379, 2024-11-19:
This is sdkconfig

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.3.1 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_PHY_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_MPU_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_RNG_SUPPORTED=y
CONFIG_SOC_LIGHT_SLEEP_SUPPORTED=y
CONFIG_SOC_DEEP_SLEEP_SUPPORTED=y
CONFIG_SOC_LP_PERIPH_SHARE_INTERRUPT=y
CONFIG_SOC_PM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_DIGI_MONITOR_NUM=0
[... 1909 lines cut ...]
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

ESP-Marius, 2024-11-20:
Without seeing your project it is a bit hard to diagnose a build failure, but since you can build the example project then this is unlikely to be a problem with esp-idf.

Please try to trim down your project bit by bit to find what introduce the error.

> E:/Dev/ESP/controller/main/main.c:14: error: expected '{' at end of input   `

Suggest you start by checking this.

ESP-Marius, 2024-11-29:
Closing due to lack of new updates. @Shiguma1379 feel free to reopen if you wanna continue troubleshoting this issue

closed 2024-11-29 by ESP-Marius