espressif/esp-idf #14913: clang-check doesn't work

LEI3550, 2024-11-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.3.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I followed the instruction to enable the function of clang-tidy in CMD.exe. The steps were shown as "Steps to reproduce". 

### What is the actual behavior?

The log showed "run-clang-tidy.py not found in your PATH". The clang-check didn't work.

### Steps to reproduce.

1. cd C:\Espressif\frameworks\esp-idf-v5.3.1
2. set IDF_TOOLCHAIN=clang
3. install.bat
4. export.bat
5. idf_tools.py install esp-clang
6. export.bat
7. cd C:\Espressif\frameworks\esp-idf-v5.3.1\examples\get-started\hello_world
8. idf.py fullclean
9. idf.py clang-check
...

### Build or installation Logs.

```plain
Microsoft Windows [Version 10.0.22631.4317]
(c) Microsoft Corporation. All rights reserved.

C:\Windows\System32>set IDF_TOOLCHAIN=clang

C:\Windows\System32>cd C:\Espressif\frameworks\esp-idf-v5.3.1

C:\Espressif\frameworks\esp-idf-v5.3.1>set IDF_TOOLCHAIN=clang

C:\Espressif\frameworks\esp-idf-v5.3.1>install.bat
Installing ESP-IDF tools
Selected targets are: esp32p4, esp32c2, esp32s2, esp32c3, esp32h2, esp32s3, esp32c6, esp32, esp32c61, esp32c5
Current system platform: win64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20240530 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20240530 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping cmake@3.24.0 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20240318 (already installed)
Skipping ninja@1.11.1 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.8 (already installed)
[... 206 lines cut ...]
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.3.1/components/app_trace C:/Espressif/frameworks/esp-idf-v5.3.1/components/app_update C:/Espressif/frameworks/esp-idf-v5.3.1/components/bootloader C:/Espressif/frameworks/esp-idf-v5.3.1/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.3.1/components/bt C:/Espressif/frameworks/esp-idf-v5.3.1/components/cmock C:/Espressif/frameworks/esp-idf-v5.3.1/components/console C:/Espressif/frameworks/esp-idf-v5.3.1/components/cxx C:/Espressif/frameworks/esp-idf-v5.3.1/components/driver C:/Espressif/frameworks/esp-idf-v5.3.1/components/efuse C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp-tls C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_adc C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_bootloader_format C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_coex C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_common C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_ana_cmpr C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_cam C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_dac C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_gpio C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_gptimer C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2c C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2s C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_isp C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_jpeg C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_ledc C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_mcpwm C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_parlio C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_pcnt C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_ppa C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_rmt C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_sdio C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_sdm C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_sdmmc C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_sdspi C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_spi C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_touch_sens C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_tsens C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_uart C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_usb_serial_jtag C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_eth C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_event C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_hid C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_http_client C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_http_server C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_https_server C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_lcd C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_mm C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_netif C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_netif_stack C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_partition C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_phy C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_pm C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_psram C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_ringbuf C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_timer C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_vfs_console C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_wifi C:/Espressif/frameworks/esp-idf-v5.3.1/components/espcoredump C:/Espressif/frameworks/esp-idf-v5.3.1/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.3.1/components/fatfs C:/Espressif/frameworks/esp-idf-v5.3.1/components/freertos C:/Espressif/frameworks/esp-idf-v5.3.1/components/hal C:/Espressif/frameworks/esp-idf-v5.3.1/components/heap C:/Espressif/frameworks/esp-idf-v5.3.1/components/http_parser C:/Espressif/frameworks/esp-idf-v5.3.1/components/idf_test C:/Espressif/frameworks/esp-idf-v5.3.1/components/ieee802154 C:/Espressif/frameworks/esp-idf-v5.3.1/components/json C:/Es [... cut ...]
  C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/build.cmake:632 (__build_process_project_includes)
  C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)

****** Filter files and dirs ******
Files to be analysed:
+ > C:\Espressif\frameworks\esp-idf-v5.3.1\examples\get-started\hello_world\main\hello_world_main.c
***********************************
run-clang-tidy.py not found in your PATH

C:\Espressif\frameworks\esp-idf-v5.3.1\examples\get-started\hello_world>
```

mfialaf, 2024-12-02:
Hi @LEI3550,
The issue should be solved by now. Please update your `pyclang` either by running the install script again or by `pip install --upgrade pyclang`.

I am closing the issue, let us know if you need anything else.

dobairoland, 2024-12-02:
For future reference, this has been fixed in `pyclang==0.6.0` by https://github.com/espressif/clang-tidy-runner/pull/50.

closed 2024-12-02 by dobairoland