espressif/esp-idf #14916: E (123456) wifi_init_default: netstack cb reg failed with 12308

kriegste, 2024-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In ESP-IDF 5.3.1 with ESP32.

After using
esp_wifi_set_mode(WIFI_MODE_STA);
to disable the AP, this error appears:

> E (123456) wifi_init_default: netstack cb reg failed with 12308

What does this error mean? And how do I fix it?

kriegste, 2024-11-21:
I do a
> esp_wifi_set_protocol(ESP_IF_WIFI_AP, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR);
just before changing the mode. This seems to be the cause of the error. How can I avoid this problem?

hansw123, 2024-11-27:
@kriegste 
sorry for late reply ，could you please put the whole code which can reproduce the issue, this may help us to find the true reason correctly and quickly

AxelLin, 2024-11-27:
> > E (123456) wifi_init_default: netstack cb reg failed with 12308
> 
> What does this error mean? And how do I fix it?

ESP_ERR_WIFI_STOP_STATE: /* 12308 0x3014 Returned when WiFi is stopping */

kriegste, 2024-11-27:
Try these two lines:

> esp_wifi_set_protocol(ESP_IF_WIFI_AP, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_11AX | WIFI_PROTOCOL_LR);
> esp_wifi_set_mode(WIFI_MODE_STA);

You might ask why I do this.

esp_wifi_set_protocol() always enables/disables the LR protocol for both the STA and AP interface. This is a general problem in ESP-IDF.
When my app needs to open an access point, it has to disable the LR protocol. Otherwise the access point cannot be found by any phone or computer.
So when it closes the access point again (this is the code above) the LR protocol needs to be enabled again.

kriegste, 2024-11-27:
Of course, it would be best to finally solve the underlying root issue, which is this:
https://github.com/espressif/esp-idf/issues/9978

hansw123, 2024-11-29:
@kriegste 
this err means the program is doing wifi stop, and we dont allow other apis to interrupt this action or doing wifi stop again.you should check your code about releated api which will do wifi stop

kriegste, 2024-11-29:
You mean esp_wifi_stop()? There is not a single call to this API in my whole app.

hansw123, 2024-11-29:
@kriegste 
other apis may have wifi stop action such as  change wifi mode, change config and so on.
may be you can post more detailed code for refer,thanks!

kriegste, 2024-12-01:
I am working on a minimum code example. It is a bit harder than I thought.

hansw123, 2025-01-23:
@kriegste  could you offer your code which can reproduce the issue ， or we can not find the root cause

kriegste, 2025-01-23:
I could narrow the problem down to the use of esp_wifi_internal_set_retry_counter() (I have very poor WiFi in one spot of the building and was experimenting with this function).

Total-Reality, 2025-02-07:
We have the same problem with this error. 
You can find a code example here:
https://www.az-delivery.de/en/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/internet-radio-mit-dem-esp32

_"Connecting to WiFi"
"E (2481) wifi_init_default: netstack cb reg failed with 12308"_

You can click on "Sketch to download" to get the file.

hansw123, 2025-02-12:
@kriegste  
esp_wifi_internal_set_retry_counter() only be called in mesh ,did you use mesh ?

closed 2025-05-08 by Alvin1Zhang