espressif/esp-idf #14917: coredump corrupted on 5.1.4+

Tarik2142, 2024-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4 , 5.3.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

Successful coredump decoding

### What is the actual behavior?

Failed to load core dump: Invalid core dump CRC 5fd7ffd, should be ffffffff

### Steps to reproduce.

```
#include <Arduino.h>

void setup() {
    delay(10000);
    assert(0);
}

void loop() {
    delay(1000);
}
```

### Debug Logs.

```plain
// First reboot - OK
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (S0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8

assert failed: void setup() main.cpp:5 (0)

Backtrace: 0x40082eb1:0x3ffb20e0 0x400876b9:0x3ffb2100 0x4008c35a:0x3ffb2120 0x400d1627:0x3ffb2250 0x400d17ee:0x3ffb2270 0x400896a1:0x3ffb2290
  #0  0x40082eb1 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x400876b9 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x4008c35a in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x400d1627 in setup() at src/main.cpp:5
  #4  0x400d17ee in loopTask(void*) at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59
  #5  0x400896a1 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 5520ccd13053a328

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr
Core  1 register dump:
PC      : 0x400837c0  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffc2630
A2      : 0x00000000  A3      : 0x70000005  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000800  A9      : 0x3ffb1f20
A10     : 0x00000000  A11     : 0x1163e1ea  A12     : 0x1163e1ea  A13     : 0x00060b24
A14     : 0xfffbfff0  A15     : 0x00000046  SAR     : 0x00000000  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x40085b88  LEND    : 0x40085b9e  LCOUNT  : 0xffffffff

Backtrace: 0x400837bd:0x3ffc2630 0x40080303:0x3ffc2660 0x65a5a5a2:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: 5520ccd13053a328

Re-entered core dump! Exception happened during core dump!
Rebooting...

// All of the following reboots coredump are broken!
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
E (169) esp_core_dump_flash: ����e dump data check failed:
Calculated checksum='a1ca788b'
Image checksum='ffffffff'

assert failed: void setup() main.cpp:5 (0)

Backtrace: 0x40082eb1:0x3ffb20e0 0x400876b9:0x3ffb2100 0x4008c35a:0x3ffb2120 0x400d1627:0x3ffb2250 0x400d17ee:0x3ffb2270 0x400896a1:0x3ffb2290
  #0  0x40082eb1 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x400876b9 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x4008c35a in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x400d1627 in setup() at src/main.cpp:5
  #4  0x400d17ee in loopTask(void*) at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59
  #5  0x400896a1 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 5520ccd13053a328

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr
Core  1 register dump:
PC      : 0x400837c0  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffc2630
A2      : 0x00000000  A3      : 0x70000005  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000800  A9      : 0x3ffb1f20
A10     : 0x00000000  A11     : 0x81c99f9d  A12     : 0x81c99f9d  A13     : 0x00060b24
A14     : 0xfffbfff0  A15     : 0x00000046  SAR     : 0x00000000  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x40085b88  LEND    : 0x40085b9e  LCOUNT  : 0xffffffff

Backtrace: 0x400837bd:0x3ffc2630 0x40080303:0x3ffc2660 0x65a5a5a2:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: 5520ccd13053a328

Re-entered core dump! Exception happened during core dump!
Rebooting...

// broken
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
E (169) esp_core_dump_flash: ����e dump data check failed:
Calculated checksum='a1ca788b'
Image checksum='ffffffff'
```

### More Information.

I am using the Arduino framework based on IDF 5.1.4. I also tried Arduino core 3.1.0 rc3 based on IDF 5.3.1 and have the same problem.
I completely wiped the chip and flashed it from scratch.
I didn't get this problem on core 2.x (IDF v4.4.3)

I get: ```E (169) esp_core_dump_flash: ����e dump data check failed: Calculated checksum='a1ca788b' Image checksum='ffffffff'```
after the first ```assert(0)``` and coredump doesn't work anymore.
Also, the device does not always reboot after a crash, it can remain in halt mode forever.

Tarik2142, 2024-11-21:
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.1.4 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_IDCACHE_PER_CORE=y
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_MAX_REGION_SIZE=64
CONFIG_SOC_DAC_CHAN_NUM=2
CONFIG_SOC_DAC_RESOLUTION=8
[... 2639 lines cut ...]
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_ULP_COPROC_RESERVE_MEM=512
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND=3000
CONFIG_MB_MASTER_DELAY_MS_CONVERT=200
CONFIG_MB_QUEUE_LENGTH=20
CONFIG_MB_SERIAL_TASK_STACK_SIZE=4096
CONFIG_MB_SERIAL_BUF_SIZE=256
CONFIG_MB_SERIAL_TASK_PRIO=10
# CONFIG_MB_CONTROLLER_SLAVE_ID_SUPPORT is not set
CONFIG_MB_CONTROLLER_NOTIFY_TIMEOUT=20
CONFIG_MB_CONTROLLER_NOTIFY_QUEUE_SIZE=20
CONFIG_MB_CONTROLLER_STACK_SIZE=4096
CONFIG_MB_EVENT_QUEUE_TIMEOUT=20
CONFIG_MB_TIMER_PORT_ENABLED=y
# End of deprecated options

Tarik2142, 2024-11-21:
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.1.4 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_IDCACHE_PER_CORE=y
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_MAX_REGION_SIZE=64
CONFIG_SOC_DAC_CHAN_NUM=2
[... 2641 lines cut ...]
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_ULP_COPROC_RESERVE_MEM=512
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_MB_MASTER_TIMEOUT_MS_RESPOND=3000
CONFIG_MB_MASTER_DELAY_MS_CONVERT=200
CONFIG_MB_QUEUE_LENGTH=20
CONFIG_MB_SERIAL_TASK_STACK_SIZE=4096
CONFIG_MB_SERIAL_BUF_SIZE=256
CONFIG_MB_SERIAL_TASK_PRIO=10
# CONFIG_MB_CONTROLLER_SLAVE_ID_SUPPORT is not set
CONFIG_MB_CONTROLLER_NOTIFY_TIMEOUT=20
CONFIG_MB_CONTROLLER_NOTIFY_QUEUE_SIZE=20
CONFIG_MB_CONTROLLER_STACK_SIZE=4096
CONFIG_MB_EVENT_QUEUE_TIMEOUT=20
CONFIG_MB_TIMER_PORT_ENABLED=y
# End of deprecated options
```

erhankur, 2024-11-22:
@Tarik2142  I have followed the steps below to run an Arduino example

```
idf.py create-project-from-example "espressif/arduino-esp32^3.0.2:hello_world"
```

I have added assert(0) to the main.cpp

```
#include "Arduino.h"

void setup() {
  Serial.begin(115200);
  assert(0);
}

void loop() {
  Serial.println("Hello world!");
  delay(1000);
}
```
And additional configs
```
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP_COREDUMP_CHECKSUM_SHA256=y
CONFIG_ESP32_PANIC_PRINT_HALT=y
```

Result:

```
I (33) boot: ESP-IDF v5.1.5-196-g64849cb703-dirty 2nd stage bootloader
I (33) boot: compile time Nov 22 2024 09:45:46
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v3.0
I (43) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 2MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot:  3 coredump         Unknown data     01 03 00110000 00010000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0e8a4h ( 59556) map
I (137) esp_image: segment 1: paddr=0001e8cc vaddr=3ffb0000 size=0174ch (  5964) load
I (140) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=234f8h (144632) map
I (195) esp_image: segment 3: paddr=00043520 vaddr=3ffb174c size=01034h (  4148) load
I (197) esp_image: segment 4: paddr=0004455c vaddr=40080000 size=0da34h ( 55860) load
I (231) boot: Loaded app from partition at offset 0x10000
I (231) boot: Disabling RNG early entropy source...
I (243) cpu_start: Multicore app
I (244) cpu_start: Pro cpu up.
I (244) cpu_start: Starting app cpu, entry point is 0x40081324
--- 0x40081324: call_start_cpu1 at /Users/erhan/dev/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (261) cpu_start: Pro cpu start user code
I (261) cpu_start: cpu freq: 160000000 Hz
I (261) cpu_start: Application information:
I (266) cpu_start: Project name:     main
I (270) cpu_start: App version:      1
I (275) cpu_start: Compile time:     Nov 22 2024 09:45:45
I (281) cpu_start: ELF file SHA256:  39b973e583883752...
I (287) cpu_start: ESP-IDF:          v5.1.5-196-g64849cb703-dirty
I (294) cpu_start: Min chip rev:     v0.0
I (298) cpu_start: Max chip rev:     v3.99 
I (303) cpu_start: Chip rev:         v3.0
I (308) heap_init: Initializing. RAM available for dynamic allocation:
I (315) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (321) heap_init: At 3FFB3820 len 0002C7E0 (177 KiB): DRAM
I (328) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (334) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (340) heap_init: At 4008DA34 len 000125CC (73 KiB): IRAM
I (348) spi_flash: detected chip: generic
I (351) spi_flash: flash io: dio
W (355) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (369) esp_core_dump_flash: Init core dump to flash
I (374) esp_core_dump_flash: Found partition 'coredump' @ 110000 65536 bytes
I (395) esp_core_dump_flash: Core dump data checksum is correct
I (395) esp_core_dump_flash: Found core dump 9476 bytes in flash @ 0x110000
I (399) app_start: Starting scheduler on CPU0
I (403) app_start: Starting scheduler on CPU1
I (403) main_task: Started on CPU0
I (413) main_task: Calling app_main()
I
assert failed: void setup() main.cpp:5 (0)

Backtrace: 0x40081956:0x3ffb8c80 0x40086ff5:0x3ffb8ca0 0x4008c609:0x3ffb8cc0 0x400dbe6a:0x3ffb8de0 0x400d8836:0x3ffb8e10 0x400895b1:0x3ffb8e30
--- 0x40081956: panic_abort at /Users/erhan/dev/esp-idf/components/esp_system/panic.c:466
0x40086ff5: esp_system_abort at /Users/erhan/dev/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x4008c609: __assert_func at /Users/erhan/dev/esp-idf/components/newlib/assert.c:81
0x400dbe6a: setup() at /Users/erhan/workspace/arduino/hello_world/main/main.cpp:5
0x400d8836: loopTask(void*) at /Users/erhan/workspace/arduino/hello_world/managed_components/espressif__arduino-esp32/cores/esp32/main.cpp:59
0x400895b1: vPortTaskWrapper at /Users/erhan/dev/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 39b973e583883752

I (185) esp_core_dump_flash: Save core dump to flash...
I (191) esp_core_dump_flash: Erase flash 12288 bytes @ 0x110000
I (452) esp_core_dump_flash: Write end offset 0x2504, check sum length 4
I (452) esp_core_dump_flash: Core dump has been saved to flash.
CPU halted.

```

Can you share the steps to prepare an application to reproduce the issue?

Tarik2142, 2024-11-22:
@erhankur here is example pio project
[coredump_test.zip](https://github.com/user-attachments/files/17870523/coredump_test.zip)

erhankur, 2024-11-22:
Still no error. I have installed from scratch. 

```
rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
[     2][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d2dd4
  #0  0x400d2dd4 in _uartDetachBus_RX at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153

[    13][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d2da4
  #0  0x400d2da4 in _uartDetachBus_TX at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:160

[    27][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d2d74
  #0  0x400d2d74 in _uartDetachBus_CTS at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:167

[    40][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d2d44
  #0  0x400d2d44 in _uartDetachBus_RTS at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:174

[    54][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d2dd4
  #0  0x400d2dd4 in _uartDetachBus_RX at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153

[    67][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d2da4
  #0  0x400d2da4 in _uartDetachBus_TX at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:160

[    81][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d2d74
  #0  0x400d2d74 in _uartDetachBus_CTS at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:167

[    94][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d2d44
  #0  0x400d2d44 in _uartDetachBus_RTS at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:174

[   108][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d2dd4
  #0  0x400d2dd4 in _uartDetachBus_RX at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153

[   121][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d2da4
  #0  0x400d2da4 in _uartDetachBus_TX at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:160

[   135][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d2d74
  #0  0x400d2d74 in _uartDetachBus_CTS at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:167

[   148][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d2d44
  #0  0x400d2d44 in _uartDetachBus_RTS at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:174

[   178][D][esp32-hal-cpu.c:264] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   196][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb60
  #0  0x3ffbdb60 in ?? at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:82

[   207][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb60
  #0  0x3ffbdb60 in ?? at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:82

=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 3.00
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
[... 40 lines cut ...]
  Core Debug Level  : 5
  Arduino Runs Core : 1
  Arduino Events on : 1
  CDC On Boot       : 0
============ Before Setup End ============
[   708][V][esp32-hal-uart.c:408] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   717][V][esp32-hal-uart.c:497] uartBegin(): UART0 not installed. Starting installation
[   728][V][esp32-hal-uart.c:560] uartBegin(): UART0 initialization done.
coredump test

assert failed: void setup() main.cpp:7 (0)

Backtrace: 0x40082f7d:0x3ffb20d0 0x400880ed:0x3ffb20f0 0x4008cf26:0x3ffb2110 0x400d17c7:0x3ffb2240 0x400d3b6b:0x3ffb2270 0x4008a26d:0x3ffb2290
  #0  0x40082f7d in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x400880ed in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
[... cut ...] ilder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x400d17c7 in setup() at src/main.cpp:7
  #4  0x400d3b6b in loopTask(void*) at /Users/erhan/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59
  #5  0x4008a26d in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: e524f3e79e80dbc2

Rebooting...
ets Jul 29 2019 12:21:46
```

Tarik2142, 2024-11-22:
> Still no error. I have installed from scratch.
>
> ```
> rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee [...]

first time load everything is fine, but the next time you load it you will see an error.

Tarik2142, 2024-11-22:
This is strange. I see that your device rebooted after the crash when mine doesn't reboot.

erhankur, 2024-11-22:
My pio.ini
```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
board = esp32dev
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
build_type = debug
build_flags =
    -DCORE_DEBUG_LEVEL=5
    -DCONFIG_ESP_COREDUMP_ENABLE=1
    -DCONFIG_ESP_COREDUMP_TO_FLASH=1
    -DCONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=1
```

Tarik2142, 2024-11-22:
This is what I get on first boot:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)  
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drx00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
[     2][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d2dd4
  #0  0x400d2dd4 in _uartDetachBus_RX at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153

[    13][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d2da4
  #0  0x400d2da4 in _uartDetachBus_TX at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:160

[    27][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d2d74
  #0  0x400d2d74 in _uartDetachBus_CTS at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:167

[    40][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d2d44
  #0  0x400d2d44 in _uartDetachBus_RTS at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:174

[    54][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d2dd4
  #0  0x400d2dd4 in _uartDetachBus_RX at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153

[    67][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d2da4
  #0  0x400d2da4 in _uartDetachBus_TX at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:160

[    81][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d2d74
  #0  0x400d2d74 in _uartDetachBus_CTS at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:167

[    94][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d2d44
  #0  0x400d2d44 in _uartDetachBus_RTS at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:174

[   108][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d2dd4
  #0  0x400d2dd4 in _uartDetachBus_RX at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153

[   121][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d2da4
  #0  0x400d2da4 in _uartDetachBus_TX at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:160

[   135][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d2d74
  #0  0x400d2d74 in _uartDetachBus_CTS at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:167

[   148][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d2d44
  #0  0x400d2d44 in _uartDetachBus_RTS at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:174

[   165][D][esp32-hal-cpu.c:264] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   181][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb60
  #0  0x3ffbdb60 in ?? at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:82

[   192][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb60
  #0  0x3ffbdb60 in ?? at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:82

=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 3.00
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
[... 174 lines cut ...]
------------------------------------------
  Arduino Board     : Espressif ESP32 Dev Module
  Arduino Variant   : esp32
  Core Debug Level  : 5
  Arduino Runs Core : 1
  Arduino Events on : 1
  CDC On Boot       : 0
============ Before Setup End ============
[   713][V][esp32-hal-uart.c:408] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   722][V][esp32-hal-uart.c:497] uartBegin(): UART0 not installed. Starting installation
[   733][V][esp32-hal-uart.c:560] uartBegin(): UART0 initialization done.
coredump test

assert failed: void setup() main.cpp:7 (0)

Backtrace: 0x40082f7d:0x3ffb20d0 0x400880ed:0x3ffb20f0 0x4008cf26:0x3ffb2110 0x400d17c7:0x3ffb2240 0x400d3b6b:0x3ffb2270 0x4008a26d:0x3ffb2290
[... cut ...]
  #2  0x4008cf26 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x400d17c7 in setup() at src/main.cpp:7
  #4  0x400d3b6b in loopTask(void*) at C:/Users/Tarik2142/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59
  #5  0x4008a26d in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 967d3c79df889a91
```

tl-sl, 2024-11-23:
Testing with esp-idf across different versions, I think this issue is caused by this setting:
`CONFIG_ESP_COREDUMP_STACK_SIZE=1024`
https://github.com/espressif/esp32-arduino-lib-builder/blob/22ce6811335c695363a37745afec36ff4f580f79/configs/defconfig.common#L116
v5.1.x  - fails
v5.2,x -  works
v5.3.x  - fails with different error.

Setting this `0` per default or larger at `2048` fixes the issues with corrupt coredumps.

Tarik2142, 2024-11-23:
I'm closing this as it's an arduino configuration problem.

closed 2024-11-23 by Tarik2142