espressif/esp-idf #14919: wifi 802.11 tx on channel 1 only

chegewara, 2024-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi team,
sorry for posting this on github issues instead of forum, but its important in my project.

We have 802.11 beaconing which is sending CCX, IBSS and WDS packets on different channels, but wireshark can catch packets only on channel 1 and im not sure its issue with wireshark or esp32.

Logic:
- run wifi in STA mode
- not associated with any STA
- TX on channels 1, 6, 11

esp-idf 5.3 and master

Thanks

zhangyanjiaoesp, 2024-11-26:
@chegewara 
Are you calling `esp_wifi_set_channel()` and then calling  `esp_wifi_80211_tx()` ?

zhangyanjiaoesp, 2024-11-26:
Are you using ESP32?

chegewara, 2024-11-26:
Yes, its esp32 pico V2

zhangyanjiaoesp, 2024-11-26:
Hi, I just test on the master branch, using the `station` example, on ESP32. When the channel change to 6/11, I can capture the packets on channel 6/11.
[image]
[image]
[image]

chegewara, 2024-11-26:
Thank you for confirming this.
Have a nice weekend.

closed 2024-11-26 by chegewara