espressif/esp-idf #14920: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT

Sotatek-HieuBui2, 2024-11-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF version.
v5.3.1

Espressif SoC revision.
ESP32-S3

Operating System used.
Windows

How did you build your project?
Command line with idf.py

If you are using Windows, please specify command line type.
CMD

Development Kit.
Custom board with esp32s3-wroom-1

Power Supply used.
USB

Context: I try to run example ble-spp-client, when run case ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT, there are error like this:
[image]

Please help me.

esp-zhp, 2024-11-25:
Is there still an issue with the latest branch?

esp-zhp, 2024-11-25:
E (483) BT_BTM: BTM_BleScan scan already active
The reason for this error is that the scan has already started, but you called "start scan" again. Please check your code to ensure that it is not being called repeatedly.

Sotatek-HieuBui2, 2024-12-05:
I only use example gatt_client, and still not adjust code ( esp_ble_gap_start_scanning() is called once in case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT ).

esp-zhp, 2024-12-09:
v4.4 is no longer maintained. Could you check if the latest branch still has this issue?  
The issue occurs because the scan hadn't stopped before another `start scan` was called.  
Perhaps this problem does exist in the v4.4 branch. You can try stopping the scan first before calling a new scan. I believe this should resolve your issue.

Alvin1Zhang, 2025-05-08:
Thanks for reporting, feel free to reopen if the issue still happens, thanks.

closed 2025-05-08 by Alvin1Zhang