espressif/esp-idf #14921: target_linker_script with ULP firmware

chrisruk, 2024-11-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm able to make use of the following syntax in my `main/CMakeLists.txt`
in order to add a custom linker script configuration to my main application.

```
target_linker_script(${COMPONENT_LIB} INTERFACE "script.ld.in" )
```

However it doesn't appear to be possible to do the same for the ULP firmware I'm creating.

I tried something like:

```
target_linker_script("__idf_ulp" INTERFACE "script.ld.in" )
````

But that doesn't appear to have any effect.

Just wondering if anyone could point me in the right direction.  

Many thanks.

o-marshmallow, 2024-11-25:
Hello @chrisruk ,

Which target are you using? ESP32-S3 or ESP32-C6? Also, which version of IDF are you working on?

On the latest version of IDF, you can add your own linker scripts to the linking phase with plain CMake commands. For example, let's say you have a file `new.ld` that you want to include in your LP project, add this to its `CMakeLists.txt` file:
```
add_custom_target(my_ld_script DEPENDS "${CMAKE_CURRENT_LIST_DIR}/new.ld")
add_dependencies(${ULP_APP_NAME} my_ld_script)
target_link_options(${ULP_APP_NAME} PRIVATE "-T${CMAKE_CURRENT_LIST_DIR}/new.ld")
```
Assuming that `new.ld` is in the same directory as your `CMakeLists.txt` file. 

Make sure to add is in the LP `CMakeLists.txt` and **not** the HP `CMakeLists.txt` file!

o-marshmallow, 2024-11-27:
Hello @chrisruk !

Thanks for the confirmation. I see what you mean, the function `ulp_embed_binary` doesn't let you customize the build for the LP core. 

I was relying on the `build_system` example which has a more complete `CMakeLists.txt` file for the ULP program and lets you customize the build even more: https://github.com/espressif/esp-idf/blob/master/examples/system/ulp/lp_core/build_system/main/ulp/CMakeLists.txt

In such file, you can add the snippet I sent previously and you should be able to integrate your lienr script file to your linking.

closed 2024-11-27 by chrisruk

o-marshmallow, 2024-11-28:
@chrisruk Thanks for your feedback!