espressif/esp-idf #14924: [esp32-p4] RTC_NOINIT_ATTR vs __NOINIT_ATTR

chegewara, 2024-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

esp32-p4

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

devkit

### Power Supply used.

USB

### What is the expected behavior?

Is it a bug, or its working differently than on S3 and other chips?

### What is the actual behavior?

has always value `29966728` after soft reboot

### Steps to reproduce.

```
#include <stdio.h>
#include <stdint.h>

#include "freertos/FreeRTOS.h"

#include "esp_system.h"
#include "esp_attr.h"

static RTC_NOINIT_ATTR uint32_t s_rtc_noinit_val;
static __NOINIT_ATTR uint32_t s_reconfigure_requested;
static const uint32_t RECONFIGURE_REQUEST = 0x1C55AA;

void app_main(void)
{

    if (esp_reset_reason() == ESP_RST_POWERON)
    {
        s_rtc_noinit_val = 1;
    }

    if (s_reconfigure_requested == RECONFIGURE_REQUEST)
    {
        printf("\t\t its working\n");
    } else {
        printf("\t\t its NOT working\n");
        printf("\t value: %lu => %ld\n", s_reconfigure_requested, RECONFIGURE_REQUEST);
    }
    s_reconfigure_requested = RECONFIGURE_REQUEST;
    printf("\t value: %ld\n", s_reconfigure_requested);
    vTaskDelay(100);

    esp_restart();
}
```

### Debug Logs.

```plain
rst:0xc (SW_CPU_RESET),boot:0x30f (SPI_FAST_FLASH_BOOT)

...

         value: 29966728 => 1856938
```
```

### More Information.

Thanks

ESP-Marius, 2024-11-25:
@chegewara What you are seeing here is probably due to DRAM being behind a cache on P4. 

So if what the app does between setting  `s_reconfigure_requested` and restarting is not sufficient to cause a cache writeback then then at the point of the restart your new value might still just exist in the cache.

You can try to add a `cache_ll_writeback_all(CACHE_LL_LEVEL_INT_MEM, CACHE_TYPE_DATA, CACHE_LL_ID_ALL);` before the restart to see if it gives you the expected behavior

ginkgm, 2025-09-24:
Closed due to lack of reply