espressif/esp-idf #14926: Clarify the Flash/PSRAM capabilities of the ESP32-S3-MINI-1

ProfFan, 2024-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

Is there any documentation on the DDR/120MHz rate support of the in-package PSRAM of the ESP32-S3-MINI-1 modules (which according to the data sheet contained a ESP32-S3FH4R2)?

Thanks!

pacucha42, 2024-11-24:
Hi @ProfFan ,
ESP32-S3FH4R2 is equipped with 4MB flash (FH4) and 2MB PSRAM (R2), both connected via QuadSPI. Quad PSRAM supports only STR mode, DDR won't work for you. See https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/flash_psram_config.html for more details

ProfFan, 2024-11-28:
@pacucha42 Will 120MHz work for flash/PSRAM? I saw a section in the data sheet that mentions this, but no support in IDF?

omnitex, 2026-11-13:
Hi @ProfFan, sorry for the late answer

Yes, 120 MHz works for both the flash and the PSRAM on the ESP32-S3FH4R2 (ESP32-S3-MINI-1). Both run in STR mode; DDR isn't available with quad PSRAM, as mentioned above. This has been supported since v4.4 ([a3dc625](https://github.com/espressif/esp-idf/commit/a3dc625da6e9634c056b48c3319980dca083f586)).

The 120 MHz flash option is hidden in menuconfig until you enable flash high-performance mode (HPM), which is probably why it looked missing. To turn it on:

```
CONFIG_SPI_FLASH_HPM_ENA=y
CONFIG_ESPTOOLPY_FLASHFREQ_120M=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_QUAD=y
CONFIG_SPIRAM_SPEED_120M=y
```

This is the same combination we test in CI: [sdkconfig.ci.esp32s3_f4r4_120sdr_120sdr](https://github.com/espressif/esp-idf/blob/master/components/esp_mspi/test_apps/mspi/sdkconfig.ci.esp32s3_f4r4_120sdr_120sdr)

**Checking it on your board:** whether flash HPM is supported depends on the flash die inside the package.
1. Run `esptool.py --chip esp32s3 -p PORT flash_id` and note the manufacturer and device ID (e.g. `0xC84016`).
2. Check whether that ID is listed in [spi_flash_hpm_enable.c](https://github.com/espressif/esp-idf/blob/master/components/esp_mspi/spi_flash_hpm_enable.c). If it's listed, 120 MHz flash is supported.
3. Build with the config above and check the boot log. You should see `Enabling flash high speed mode by ...`. If you see `High performance mode of this flash model hasn't been supported.` instead, that flash can't run at 120 MHz. If the flash uses the dummy-cycle method (most XMC and GD E-series parts), also enable `CONFIG_SPI_FLASH_HPM_DC`.

Docs:
- [Flash and PSRAM modes and speeds](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/flash_psram_config.html) (your chip falls under the "F4R4 Hardware" table, quad flash with quad PSRAM)
- [Optional flash features (HPM)](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html)

Code:
- [Kconfig.flash_freq](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/esp32s3/Kconfig.flash_freq): the 120 MHz flash option and its HPM dependency
- [Kconfig.spiram](https://github.com/espressif/esp-idf/blob/master/components/esp_psram/esp32s3/Kconfig.spiram#L89): `SPIRAM_SPEED_120M`
- [mspi_timing_tuning_configs.h](https://github.com/espressif/esp-idf/blob/master/components/esp_mspi/mspi_timing_tuning/port/esp32s3/mspi_timing_tuning_configs.h): timing tuning at 120 MHz

I'll close this as answered. Feel free to reopen if anything doesn't work.

closed 2026-11-13 by omnitex