espressif/esp-idf #14927: 请问如何实现HID master mode

jankens, 2024-11-24:
### Is your feature request related to a problem?

      我之前使用csr8811 模块实现蓝牙HID通讯，最近希望使用esp32模块替换csr8811。移植过程中遇到一个HID master mode（设备主动发起连接）无法连接windows的问题。我使用idf-5.1.3。使用esp_hid_mouse_device——mouse 例程。如果使用HID slave mode，windows可以正常发现并连接esp32 module。当使用HID master mode的时候，windows会弹出确认连接对话框。但是点击允许连接后，无法正常连上hid的control和interrupt PSM。
       我使用Microsoft提供的抓包工具。分别抓了ESP32 和CSR8811的封包。并且修改ESP32 的代码，使得封包基本与CSR8811完全一致。但是仍然无法正常连上。请问有没有人能帮我看看问题是出在什么地方。谢谢！
      附件为2个模块的抓包文件
[hid master.zip](https://github.com/user-attachments/files/17893561/hid.master.zip)

ESP-YTGerd, 2024-11-28:
哈咯, @jankens 
附件中的抓包文件可看到交互是正常的, 方便发一下修改完example esp_hid_mouse_device让esp32作为master和hid device的源码嘛？

jankens, 2024-12-09:
问题已经解决了，谢谢你的支持！
[image]

closed 2025-03-07 by Alvin1Zhang