espressif/esp-idf #14930: ULP interrupt handler corrupts stack

baoshi, 2024-11-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a program using IDF 5.3.1 on ESP32-S3 (on Mac, dev kit C at the moment).  The program has extensive using of ULP RTC IO interrupt. I found in some cases the program behaviors strange after interrupt is triggered. I'm able to zoom into the problem as follows:
In ulp_riscv_vectors.S, saving of registers is implemented as:
```
    .equ SAVE_REGS, 17
    .equ CONTEXT_SIZE, (SAVE_REGS * 4)

.macro save_general_regs cxt_size=CONTEXT_SIZE
    addi sp, sp, -\cxt_size
    sw   ra, RV_STK_RA(sp)
    sw   tp, RV_STK_TP(sp)
    sw   t0, RV_STK_T0(sp)
    sw   t1, RV_STK_T1(sp)
    sw   t2, RV_STK_T2(sp)
    sw   a0, RV_STK_A0(sp)
    sw   a1, RV_STK_A1(sp)
    sw   a2, RV_STK_A2(sp)
    sw   a3, RV_STK_A3(sp)
    sw   a4, RV_STK_A4(sp)
    sw   a5, RV_STK_A5(sp)
    sw   a6, RV_STK_A6(sp)
    sw   a7, RV_STK_A7(sp)
    sw   t3, RV_STK_T3(sp)
    sw   t4, RV_STK_T4(sp)
    sw   t5, RV_STK_T5(sp)
    sw   t6, RV_STK_T6(sp)
.endm
```

This macro compiles into the assembly code:
```
00000010 <irq_vector>:
  10:	fbc10113          	add	sp,sp,-68
  14:	c206                	sw	ra,4(sp)
  16:	c812                	sw	tp,16(sp)
  18:	ca16                	sw	t0,20(sp)
  1a:	cc1a                	sw	t1,24(sp)
  1c:	ce1e                	sw	t2,28(sp)
  1e:	d42a                	sw	a0,40(sp)
  20:	d62e                	sw	a1,44(sp)
  22:	d832                	sw	a2,48(sp)
  24:	da36                	sw	a3,52(sp)
  26:	dc3a                	sw	a4,56(sp)
  28:	de3e                	sw	a5,60(sp)
  2a:	c0c2                	sw	a6,64(sp)
  2c:	c2c6                	sw	a7,68(sp)
  2e:	d8f2                	sw	t3,112(sp)
  30:	daf6                	sw	t4,116(sp)
  32:	dcfa                	sw	t5,120(sp)
  34:	defe                	sw	t6,124(sp)
```

It seems some registers are saved outside of the allocated stack space, that causes the stack to be corrupted.

Hope there is an easy fix.

baoshi, 2024-11-25:
I modified ulp_riscv_vectors.S as follows to solve this problem:
```
.macro save_general_regs cxt_size=CONTEXT_SIZE
    addi sp, sp, -\cxt_size
    sw   ra, 0(sp)
    sw   tp, 4(sp)
    sw   t0, 8(sp)
    sw   t1, 12(sp)
    sw   t2, 16(sp)
    sw   a0, 20(sp)
    sw   a1, 24(sp)
    sw   a2, 28(sp)
    sw   a3, 32(sp)
    sw   a4, 36(sp)
    sw   a5, 40(sp)
    sw   a6, 44(sp)
    sw   a7, 48(sp)
    sw   t3, 52(sp)
    sw   t4, 56(sp)
    sw   t5, 60(sp)
    sw   t6, 64(sp)
.endm

/* Restore the general purpose registers (excluding gp) from the context on
 * the stack. The context is then deallocated. The default size is CONTEXT_SIZE
 * but it can be overridden. */
.macro restore_general_regs cxt_size=CONTEXT_SIZE
    lw   ra, 0(sp)
    lw   tp, 4(sp)
    lw   t0, 8(sp)
    lw   t1, 12(sp)
    lw   t2, 16(sp)
    lw   a0, 20(sp)
    lw   a1, 24(sp)
    lw   a2, 28(sp)
    lw   a3, 32(sp)
    lw   a4, 36(sp)
    lw   a5, 40(sp)
    lw   a6, 44(sp)
    lw   a7, 48(sp)
    lw   t3, 52(sp)
    lw   t4, 56(sp)
    lw   t5, 60(sp)
    lw   t6, 64(sp)
    addi sp,sp, \cxt_size
.endm
```

But it requires modify file inside ESP-IDF, which is problem for future upgrading.

o-marshmallow, 2024-11-25:
Hello @baoshi !

Thanks for reporting this issue and for your diagnostic, I am going to look into this.

closed 2024-12-03 by espressif-bot