espressif/esp-idf #14931: GPIO checks for PULL_UP/DOWN doesn't work for RTC pins.

IhorNehrutsa, 2024-11-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Functions gpio_ll_get_io_config(), gpio_ll_pullup_is_enabled(), gpio_ll_pulldown_is_enabled() work through the IO_MUX_x_REG

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/hal/esp32/include/hal/gpio_ll.h#L66-L68

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/hal/esp32/include/hal/gpio_ll.h#L108-L111

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/hal/esp32/include/hal/gpio_ll.h#L143-L146

But for pins that can operate via RTC, PULL_UP/DOWN is set via the RTC mechanism (see line 86, line 105), not MUX_REG mechanism.

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/esp_driver_gpio/src/gpio.c#L76-L112

See;
https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/hal/esp32/include/hal/rtc_io_ll.h#L178-L183

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/hal/esp32/include/hal/rtc_io_ll.h#L190-L195

IhorNehrutsa, 2024-11-25:
May set SOC_GPIO_SUPPORT_RTC_INDEPENDENT to 1 to resolve the issue.

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/esp_driver_gpio/src/gpio.c#L37

closed 2024-12-20 by espressif-bot

safocl, 2024-12-20:
maybe this version will look and work better:
```
struct get_io_config_info {
    bool *pu;
    bool *pd;
    bool *ie;
    bool *oe;
    bool *od;
    uint32_t *drv;
    uint32_t *fun_sel;
    uint32_t *sig_out;
    bool *slp_sel;
};

static inline get_io_config_info gpio_ll_get_io_config(gpio_dev_t *hw, uint32_t gpio_num)
{
    uint32_t bit_shift = (gpio_num < 32) ? gpio_num : (gpio_num - 32);
    uint32_t bit_mask = 1 << bit_shift;
    uint32_t iomux_reg_val = REG_READ(GPIO_PIN_MUX_REG[gpio_num]);

    get_io_config_info ret;
    get_io_config_info.pu = (iomux_reg_val & FUN_PU_M) >> FUN_PU_S;
    get_io_config_info.pd = (iomux_reg_val & FUN_PD_M) >> FUN_PD_S;
    get_io_config_info.ie = (iomux_reg_val & FUN_IE_M) >> FUN_IE_S;
    get_io_config_info.oe = (((gpio_num < 32) ? hw->enable : hw->enable1.val) & bit_mask) >> bit_shift;
    get_io_config_info.od = hw->pin[gpio_num].pad_driver;
    get_io_config_info.drv = (iomux_reg_val & FUN_DRV_M) >> FUN_DRV_S;
    get_io_config_info.fun_sel = (iomux_reg_val & MCU_SEL_M) >> MCU_SEL_S;
    get_io_config_info.sig_out = hw->func_out_sel_cfg[gpio_num].func_sel;
    get_io_config_info.slp_sel = (iomux_reg_val & SLP_SEL_M) >> SLP_SEL_S;

    return ret;
}
```
or
```
    return { 
        .pu = (iomux_reg_val & FUN_PU_M) >> FUN_PU_S,
        .pd = (iomux_reg_val & FUN_PD_M) >> FUN_PD_S,
        .ie = (iomux_reg_val & FUN_IE_M) >> FUN_IE_S,
        .oe = (((gpio_num < 32) ? hw->enable : hw->enable1.val) & bit_mask) >> bit_shift,
        .od = hw->pin[gpio_num].pad_driver,
        .drv = (iomux_reg_val & FUN_DRV_M) >> FUN_DRV_S,
        .fun_sel = (iomux_reg_val & MCU_SEL_M) >> MCU_SEL_S,
        .sig_out = hw->func_out_sel_cfg[gpio_num].func_sel,
        .slp_sel = (iomux_reg_val & SLP_SEL_M) >> SLP_SEL_S
        };
```
?

This gives some advantages -- at least you won't have to check for nullptr pointers.

IhorNehrutsa, 2024-12-20:
@safocl @songruo 
The main problem is that when we get the PULLUP/PULLDOWN status, we check the IO_MUX_x_REG in the GPIO functions
```
 *pu = (iomux_reg_val & FUN_PU_M) >> FUN_PU_S; 
 *pd = (iomux_reg_val & FUN_PD_M) >> FUN_PD_S; 
```
but to set PULLUP/PULLDOWN we use RTC function
```
rtc_gpio_pullup_en(gpio_num); 
rtc_gpio_pullup_dis(gpio_num); 
```
not the GPIO function
```
gpio_hal_pullup_en(gpio_context.gpio_hal, gpio_num); 
gpio_hal_pullup_dis(gpio_context.gpio_hal, gpio_num); 
```

Fast solution in https://github.com/espressif/esp-idf/issues/14931#issuecomment-2498244386

IhorNehrutsa, 2024-12-20:
#14923 doesn't solve this issue/bug.

IhorNehrutsa, 2024-12-20:
Do I must to reopen this issue?

songruo, 2024-12-20:
@IhorNehrutsa The fix is here: https://github.com/espressif/esp-idf/blob/c749ec66f64527e09f2f1d50919914deca085e5c/components/esp_driver_gpio/src/gpio.c#L1026-L1039
Are you sure this doesn't fix the issue?

safocl, 2024-12-20:
@IhorNehrutsa 
I'm talking about the general case with this function, the form to which it was brought looks with a lot of overhead in my opinion...
but you can avoid such things by simply outputting the data in return

IhorNehrutsa, 2024-12-20:
@songruo 
Does [c749ec6](https://github.com/espressif/esp-idf/commit/c749ec66f64527e09f2f1d50919914deca085e5c) apply to other releases?

songruo, 2024-12-20:
Will backport to previous releases.

AxelLin, 2025-05-14:
@songruo 

It looks like 8fd103928b5eda6d455abb0f296071d34f5c414f causes regression
```
due to below changes:
--- a/components/driver/gpio/gpio.c
+++ b/components/driver/gpio/gpio.c
@@ -77,7 +77,7 @@ static gpio_context_t gpio_context = {

 esp_err_t gpio_pullup_en(gpio_num_t gpio_num)
 {
-    GPIO_CHECK(GPIO_IS_VALID_GPIO(gpio_num), "GPIO number error", ESP_ERR_INVALID_ARG);
+    GPIO_CHECK(GPIO_IS_VALID_OUTPUT_GPIO(gpio_num), "GPIO number error", ESP_ERR_INVALID_ARG);

     if (!rtc_gpio_is_valid_gpio(gpio_num) || SOC_GPIO_SUPPORT_RTC_INDEPENDENT) {
         portENTER_CRITICAL(&gpio_context.gpio_spinlock);
@@ -115,7 +115,7 @@ esp_err_t gpio_pullup_dis(gpio_num_t gpio_num)

 esp_err_t gpio_pulldown_en(gpio_num_t gpio_num)
 {
-    GPIO_CHECK(GPIO_IS_VALID_GPIO(gpio_num), "GPIO number error", ESP_ERR_INVALID_ARG);
+    GPIO_CHECK(GPIO_IS_VALID_OUTPUT_GPIO(gpio_num), "GPIO number error", ESP_ERR_INVALID_ARG);

```
I use ESP32 GPIO34 as input.
My button is low-active, and I have gpio_conf.pull_up_en = GPIO_PULLUP_ENABLE;.
But with above changes, I got "GPIO number error".
I'm using esp-idf v5.2 branch, but seems this issue also happens on other newer branches.
Can you fix this ASAP?

songruo, 2025-05-14:
@AxelLin This can be considered as a bugfix. ESP32 GPIO34 is an input-only pin, and for these input-only pins, there are no internal pull-up / pull-down resistors inside the pad. Therefore, the previous call to `gpio_pulldown_en` or `gpio_pullup_en` to set the pu, pd register bits indeed do no effect to the input-only IOs.

AxelLin, 2025-05-14:
@songruo Thanks for the prompt reply.