espressif/esp-idf #14933: How to configure correctly for bridging in eth2ap and sta2eth projects?

02-bmd-44, 2024-11-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I was able to run the [eth2ap](https://github.com/espressif/esp-idf/tree/master/examples/network/eth2ap) and [sta2eth ](https://github.com/espressif/esp-idf/tree/master/examples/network/sta2eth)projects separately without any problems (as in 1 and 2 in the image). But what I want is to be able to use these two projects together (as in image 3). I don't know exactly what configurations I need to do for this.

[image]

### Modules I Use
2X ESP32 WROOM 32U
2X LAN8720

If you can help me with this, you will save me a great deal of trouble.

bogdankolendovskyy, 2024-11-27:
I've tried and was unable to reproduce the issue. Could you elaborate on your configuration for each device ?

02-bmd-44, 2024-11-28:
I used esp-idf version 5.3.1. I made a few changes to use lan8720 module in et2ap according to lan8720 module and in sta2eth to use lan8720 module instead of usb only. I got some of them by trial and error. Honestly, I didn't have much experience in esp-idf and networking. If there is no error in your system, can you send me the codes you edited. Or can you indicate where I need to change in the first use of the sample project.

02-bmd-44, 2024-11-28:
I finally managed to get a connection with some changes. I have done speed tests and I can get enough internet speed. I also did a ping test by playing a game. The problem is that sometimes when I am in the game I get the following message. The internet cuts out for 2-3 seconds and then it starts working again. Why exactly is this happening and how can I solve it.

```
E (8938636) eth2ap_example: send flow control message failed or timeout
E (8938736) eth2ap_example: send flow control message failed or timeout
E (8938836) eth2ap_example: send flow control message failed or timeout
E (8938936) eth2ap_example: send flow control message failed or timeout
E (8939036) eth2ap_example: send flow control message failed or timeout
E (8939066) eth2ap_example: WiFi send packet failed: 12309
E (9091736) eth2ap_example: WiFi send packet failed: 12309
E (9091906) eth2ap_example: send flow control message failed or timeout
E (9092006) eth2ap_example: send flow control message failed or timeout
E (9092106) eth2ap_example: send flow control message failed or timeout 
```

bogdankolendovskyy, 2024-11-29:
I didn't edit anything in the example, the only changes I made were following in the project configuration:

**eth2ap**
- SSID of the WiFi access point created by ESP32 `Example Configuration ---> Wi-Fi SSID`
- Password for said access point `Example Configuration ---> Wi-Fi Password`
- Ethernet configuration (PHY Device, MDC/MDIO GPIO number, clock source according to the board you use) in `Example Ethernet Configuration`

**sta2eth**
- Set it to be configured via web interface `Example Configuration ---> WiFi configuration ---> Manual configuration via http server`
- Ethernet configuration (PHY Device, MDC/MDIO GPIO number, clock source according to the board you use) in `Example Ethernet Configuration`

Other settings were left default. The devices were then connected as described below
```
PC 1 -- eth2ap -- sta2eth -- PC 2
```

iperf server was ran on PC 2 and iperf client on PC 1 to check the connection.

02-bmd-44, 2024-12-13:
It took me a while to get back as I was busy with the project. I downloaded the code again and made only the basic changes you mentioned and it worked fine. I think I made too many changes with the configurations before, thank you for your help.

closed 2024-12-13 by 02-bmd-44