espressif/esp-idf #14938: assert lld_con.c 2387, param 00000000 00000000

thekurtovic, 2024-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4.241008

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Print heap statistics without any issue.

### What is the actual behavior?

Occasionally will crash with the message assert lld_con.c 2387, param 00000000 00000000

### Steps to reproduce.

Not reproducible. Occurs in a project which uses BLE to connection to peripheral devices.
Will try to create minimal reproducible example.

### Debug Logs.

```plain
lld_con.c 2387, param 00000000 00000000
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40006f93  PS      : 0x00060b34  A0      : 0x8001b520  A1      : 0x3fca3030  
A2      : 0x00000000  A3      : 0x3ff19e72  A4      : 0x00000953  A5      : 0x0000071a  
A6      : 0x07fffffe  A7      : 0x339d2652  A8      : 0x00000001  A9      : 0x3fcef9e4  
A10     : 0x3fcef9e4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40006f8c  LEND    : 0x40006f93  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x40388f9f  EPC2    : 0x00000000  EPC3    : 0x400559e0  EPC4    : 0x40006f93

Backtrace: 0x40006f90:0x3fca3030 0x4001b51d:0x3fca3050 0x4037c571:0x3fca3070 0x4002ce21:0x3fca3090 0x4002baf5:0x3fca30b0 0x4002c436:0x3fca30d0 0x4037e041:0x3fca30f0 0x4000ce63:0x3fca3120 0x4037b643:0x3fca3140 0x40379485:0x3fca3160 0x400559dd:0x3fccf6e0 0x40386fca:0x3fccf6f0 0x403890a3:0x3fccf710 0x4211a80f:0x3fccf730 0x4211a865:0x3fccf750 0x4204cf5e:0x3fccf770 0x4204cfa9:0x3fccf7b0 0x4201a2b7:0x3fccf7f0 0x42005ed3:0x3fccf810 0x42010cd7:0x3fccf830 0x4210f0b6:0x3fccf890 0x4202cfb9:0x3fccf8b0 0x4202d2f7:0x3fccf8e0 0x420e6355:0x3fccf950 0x420e6669:0x3fccf970 0x420e8062:0x3fccf9a0 0x420f00ab:0x3fccfa00 0x420e2da9:0x3fccfa30 0x420e2c3b:0x3fccfa60 0x420e2065:0x3fccfa90 0x420e2077:0x3fccfab0 0x42115a1f:0x3fccfad0 0x403800d6:0x3fccfaf0 0x4202db4a:0x3fccfb10 0x40386d55:0x3fccfb30

Core  1 register dump:
PC      : 0x4037587d  PS      : 0x00060034  A0      : 0x8038058c  A1      : 0x3fca3900  
A2      : 0x00ca7f40  A3      : 0x00000000  A4      : 0x3fcaedce  A5      : 0x01ffffff  
A6      : 0x00060523  A7      : 0x00000000  A8      : 0x8037587d  A9      : 0x3fca38e0  
A10     : 0xac022001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00060023  
[... 136 lines cut ...]
Normally the code works as expected, but sometimes it seemingly crashes when trying to print heap values at the end of the NimBLE `onNotify` callback.
```
void helper::getMemInfo(u8_t type, mem_t &h)
{
    h.free = (type == MEM_INT) ? ESP.getFreeHeap() : ESP.getFreePsram();
    h.max = (type == MEM_INT) ? ESP.getMaxAllocHeap() : ESP.getMaxAllocPsram();
    h.frag = h.free > 0
            ? (100 * (h.free - h.max)) / h.free
            : 100;
}

#define PRINT_HEAP() { mem_t i, e;                                              \
                    helper::getMemInfo(MEM_INT, i);                             \
                    helper::getMemInfo(MEM_EXT, e);                             \
                    LOG_D("Mem %lu B (-%lu, %u%%), %lu B (-%lu, %u%%)",         \
                                            i.free, i.free - i.max, i.frag,     \
                                            e.free, e.free - e.max, e.frag); }

void ble::onNotify(const NimBLERemoteCharacteristic *ch, const u8_t *data, size_t len, bool isNotify)
{
    ...
    PRINT_HEAP();
}

What does the assert message mean?

esp-zhp, 2024-11-28:
Thank you for report the issue.I will check it.

thekurtovic, 2025-01-03:
I've migrated to IDF version 5.3.2.241210.
I'm still getting the issue occasionally, but not as frequent anymore, sometimes running for 3 days straight.
The nimble host task runs on CPU0. Instead of doing everything within the `onNotify` callback, I changed it to only copy the data into a buffer, and then process the buffered data during the main loop on CPU1.
Sometimes it will assert within the callback, but other times after when CPU1 is processing the data.
The backtraces don't contained as much information anymore, but still show a similar pattern.
I'm assuming the assertion is regarding a nullptr since it's printing 00000000, but not sure what it could be.

<details>
<summary>
Debug
</summary>

```
D [251624414] >>onNotify
D [251624414] Pushing notification from d01ec7d8b54f
D [251624420] >>push
D [251624424] <<push
D [251624429] <
BLE assert lld_con.c 2387, param 00000000 00000000
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40006f93  PS      : 0x00060a34  A0      : 0x8001b520  A1      : 0x3fca59f0
A2      : 0x00000000  A3      : 0x3ff19e72  A4      : 0x00000953  A5      : 0x40385e3e
A6      : 0x3fca68e8  A7      : 0xffffffff  A8      : 0x00000001  A9      : 0x3fcef9e4
A10     : 0x3fcef9e4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40006f8c  LEND    : 0x40006f93  LCOUNT  : 0x00000000
Core  0 was running in ISR context:
EPC1    : 0x4038ad23  EPC2    : 0x00000000  EPC3    : 0x400559e0  EPC4    : 0x40006f93

Backtrace: 0x40006f90:0x3fca59f0 0x4001b51d:0x3fca5a10 0x4037edd9:0x3fca5a30 0x4002ce21:0x3fca5a50 0x4002baf5:0x3fca5a70 0x4002c436:0x3fca5a90 0x403801f9:0x3fca5ab0 0x4000ce63:0x3fca5ae0 0x4037d413:0x3fca5b00 0x4037ca39:0x3fca5b20 0x400559dd:0x3fcc9be0 0x40385dbb:0x3fcc9bf0 0x4037c393:0x3fcc9c10 0x4002cc57:0x3fcc9c30 0x4037e51b:0x3fcc9c70 0x4001c246:0x3fcc9ca0 0x420e6089:0x3fcc9d20 0x400131c5:0x3fcc9d50 0x420e53c2:0x3fcc9db0 0x40029007:0x3fcc9dd0 0x420f1357:0x3fcc9e50 0x4000d025:0x3fcc9e80 0x4002c4a5:0x3fcc9ea0 0x420e2573:0x3fcc9ec0 0x4037cf99:0x3fcc9ee0 0x4037d167:0x3fcc9f00 0x40385a39:0x3fcc9f30

Core  1 register dump:
PC      : 0x40056967  PS      : 0x00060034  A0      : 0x82014668  A1      : 0x3fca6280
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x40382f68  A8      : 0x0000fa00  A9      : 0x00000eff
A10     : 0xfa000000  A11     : 0x70bc4dc0  A12     : 0x4dc00000  A13     : 0x70bc0000
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x0000000a  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40056964:0x3fca6280 0x42014665:0x3fca62a0 0x40375fde:0x3fca62c0 0x40380d3d:0x3fca62e0 0x40380f98:0x3fca6300 0x4037729d:0x3fca6330 0x4037c8ed:0x3fca6350 0x403811cf:0x3fcb6920 0x4203ffbb:0x3fcb6940 0x40386d29:0x3fcb6960 0x40385a39:0x3fcb6980

ELF file SHA256: b38a484b69ad599f882ca68a76ee5a116f10efbb917a9e977a41b32c770a886f

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4039294f  PS      : 0x00020f34  A0      : 0x82045574  A1      : 0x3fca5840
A2      : 0x00020f23  A3      : 0xa5a5a5a5  A4      : 0x82045574  A5      : 0x00060f25
A6      : 0xfffbfff0  A7      : 0x00000076  A8      : 0x00000000  A9      : 0x3fca5810
A10     : 0x00000001  A11     : 0x3fca581c  A12     : 0x00000001  A13     : 0x3fca582c
A14     : 0x00000059  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
Core  0 was running in ISR context:
EPC1    : 0x4038ad23  EPC2    : 0x00000000  EPC3    : 0x400559e0  EPC4    : 0x4039294f

Backtrace: 0x4039294c:0x3fca5840 0x42045571:0x3fca5850 0x42045653:0x3fca5870 0x4204408e:0x3fca5890 0x420431b5:0x3fca58c0 0x40378065:0x3fca5910 0x4037841c:0x3fca5930 0x00040022:0x3fca59f0 |<-CORRUPTED

Core  1 register dump:
PC      : 0x40056967  PS      : 0x00060034  A0      : 0x82014668  A1      : 0x3fca6280
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x40382f68  A8      : 0x0000fa00  A9      : 0x00000eff
A10     : 0xfa000000  A11     : 0x70bc4dc0  A12     : 0x4dc00000  A13     : 0x70bc0000
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x0000000a  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40056964:0x3fca6280 0x42014665:0x3fca62a0 0x40375fde:0x3fca62c0 0x40380d3d:0x3fca62e0 0x40380f98:0x3fca6300 0x4037729d:0x3fca6330 0x4037c8ed:0x3fca6350 0x403811cf:0x3fcb6920 0x4203ffbb:0x3fcb6940 0x40386d29:0x3fcb6960 0x40385a39:0x3fcb6980

ELF file SHA256: b38a484b69ad599f882ca68a76ee5a116f10efbb917a9e977a41b32c770a886f
[... 41 lines cut ...]

</details>

<details>
<summary>Code</summary>

```
void ble::onNotify(const NimBLERemoteCharacteristic *ch, const u8_t *data, size_t len, bool isNotify)
{
    LOG_D(">>onNotify");
    bleNotify_t notify{};
    
    if (len > BLE_MAX_PACKET || len < BLE_MIN_PACKET)
        return;

    LOG_D("Pushing notification from %s", notify.addr);
    memcpy(notify.data, data, len);
    notify.len = len;
    notify.notify = isNotify;
    _notifs.push(notify);
    LOG_D("<<onNotify");
}
```

</details>

thekurtovic, 2025-01-06:
This assert seems to eventually occur when I have the esp32-s3 operating in multi-connect mode, where it will restart scanning after a connection is made.
If instead it only handles one connection at a time, and waits until disconnect before restarting scan, the issue doesn't happen.

I have 4 peripheral devices which are advertising. If the esp32-s3 connects to a peripheral and sends data, the peripheral waits 4 seconds before advertising again. When the esp32-s3 receives the advertisements it will attempt to connect asynchronously from the `onResult` callback upon receiving an advertisement.

esp-cjh, 2025-01-07:
Hi, @thekurtovic How long does it take to reproduce this issue?

thekurtovic, 2025-01-07:
Looking through some of my logs it can range from a few minutes to several days, appears random to me. 
<details>
<summary>Logs</summary>

```
3.3 hr
3.7 hr
4.6 hr
3.9 hr
4.7 hr
6.1 hr
8.3 hr
4.8 hr
12  hr
7.9 hr
25  hr
2.9 hr
1.8 hr
70  hr
8.5 min
34  min
3.4 hr
3.5 hr
```

</details>

I think increasing the rate of the "connect/scan/write to characteristic/receive notification" cycle helps trigger it faster. With 4 peripherals my esp32-s3 starts that cycle every ~1.5 seconds, and finishes in about half that time.
I'll try setting up a minimal example project this week.

esp-cjh, 2025-01-08:
@thekurtovic Thank you very much. I will also provide the debug library on my end.

thekurtovic, 2025-01-08:
This is as barebones as I could get it.
https://github.com/thekurtovic/esp32-nimble-assert

esp-cjh, 2025-01-09:
@thekurtovic Could you provide the code for the peripheral side?

thekurtovic, 2025-01-09:
I'm not using any espressif chips for peripheral devices. I will try to rewrite the peripheral application for esp32 and see if I can reproduce again.

thekurtovic, 2025-01-09:
nRF52840

esp-cjh, 2025-01-10:
Please upload both the ESP32 and NRF5340 peripheral example, and I will provide a controller debug library for you.

esp-cjh, 2025-01-10:
@thekurtovic Please use this lib [266551d.zip](https://github.com/user-attachments/files/18372557/266551d.zip) to reproduce the issue and provide the log.
1 Uzip the file and replace the controller lib (components/bt/controller/lib_esp32c3_family/esp32s3/libbtdm_app.a)
2 Set the CONFIG_ESP_INT_WDT_TIMEOUT_MS=5000

thekurtovic, 2025-01-10:
Looks like no issues after running for 18 hours when connecting to esp32 peripherals.
I will need some time to create a minimal nRF52840 project before I can share the code.

Thanks. I've setup the original test again, replaced the file, and added the config to `sdkconfig.defaults`.
Do I need a certain log level to be enabled? I'm currently set to `CONFIG_LOG_DEFAULT_LEVEL_WARN`

esp-cjh, 2025-01-13:
@thekurtovic I'm very sorry, the lib provided last time had an issue.
Please use this one [e178c96.zip](https://github.com/user-attachments/files/18392288/e178c96.zip) to reproduce the problem.

thekurtovic, 2025-01-13:
No worries. Here's output with the new file.

<details>
<summary>Log 1</summary>

```
(2275404) main.cpp:215 Connecting to e3:e4:5b:75:ad:36
(2275410) main.cpp:225 <<onResult
(2275626) main.cpp:289 Sending data to d1:bb:4d:b5:30:72...
(2275666) main.cpp:298 <<send
(2275667) main.cpp:239 >>onNotify
(2275667) main.cpp:242 Notif from d1:bb:4d:b5:30:72, 36 B
c1 e8 14 0a 00 00 1c 00 00 09 ff f3 01 09 30 00 03 09 2a 00 07 07 a0 04 03 08 01 00 00 10 ff 00 01 10 e2 01
(2275676) main.cpp:244 free heap h 249907 163828
(2275680) main.cpp:245 free heap p 8365252 8257524
(2275685) main.cpp:246 <<onNotify
BLE assert lld_con.c 2387, param 00000000 00000000
=== BLE DUMP START(4ae0) ===
ET[0] 501a (7278545 624) 10000
ET[1] 801a (7278600 624) 2550
ET[2] 501a (7278609 624) 10000
ET[3] 701a (7278664 624) 2550
ET[4] 501a (7278673 624) 10000
ET[5] 801a (7278728 624) 2550
ET[6] 501a (7278737 624) 10000
ET[7] 701a (7278792 624) 2550
ET[8] 501a (7278801 624) 10000
ET[9] 801a (7278856 624) 2550
ET[10] 501a (7278865 624) 10000
ET[11] 701a (7278920 624) 2550
ET[12] 501a (7278929 624) 10000
ET[13] 801a (7278472 624) 2550
ET[14] 501a (7278481 624) 10000
ET[15] 701a (7278536 624) 2550
REG 0010070f 00640b66 00000800
DIAG[0]: 00000000
DIAG[1]: 0811f960
LLM[0] 6 (0 0 0 0)
LLM[1] 9 (4 32 2 2)
LLM[2] 5 (4 32 2 2)
CON[1 0x3fcb6d38] 2 (38 1) (7276896 5100 64)
CON[2 0x3fcb6c48] 0 (0 0) (7278984 1516 24)
SCH_WAIT[0 0x3fcb6c48] 31 (7278984 0) 1516
SCH_PLAN[0] 1 4 (2 2)
SCH_PROG 0 13 13
SLICE[0] 0 0
SLICE[1] 64 5100
SLICE[2] 24 1516
SLICE[3] 0 5100
SLICE[4] 0 0
SLICE[5] 0 0
SLICE[6] 0 0
SLICE[7] 0 0
SLICE[8] 0 0
SLICE[9] 0 0
=== BLE DUMP END ===
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40006f93  PS      : 0x00060634  A0      : 0x8001b520  A1      : 0x3fc9cc90
A2      : 0x00000000  A3      : 0x3ff19e72  A4      : 0x00000953  A5      : 0x40382c92
A6      : 0x3fc9d998  A7      : 0xffffffff  A8      : 0x00000001  A9      : 0x3fcef9e4
A10     : 0x3fcef9e4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40006f8c  LEND    : 0x40006f93  LCOUNT  : 0x00000000
Core  0 was running in ISR context:
EPC1    : 0x4038727f  EPC2    : 0x00000000  EPC3    : 0x400559e0  EPC4    : 0x40006f93

Backtrace: 0x40006f90:0x3fc9cc90 0x4001b51d:0x3fc9ccb0 0x4037b891:0x3fc9ccd0 0x4002ce21:0x3fc9ccf0 0x4002baf5:0x3fc9cd10 0x4002c436:0x3fc9cd30 0x4037ccb9:0x3fc9cd50 0x4000ce63:0x3fc9cd80 0x40379ecb:0x3fc9cda0 0x40378f19:0x3fc9cdc0 0x400559dd:0x3fcba3c0 0x40382bc3:0x3fcba3d0 0x40379123:0x3fcba3f0 0x4002cc57:0x3fcba410 0x4037afd3:0x3fcba450 0x4001c246:0x3fcba480 0x42040db9:0x3fcba500 0x400131c5:0x3fcba530 0x4204007e:0x3fcba590 0x40029007:0x3fcba5b0 0x4204c763:0x3fcba630 0x4000d025:0x3fcba660 0x4002c4a5:0x3fcba680 0x4203d1db:0x3fcba6a0 0x40379a4d:0x3fcba6c0 0x40379c1f:0x3fcba6e0 0x40382841:0x3fcba710

[... 42 lines cut ...]
r_rwble_isr at ??:?
?? ??:0
r_rwbtdm_isr_wrapper at ??:?
_xt_medint3 at C:\Users\User\.platformio\packages\framework-espidf\components\xtensa/xtensa_vectors.S:1409
?? ??:0
vPortClearInterruptMaskFromISR at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\include\freertos/portmacro.h:560
 (inlined by) vPortExitCritical at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:514
global_interrupt_restore at C:/Users/User/.platformio/packages/framework-espidf/components/bt/controller/esp32c3/bt.c:546
?? ??:0
r_lld_con_sched at ??:?
?? ??:0
r_lld_con_start_eco at ??:?
?? ??:0
r_llc_start_eco at ??:?
?? ??:0
r_ke_task_schedule at ??:?
?? ??:0
?? ??:0
rw_schedule at ??:?
ble_try_turn_on_pll_track at ??:?
btdm_controller_task at ??:?
vPortTaskWrapper at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:139
```

</details>

esp-cjh, 2025-01-14:
@thekurtovic Please use this lib [f965908.zip](https://github.com/user-attachments/files/18405086/f965908.zip) to test whether the issue can be fixed.
Could you provide an example for the nRF52840? It would be faster for me to resolve this issue if I can reproduce it locally.

thekurtovic, 2025-01-14:
Issue still reproduced.
Yes I will try to do it soon.

esp-cjh, 2025-01-14:
Thanks. Please provide the new log.

thekurtovic, 2025-01-14:
<details>
<summary>Log</summary>

```
CANCEL[1] 2 0x3fcb6d58 (563736 5100)
BLE assert lld_con.c 2387, param 00000000 00000000
=== BLE DUMP START(4ee0) ===
ET[0] 501a (563297 624) 10000 287
ET[1] 801a (563352 624) 2550 22d
ET[2] 501a (563361 624) 10000 287
ET[3] 801a (563416 624) 2550 25a
ET[4] 501a (563425 624) 10000 287
ET[5] 781a (563480 624) 2550 22d
ET[6] 501a (563489 624) 10000 287
ET[7] 801a (563544 624) 2550 22d
ET[8] 501a (563553 624) 10000 287
ET[9] 801a (563608 624) 2550 25a
ET[10] 501a (563617 624) 10000 287
ET[11] 781a (563672 624) 2550 22d
ET[12] 501a (563681 624) 10000 287
ET[13] 801a (563224 624) 2550 25a
ET[14] 501a (563233 624) 10000 287
ET[15] 781a (563288 624) 2550 22d
REG 0010070f 00640b66 00000800
DIAG[0]: 00000000
DIAG[1]: 009aa302
LLM[0] 6 (0 0 0 0)
LLM[1] 9 (12 32 2 2)
LLM[2] 9 (12 32 2 2)
LLM[3] 5 (12 32 2 2)
CON[1 0x3fcb6d58] 2 22d (51 1) (560832 5100 64)
CON[2 0x3fcb6c68] 0 25a (170 3) (553656 5100 64)
CON[3 0x3fcb6b78] 0 287 (0 0) (563736 2500 24)
SCH_WAIT[0 0x3fcb6b78] 31 (563736 0) 2500
SCH_WAIT[1 0x3fcb6c68] 16 (563800 0) 5100
SCH_PLAN[0] 2 12 (2 2)
SCH_PLAN[1] 1 12 (2 2)
SCH_PROG 0 13 13
SLICE[0] 0 0
SLICE[1] 64 5100
SLICE[2] 64 5100
SLICE[3] 24 2500
SLICE[4] 0 0
SLICE[5] 0 0
SLICE[6] 0 0
SLICE[7] 0 0
SLICE[8] 0 0
SLICE[9] 0 0
=== BLE DUMP END ===
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40006f93  PS      : 0x00060634  A0      : 0x8001b520  A1      : 0x3fc9cc90  
A2      : 0x00000000  A3      : 0x3ff19e72  A4      : 0x00000953  A5      : 0x3fc9cc04  
A6      : 0x00000018  A7      : 0xfffffffb  A8      : 0x00000001  A9      : 0x3fcef9e4  
A10     : 0x3fcef9e4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x3fc9cbb4  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40006f8c  LEND    : 0x40006f93  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x403872a3  EPC2    : 0x00000000  EPC3    : 0x400559e0  EPC4    : 0x40006f93

Backtrace: 0x40006f90:0x3fc9cc90 0x4001b51d:0x3fc9ccb0 0x4037b8b6:0x3fc9ccd0 0x4002ce21:0x3fc9ccf0 0x4002baf5:0x3fc9cd10 0x4002c436:0x3fc9cd30 0x4037ccdd:0x3fc9cd50 0x4000ce63:0x3fc9cd80 0x40379ecb:0x3fc9cda0 0x403791cd:0x3fc9cdc0 0x400559dd:0x3fcba3e0 0x40382be7:0x3fcba3f0 0x40378b27:0x3fcba410 0x4002cc57:0x3fcba430 0x4037afd3:0x3fcba470 0x4001c246:0x3fcba4a0 0x42040dc5:0x3fcba520 0x400131c5:0x3fcba550 0x42040076:0x3fcba5b0 0x40029007:0x3fcba5d0 0x4204c7b3:0x3fcba650 0x4000d025:0x3fcba680 0x4002c4a5:0x3fcba6a0 0x4203d1d3:0x3fcba6c0 0x40379a4d:0x3fcba6e0 0x40379c1f:0x3fcba700 0x40382865:0x3fcba730

Core  1 register dump:
PC      : 0x4037f096  PS      : 0x00060734  A0      : 0x8201119e  A1      : 0x3fcafc80  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80382bea  A5      : 0x3fcbfcb0  
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x8201198e  A9      : 0x3fcafc40  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80382bea  A13     : 0x3fcbfe30  
A14     : 0x00060023  A15     : 0x3fcafe60  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[... 35 lines cut ...]
r_rwble_isr at ??:?
?? ??:0
r_rwbtdm_isr_wrapper at ??:?
_xt_medint3 at C:\Users\User\.platformio\packages\framework-espidf\components\xtensa/xtensa_vectors.S:1409
?? ??:0
vPortClearInterruptMaskFromISR at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\include\freertos/portmacro.h:560
 (inlined by) vPortExitCritical at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:514
global_interrupt_restore at C:/Users/User/.platformio/packages/framework-espidf/components/bt/controller/esp32c3/bt.c:546
?? ??:0
r_lld_con_sched at ??:?
?? ??:0
r_lld_con_start_eco at ??:?
?? ??:0
r_llc_start_eco at ??:?
?? ??:0
r_ke_task_schedule at ??:?
?? ??:0
?? ??:0
rw_schedule at ??:?
ble_try_turn_on_pll_track at ??:?
btdm_controller_task at ??:?
vPortTaskWrapper at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:139
```

</details>

esp-cjh, 2025-01-17:
Hi, @thekurtovic 
Please use this lib [ffba7b6.zip](https://github.com/user-attachments/files/18454510/ffba7b6.zip) to check whether the issue can be fixed. Notice to saved the log.

thekurtovic, 2025-01-17:
@esp-cjh Did you manage to reproduce the issue on your end?

<details>
<summary>Log</summary>

```
CANCEL[1] 2 0x3fcb6de8 (5144880 5100)
BLE assert lld_con.c 2387, param 00000000 00000000
=== BLE DUMP START(4ee0) ===
ET[0] 501a (5144761 624) 10000 25a
ET[1] 701a (5144816 624) 2550 22d
ET[2] 501a (5144825 624) 10000 25a
ET[3] 801a (5144368 624) 2550 22d
ET[4] 501a (5144377 624) 10000 25a
ET[5] 701a (5144432 624) 2550 22d
ET[6] 501a (5144441 624) 10000 25a
ET[7] 801a (5144496 624) 2550 22d
ET[8] 501a (5144505 624) 10000 25a
ET[9] 701a (5144560 624) 2550 22d
ET[10] 501a (5144569 624) 10000 25a
ET[11] 801a (5144624 624) 2550 22d
ET[12] 501a (5144633 624) 10000 25a
ET[13] 701a (5144688 624) 2550 22d
ET[14] 501a (5144697 624) 10000 25a
ET[15] 801a (5144752 624) 2550 22d
REG 0010070f 00640b66 00000800
DIAG[0]: 00000000
DIAG[1]: 0081c50b
LLM[0] 6 (0 0 0 0)
LLM[1] 9 (24 32 2 2)
LLM[2] 5 (40 120 2 2)
CON[1 0x3fcb6de8] 2 22d (37 1) (5142840 5100 64)
CON[2 0x3fcb6cf8] 0 25a (0 0) (5144880 2500 24)
SCH_WAIT[0 0x3fcb6cf8] 31 (5144880 0) 2500
SCH_PLAN[0] 1 24 (2 2)
SCH_PROG 0 3 3
SLICE[0] 0 0
SLICE[1] 64 5100
SLICE[2] 24 2500
SLICE[3] 0 5100
SLICE[4] 0 0
SLICE[5] 0 0
SLICE[6] 0 0
SLICE[7] 0 0
SLICE[8] 0 0
SLICE[9] 0 0
=== BLE DUMP END ===
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40006f93  PS      : 0x00060634  A0      : 0x8001b520  A1      : 0x3fc9cc90
A2      : 0x00000000  A3      : 0x3ff19e72  A4      : 0x00000953  A5      : 0x3fc9cc04
A6      : 0x00000018  A7      : 0xfffffffb  A8      : 0x00000001  A9      : 0x3fcef9e4
A10     : 0x3fcef9e4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x3fc9cbb4  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40006f8c  LEND    : 0x40006f93  LCOUNT  : 0x00000000
Core  0 was running in ISR context:
EPC1    : 0x403872a3  EPC2    : 0x00000000  EPC3    : 0x400559e0  EPC4    : 0x40006f93

Backtrace: 0x40006f90:0x3fc9cc90 0x4001b51d:0x3fc9ccb0 0x4037b8b6:0x3fc9ccd0 0x4002ce21:0x3fc9ccf0 0x4002baf5:0x3fc9cd10 0x4002c436:0x3fc9cd30 0x4037ccdd:0x3fc9cd50 0x4000ce63:0x3fc9cd80 0x40379ecb:0x3fc9cda0 0x403791cd:0x3fc9cdc0 0x400559dd:0x3fcba470 0x40382be7:0x3fcba480 0x40378b27:0x3fcba4a0 0x4002cc57:0x3fcba4c0 0x4037afd3:0x3fcba500 0x4001c246:0x3fcba530 0x42040ee5:0x3fcba5b0 0x400131c5:0x3fcba5e0 0x42040166:0x3fcba640 0x40029007:0x3fcba660 0x4204c8ef:0x3fcba6e0 0x4000d025:0x3fcba710 0x4002c4a5:0x3fcba730 0x4203d2c3:0x3fcba750 0x40379a4d:0x3fcba770 0x40379c1f:0x3fcba790 0x40382865:0x3fcba7c0

Core  1 register dump:
PC      : 0x4037f096  PS      : 0x00060734  A0      : 0x82011236  A1      : 0x3fcafc80
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80382bea  A5      : 0x3fcbfca0
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x82011a26  A9      : 0x3fcafc40
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80382bea  A13     : 0x3fcbfca0
A14     : 0x00060023  A15     : 0x3fcafe60  SAR     : 0x0000001f  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4037f093:0x3fcafc80 0x42011233:0x3fcafca0 0x40383851:0x3fcafcc0 0x40382865:0x3fcafce0

ELF file SHA256: cbee1f148f968064fbf73a4e2a6219619aa8d9395f0b2e293e25a6022d4c079e

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4038ae1f  PS      : 0x00020b34  A0      : 0x82013418  A1      : 0x3fc9cae0
A2      : 0x00020b23  A3      : 0xa5a5a5a5  A4      : 0x82013418  A5      : 0x00060b25
A6      : 0xfffbfff0  A7      : 0x00000076  A8      : 0x00000000  A9      : 0x3fc9cab0
A10     : 0x00000001  A11     : 0x3fc9cabc  A12     : 0x00000001  A13     : 0x3fc9cacc
A14     : 0x00000060  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
Core  0 was running in ISR context:
EPC1    : 0x403872a3  EPC2    : 0x00000000  EPC3    : 0x400559e0  EPC4    : 0x4038ae1f

Backtrace: 0x4038ae1c:0x3fc9cae0 0x42013415:0x3fc9caf0 0x420134f5:0x3fc9cb10 0x42012dee:0x3fc9cb30 0x42011f21:0x3fc9cb60 0x40376679:0x3fc9cbb0 0x40376a30:0x3fc9cbd0 0x00040022:0x3fc9cc90 |<-CORRUPTED

Core  1 register dump:
PC      : 0x4037f096  PS      : 0x00060734  A0      : 0x82011236  A1      : 0x3fcafc80
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80382bea  A5      : 0x3fcbfca0
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x82011a26  A9      : 0x3fcafc40
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80382bea  A13     : 0x3fcbfca0
A14     : 0x00060023  A15     : 0x3fcafe60  SAR     : 0x0000001f  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4037f093:0x3fcafc80 0x42011233:0x3fcafca0 0x40383851:0x3fcafcc0 0x40382865:0x3fcafce0

?? ??:0
?? ??:0
r_lld_con_evt_canceled_cbk_eco at ??:?
?? ??:0
?? ??:0
?? ??:0
r_rwble_isr at ??:?
?? ??:0
r_rwbtdm_isr_wrapper at ??:?
_xt_medint3 at C:\Users\User\.platformio\packages\framework-espidf\components\xtensa/xtensa_vectors.S:1409
?? ??:0
vPortClearInterruptMaskFromISR at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\include\freertos/portmacro.h:560
 (inlined by) vPortExitCritical at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:514
global_interrupt_restore at C:/Users/User/.platformio/packages/framework-espidf/components/bt/controller/esp32c3/bt.c:546
?? ??:0
r_lld_con_sched at ??:?
?? ??:0
r_lld_con_start_eco at ??:?
?? ??:0
r_llc_start_eco at ??:?
?? ??:0
r_ke_task_schedule at ??:?
?? ??:0
?? ??:0
rw_schedule at ??:?
ble_try_turn_on_pll_track at ??:?
btdm_controller_task at ??:?
vPortTaskWrapper at C:\Users\User\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:139
```

</details>

esp-cjh, 2025-01-20:
@thekurtovic I failed to flash the device. And could you please provide the complete log? I add some print in previous lib.

[image]

thekurtovic, 2025-01-20:
@esp-cjh Could you try opening a terminal in VSC and enter `make --version`? Seems like your system doesn't have `make`.
```
PS C:\nRF5_Projects\assert-repro> make --version
GNU Make 4.4.1
Built for Windows32
Copyright (C) 1988-2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

```
PS C:\nRF5_Projects\assert-repro> Get-Command make

CommandType     Name                                               Version    Source                                                                                                                                                                    
-----------     ----                                               -------    ------                                                                                                                                                                    
Application     make.exe                                           1.0.0.0    C:\ProgramData\chocolatey\bin\make.exe  
```

>And could you please provide the complete log? I add some print in previous lib.

I'll run the test again and get a complete log.

esp-cjh, 2025-01-20:
@thekurtovic Yes, I don't have `make` on my windows. May I should be missing the develop environment setup.

esp-cjh, 2025-01-20:
@thekurtovic Please use this lib [9f68669.zip](https://github.com/user-attachments/files/18475224/9f68669.zip) for test and provide the complete log. I have add a fixed and assert will no occur.

thekurtovic, 2025-01-20:
Thanks. I'll try testing again this week and report back by Friday.

thekurtovic, 2025-01-21:
@esp-cjh Here is [log](https://github.com/thekurtovic/esp32-nimble-assert/blob/main/log_esp32_2025-01-20_175724-bledump.ans)

thekurtovic, 2025-01-21:
@esp-cjh [Uploaded](https://github.com/thekurtovic/esp32-nimble-assert/blob/main/firmware.elf)

esp-cjh, 2025-01-21:
@thekurtovic I have found out the root cause. Please use this lib [d15948f.zip](https://github.com/user-attachments/files/18488533/d15948f.zip) to fixed the issue.

esp-cjh, 2025-01-22:
@thekurtovic The debug print is used for me to check. I will remove the debug print and give a fixed lib.

esp-cjh, 2025-01-22:
@thekurtovic Please use this lib [edff442.zip](https://github.com/user-attachments/files/18499515/edff442.zip) to fixed the issue.

thekurtovic, 2025-01-22:
@esp-cjh Oh sorry, I thought maybe it was still bad to be seeing.
Thanks for fixing it.

esp-cjh, 2025-01-22:
@thekurtovic  It's okay. Please test if the issue has been resolved. Thanks!

thekurtovic, 2025-01-22:
Tested for 20 hours so far, no issues observed. Previously would happen within 10-20 minutes of runtime.
I'll test for one more day just to make sure.

thekurtovic, 2025-01-23:
@esp-cjh Issue has been solved, thanks again.

Indastri, 2025-02-20:
@thekurtovic is the issue solved? If so please close this issue.

Cheers!

thekurtovic, 2025-02-20:
See my previous comment above.
A patched file was provided to test with, I don't know of any commits which actually address it.

esp-cjh, 2025-02-26:
@thekurtovic @Indastri 
I have submitted the MR, but it has not been merged yet.
Once it is merged, I will provide the GitHub commit.

closed 2025-03-28 by Alvin1Zhang