espressif/esp-idf #14939: on dependencies new version - ESP32-H2 HA_on_off_light does not connect to the Zigbee network.

KMTsvetanov, 2024-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDV v5.3.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It works if you don;t update to
  espressif/esp-zigbee-lib: "^1.6.0"
  espressif/esp-zboss-lib: "^1.6.0"

### What is the actual behavior?

It works only with:
  espressif/esp-zboss-lib: "1.0.9"
  espressif/esp-zigbee-lib: "1.0.9"

### Steps to reproduce.

idf.py update-dependencies
idf.py build

### More Information.

It breaks on line:
esp_zb_init(&zb_nwk_cfg);

Maybe it makes some changes inside:
start idf.py menuconfig

xieqinan, 2024-11-27:
Hi @KMTsvetanov ,

I tried using esp-idf v5.3.1 (commit: d479c520e1) to reproduce this issue, and it worked fine. My changes are detailed below; please follow them and test again:  

By the way, I recommend using the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) for developing your application, as it includes the latest updates and more examples.

```
diff --git a/examples/zigbee/light_sample/HA_on_off_light/main/esp_zb_light.h b/examples/zigbee/light_sample/HA_on_off_light/main/esp_zb_light.h
index 5def811a87f..a5405d8925d 100644
--- a/examples/zigbee/light_sample/HA_on_off_light/main/esp_zb_light.h
+++ b/examples/zigbee/light_sample/HA_on_off_light/main/esp_zb_light.h
@@ -34,10 +34,10 @@
 
 #define ESP_ZB_DEFAULT_RADIO_CONFIG()                           \
     {                                                           \
-        .radio_mode = RADIO_MODE_NATIVE,                        \
+        .radio_mode = ZB_RADIO_MODE_NATIVE,                        \
     }
 
 #define ESP_ZB_DEFAULT_HOST_CONFIG()                            \
     {                                                           \
-        .host_connection_mode = HOST_CONNECTION_MODE_NONE,      \
+        .host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE,      \
     }
diff --git a/examples/zigbee/light_sample/HA_on_off_light/main/idf_component.yml b/examples/zigbee/light_sample/HA_on_off_light/main/idf_component.yml
index 8578e73e678..31334ea1bb0 100644
--- a/examples/zigbee/light_sample/HA_on_off_light/main/idf_component.yml
+++ b/examples/zigbee/light_sample/HA_on_off_light/main/idf_component.yml
@@ -1,7 +1,7 @@
 ## IDF Component Manager Manifest File
 dependencies:
-  espressif/esp-zboss-lib: "1.0.9"
-  espressif/esp-zigbee-lib: "1.0.9"
+  espressif/esp-zboss-lib: "1.6.0"
+  espressif/esp-zigbee-lib: "1.6.0"
   espressif/led_strip: "~2.0.0"
   ## Required IDF version
   idf:
diff --git a/examples/zigbee/light_sample/HA_on_off_switch/main/esp_zb_switch.h b/examples/zigbee/light_sample/HA_on_off_switch/main/esp_zb_switch.h
index d4d95a0a0e4..08c233e2024 100644
--- a/examples/zigbee/light_sample/HA_on_off_switch/main/esp_zb_switch.h
+++ b/examples/zigbee/light_sample/HA_on_off_switch/main/esp_zb_switch.h
@@ -31,10 +31,10 @@
 
 #define ESP_ZB_DEFAULT_RADIO_CONFIG()                           \
     {                                                           \
-        .radio_mode = RADIO_MODE_NATIVE,                        \
+        .radio_mode = ZB_RADIO_MODE_NATIVE,                 \
     }
 
 #define ESP_ZB_DEFAULT_HOST_CONFIG()                            \
     {                                                           \
-        .host_connection_mode = HOST_CONNECTION_MODE_NONE,      \
+        .host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE, \
     }
diff --git a/examples/zigbee/light_sample/HA_on_off_switch/main/idf_component.yml b/examples/zigbee/light_sample/HA_on_off_switch/main/idf_component.yml
index c186c3c8cad..fc2c96c7de9 100644
--- a/examples/zigbee/light_sample/HA_on_off_switch/main/idf_component.yml
+++ b/examples/zigbee/light_sample/HA_on_off_switch/main/idf_component.yml
@@ -1,7 +1,7 @@
 ## IDF Component Manager Manifest File
 dependencies:
-  espressif/esp-zboss-lib: "1.0.9"
-  espressif/esp-zigbee-lib: "1.0.9"
+  espressif/esp-zboss-lib: "1.6.0"
+  espressif/esp-zigbee-lib: "1.6.0"
   ## Required IDF version
   idf:
     version: ">=5.0.0"
```

closed 2025-02-26 by chshu

xieqinan, 2025-07-16:
@Postafen ,

Could you please try again with the newest `~1.6.0` version ? If you MUST use the `~1.5.0`  version, please share the failed log.

xieqinan, 2025-07-21:
@Postafen ,

> the issue might not be the same, my problem is that the ~1.6.0 won't find any dependencies, causing

If the issue still confuses you, feel free to open a new issue for further discussion.