espressif/esp-idf #14940: Duplicate - not fixed - release/v4.4 install.sh venv error (IDFGH-10060) #11336

hubbardmeister, 2024-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4-dev-4375-gf420609c33-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Diff - for fix.

diff --git a/tools/idf_tools.py b/tools/idf_tools.py
index 5ae9f4cae2..2186a1604a 100755
--- a/tools/idf_tools.py
+++ b/tools/idf_tools.py
@@ -2581,7 +2581,7 @@ def action_install_python_env(args):  # type: ignore
     nix_store = os.environ.get('NIX_STORE')
     is_nix = nix_store is not None and sys.base_prefix.startswith(nix_store) and sys.prefix.startswith(nix_store)
 
-    is_virtualenv = not is_nix and (hasattr(sys, 'real_prefix') or (hasattr(sys, 'base_prefix') and sys.base_prefix != sys.prefix))
+    is_virtualenv = not is_nix and (hasattr(sys, 'real_prefix') and (hasattr(sys, 'base_prefix') and sys.base_prefix != sys.prefix))
     if is_virtualenv and (not os.path.exists(idf_python_env_path) or reinstall):
         fatal('This script was called from a virtual environment, can not create a virtual environment again')
         raise SystemExit(1)

### What is the actual behavior?

fails  on venv  error

### Steps to reproduce.

running install.sh in a python venv environment.

### Build or installation Logs.

```plain
This script was called from a virtual environment, can not create a virtual environment again'
```

### More Information.

No, known error, pulled branch above, diff still not in code base.

dobairoland, 2024-11-27:
Hi @hubbardmeister. Thanks for your report. I wonder if your issue really relates to the linked duplicate.

Are you using NIXOS or something else? Could you please clarify the diff you provided? I think some characters are missing.

dobairoland, 2024-12-16:
I'm closing this for inactivity.

hubbardmeister, 2024-12-31:
Apologies for tardy response. Still an issue -

i'm running Ubuntu, have venv for my python/shell environment. This is still an issue, just did a fresh download today and had to modify idf_tools.py again.

 (venv) rhubbard@rhubbard-XPS-8900:~/esp/esp-idf$ get_idf
Checking "python3" ...
Python 3.12.8
"python3" has been detected
Activating ESP-IDF 5.5
Setting IDF_PATH to '/home/rhubbard/esp/esp-idf'.
* Checking python version ... 3.12.8
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... bash
* Detecting outdated tools in system ... OK - no outdated tools found
* Shell completion ... Autocompletion code generated

Attached diff -
[gitdiff.txt](https://github.com/user-attachments/files/18283425/gitdiff.txt)

Her is the Diff plain text.

diff --git a/tools/idf_tools.py b/tools/idf_tools.py
index b5e4f024ec..b7dbd24c11 100755
--- a/tools/idf_tools.py
+++ b/tools/idf_tools.py
@@ -2581,7 +2581,7 @@ def action_install_python_env(args):  # type: ignore
     nix_store = os.environ.get('NIX_STORE')
     is_nix = nix_store is not None and sys.base_prefix.startswith(nix_store) and sys.prefix.startswith(nix_store)

-    is_virtualenv = not is_nix and (hasattr(sys, 'real_prefix') or (hasattr(sys, 'base_prefix') and sys.base_prefix != sys.prefix))
+    is_virtualenv = not is_nix and (hasattr(sys, 'real_prefix') and (hasattr(sys, 'base_prefix') and sys.base_prefix != sys.prefix))
     if is_virtualenv and (not os.path.exists(idf_python_env_path) or reinstall):
         fatal('This script was called from a virtual environment, can not create a virtual environment again')
         raise SystemExit(1)
~                             

This is pulled from mater branch.
git branch --show-current
master