espressif/esp-idf #14941: [USB] CONFIG_USB_HOST_HW_BUFFER_BIAS_* configs not working

zry98, 2024-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

According to the docs, setting one of the [USB host hardware FIFO size biasing configs](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s3/api-reference/kconfig.html#config-usb-host-hw-buffer-bias) (`CONFIG_USB_HOST_HW_BUFFER_BIAS_BALANCED`, `CONFIG_USB_HOST_HW_BUFFER_BIAS_IN` and `CONFIG_USB_HOST_HW_BUFFER_BIAS_PERIODIC_OUT`) should change the MPS limits.

### What is the actual behavior?

MPS limits not changed whatever bias has been set.

### Steps to reproduce.

1. Modify the source in [`components/usb/hcd_dwc.c`](https://github.com/espressif/esp-idf/blob/c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74/components/usb/hcd_dwc.c#L1651), add some conditional log lines to print the 3 MPS limits set for FIFOs:
```cpp
    // Check if pipe MPS exceeds HCD MPS limits (due to DWC FIFO sizing)
#ifdef CONFIG_USB_HOST_HW_BUFFER_BIAS_IN
    ESP_LOGI(HCD_DWC_TAG, "CONFIG_USB_HOST_HW_BUFFER_BIAS_IN: %d, %d, %d", mps_limits.in_mps, mps_limits.non_periodic_out_mps, mps_limits.periodic_out_mps);
#elif CONFIG_USB_HOST_HW_BUFFER_BIAS_PERIODIC_OUT
    ESP_LOGI(HCD_DWC_TAG, "CONFIG_USB_HOST_HW_BUFFER_BIAS_PERIODIC_OUT: %d, %d, %d", mps_limits.in_mps, mps_limits.non_periodic_out_mps, mps_limits.periodic_out_mps);
#else   // CONFIG_USB_HOST_HW_BUFFER_BIAS_BALANCED
    ESP_LOGI(HCD_DWC_TAG, "CONFIG_USB_HOST_HW_BUFFER_BIAS_BALANCED: %d, %d, %d", mps_limits.in_mps, mps_limits.non_periodic_out_mps, mps_limits.periodic_out_mps);
#endif
    int limit;
```

2. Build & flash & monitor 3 times, each with different bias option set in `sdkconfig`.
  - With `CONFIG_USB_HOST_HW_BUFFER_BIAS_BALANCED=y`:
```plaintext
I (2713) HCD DWC: CONFIG_USB_HOST_HW_BUFFER_BIAS_BALANCED: 408, 256, 128
E (2714) HCD DWC: EP MPS (192) exceeds supported limit (128)
```
  - With `CONFIG_USB_HOST_HW_BUFFER_BIAS_IN=y`:
```plaintext
I (3736) HCD DWC: CONFIG_USB_HOST_HW_BUFFER_BIAS_IN: 408, 256, 128
E (3736) HCD DWC: EP MPS (192) exceeds supported limit (128)
```

  - With `CONFIG_USB_HOST_HW_BUFFER_BIAS_PERIODIC_OUT=y`:
```plaintext
I (5782) HCD DWC: CONFIG_USB_HOST_HW_BUFFER_BIAS_PERIODIC_OUT: 408, 256, 128
E (5783) HCD DWC: EP MPS (192) exceeds supported limit (128)
```

roma-jam, 2024-11-27:
Hi @zry98 ,

thanks for the reporting the issue. 

Unfortunately, I am not able to reproduce the issue not on v5.3 with the same log as you did (neither on latest, which is v5.5 so far). Values are changed in my case, based on the `menuconfig` settings. 

Just to be clear what I did: 
- Add the debug output line you have provided
- Run `menuconfig` and change the USB-OTG settings
- Rebuild and flash the application

Please, verify the steps and confirm that I didn't miss anything. 

In my case the limits are next:
```
[in, n/p_out, p_out] = {
    408, 256, 128, // Balanced
    600, 64, 128,  // Bias In
    128, 64, 600 // Bias PTX
}

```
Thus, I could confirm another problem, that FIFO distribution has changed in balanced mode (256,128 instead of 192,192) from v5.0.

Could you please try out your scenario with the same debug lines with esp-idf `release/v5.0` and share the result here?

zry98, 2025-01-21:
Hi @roma-jam , I've tested with `release/v5.0` but my added log lines are not getting printed. I've also tested with `release/v5.4` but same results as before with v5.3.1. I'm using the example [`usb_camera_mic_spk`](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk) from espressif/esp-iot-solution.

roma-jam, 2025-01-21:
Hi @zry98, 

got it. Somehow I missed that you are trying to build `usb_camera_mic_spk` example. 

This example doesn't use the USB Host Library, it uses another component, which is called `usb_stream`. 
You may find it's code [here](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/usb_stream/usb_stream.c)

`usb_stream` doesn't use USB Host Library, so the configuration for USB Host Library doesn't affect it. This means that changing the configuration for USB Host Library won't change the bias in the `usb_camera_mik_spk` example.  

I will close the issue as it is not related to USB Host Library or esp-idf itself. 

If there is still a problem or you have any other questions regarding this example, please use the esp-iot-solutions issue tracker: https://github.com/espressif/esp-iot-solution/issues

closed 2025-01-21 by roma-jam