espressif/esp-idf #14942: esp_timer定时约24小时导致Crash的BUG

shu7734, 2024-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

 ESP-IDF v5.3-dirty

### Espressif SoC revision.

ESP32-C3 v0.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

官方开发板

### Power Supply used.

USB

### What is the expected behavior?

应该等待定时的时间到期，调用Callback

### What is the actual behavior?

声明了一个全局timer handle
根据官方文档
esp_timer_init
esp_timer_create
esp_timer_start_once
在callback函数调用结束后，发送通知消息，主task在收到通知消息后，重新esp_timer_start_once
esp_timer_start_once的timeout_us参数值，在接近或超过24小时，比如85899*1000000
esp_timer_start_once调用不返回错误，
但是马上调用err = esp_timer_get_expiry_time(sched_timer[index], &sec);
并不会响应错误，但是sec值是0，并且esp_timer_start_once后会马上调用callback函数，最终导致定时器无限循环被调用
之前我以为是我配置了Newlib Nano导致的，但是禁用后问题依然
是否timeout_us有最大值限制？超过限制会置0？还是BUG导致？
__

### Steps to reproduce.

esp_timer_handle_t timer1 = NULL;
QueueHandle_t sched_timer_queue = NULL;
int next=60000;
const esp_timer_create_args_t task_timer_args = {
			.callback = &Sched_Task_Callback,
			.arg = (void *)0, 
			.name = "task_timer"};
		ESP_ERROR_CHECK(esp_timer_create(&task_timer_args, &timer1)

sched_timer_queue = xQueueCreate(10, sizeof(uint8_t));
ESP_ERROR_CHECK(esp_timer_start_once(timer1 , next * 1000000));

static void IRAM_ATTR Sched_Task_Callback(void *arg)
{
int val = 0;
ESP_LOGI("Sched_Task_Callback", "Sched_Task_Callback");
xQueueSend(sched_timer_queue,
			   &val ,
			   0);
}
另一个task
while (1)
	{
		BaseType_t xReturn = pdTRUE;
		uint8_t val= 0;
		xReturn = xQueueReceive(sched_timer_queue, &val, portMAX_DELAY);
		if (pdTRUE == xReturn)
		{
                                 next+=2000
                                 uint64_t sec = next * 1000000;
				//esp_err_t err = esp_timer_start_once(timer1 , sec);
				esp_err_t err = esp_timer_start_once(timer1 , sec);
				if (err != ESP_OK)
				{
					ESP_LOGE("timer_handler_task", "Failed to start timer: %s", esp_err_to_name(err));
				}
				// ESP_ERROR_CHECK(esp_timer_start_periodic(timer1 , next * 1000000));
				sec = 0;
				err = esp_timer_get_expiry_time(timer1 , &sec);
				if (err != ESP_OK)
				{
					ESP_LOGE("timer_handler_task", "Failed to esp_timer_get_expiry_time: %s", esp_err_to_name(err));
				}
				ESP_LOGI("timer_handler_task", "Timer Sec is %llu us \n", sec);//如果next值达到一定大小，此处sec为0,会立即调用callback，把esp_timer_start_once换成esp_timer_start_periodic也是一样
}

KonstantinKondrashov, 2024-11-27:
Hi @shu7734!
I run your code (please next time provide well-formatted code). There is an overflow issue when you use `next` as `int`. Use `uint64_t` for `next` or `1000000ULL` for conversion to usec.
```c
            next = 60000;
            next+=85899;
            // (60000 + 85899 * 1000000) = 85899060000
            // 0x0013FFFBA320 -> int can take only 0xFFFBA320
            // 
            uint64_t sec = next * 1000000;
```

shu7734, 2024-11-27:
sorry,It was my mistake,Thanks!

closed 2024-11-27 by shu7734