espressif/esp-idf #14943: CONFIG_LWIP_LOCAL_HOSTNAME issues

pavel808, 2024-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have an application running on an ESP32-S3, on an ESP Thread Border Router device.

I have the default hostname assigned inside sdkconfig as follows : 

`CONFIG_LWIP_LOCAL_HOSTNAME="espressif" `

It even appears on my router web page : 

`espressif     WLAN    192.168.179.72  2,4 GHz, 72 / 72 Mbit/s`

When I try to ping this hostname from an Ubuntu computer on the same network, it is not recognized : 

```
$ ping espressif
ping: espressif: Name or service not known
```
How would I go about solving this? Thank you in advance.

KaeLL, 2024-11-27:
https://components.espressif.com/components/espressif/mdns

pavel808, 2024-12-01:
@KaeLL Thanks for the information

closed 2024-12-01 by pavel808