espressif/esp-idf #14944: Undefined reference to `floor'

AlexanderDKazakov, 2024-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Get successful compilation

### What is the actual behavior?

Linker error in **libstdc++.a**  undefined references to 'floor':

```bash          
Error: Linker .../embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          ....embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: ...embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/lib/rv32imc_zicsr_zifencei/ilp32/no-rtti/libstdc++.a(hashtable_c++0x.o): in function `std::__detail::_Prime_rehash_policy::_M_next_bkt(unsigned int) const':
          /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/gcc/libstdc++-v3/src/c++11/hashtable_c++0x.cc:85:(.text._ZNKSt8__detail20_Prime_rehash_policy11_M_next_bktEj+0x4e): undefined reference to `floor'
          ...embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: ...embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/lib/rv32imc_zicsr_zifencei/ilp32/no-rtti/libstdc++.a(hashtable_c++0x.o): in function `std::__detail::_Prime_rehash_policy::_M_need_rehash(unsigned int, unsigned int, unsigned int) const':
          /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/gcc/libstdc++-v3/src/c++11/hashtable_c++0x.cc:114:(.text._ZNKSt8__detail20_Prime_rehash_policy14_M_need_rehashEjjj+0x78): undefined reference to `floor'
          ...embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/gcc/libstdc++-v3/src/c++11/hashtable_c++0x.cc:118:(.text._ZNKSt8__detail20_Prime_rehash_policy14_M_need_rehashEjjj+0xdc): undefined reference to `floor'
          collect2: error: ld returned 1 exit status
```

### Steps to reproduce.

I am compiling some chunk of foreign code and my linkage fails with the error what I've described earlier. Compilation goes fine.

### More Information.

v5.3.1 provide even more linker error for **libnosys.a**:
* rewinddir
* seekdir
* telldir
* readdir
* chdir
* getcwd
* mkdir
* closedir
* opendir

AlexanderDKazakov, 2024-11-27:
Could be related to the issue:
https://github.com/gcc-mirror/gcc/blob/master/libstdc%2B%2B-v3/src/c%2B%2B11/hashtable_c%2B%2B0x.cc

igrr, 2024-11-27:
Hi @AlexanderDKazakov, could you please try to share a minimal project which reproduces this issue?

Simply using std::unordered_map (which uses this hashtable implementation internally) in a minimal project doesn't reproduce this linking error, `floor` gets linked correctly:

<details><summary>(test code)</summary>
Code from https://en.cppreference.com/w/cpp/container/unordered_map:

```c++
#include <unordered_map>

#include <iostream>
#include <string>
#include <unordered_map>
 
extern "C" void app_main(void)
{
    // Create an unordered_map of three strings (that map to strings)
    std::unordered_map<std::string, std::string> u =
    {
        {"RED", "#FF0000"},
        {"GREEN", "#00FF00"},
        {"BLUE", "#0000FF"}
    };
 
    // Helper lambda function to print key-value pairs
    auto print_key_value = [](const auto& key, const auto& value)
    {
        std::cout << "Key:[" << key << "] Value:[" << value << "]\n";
    };
 
    std::cout << "Iterate and print key-value pairs of unordered_map, being\n"
                 "explicit with their types:\n";
    for (const std::pair<const std::string, std::string>& n : u)
        print_key_value(n.first, n.second);
 
    std::cout << "\nIterate and print key-value pairs using C++17 structured binding:\n";
    for (const auto& [key, value] : u)
        print_key_value(key, value);
 
    // Add two new entries to the unordered_map
    u["BLACK"] = "#000000";
    u["WHITE"] = "#FFFFFF";
 
    std::cout << "\nOutput values by key:\n"
                 "The HEX of color RED is:[" << u["RED"] << "]\n"
                 "The HEX of color BLACK is:[" << u["BLACK"] << "]\n\n";
 
    std::cout << "Use operator[] with non-existent key to insert a new key-value pair:\n";
    print_key_value("new_key", u["new_key"]);
 
    std::cout << "\nIterate and print key-value pairs, using `auto`;\n"
                 "new_key is now one of the keys in the map:\n";
    for (const auto& n : u)
        print_key_value(n.first, n.second);
}
```

</details>

results in:
```
$ grep floor build/test.map
...
                0x400f5170                floor
```

I think the issue you met is related to the order of the libraries on the linker command line. Perhaps you can extract the full linker command line from the build logs and check that the order is correct, i.e. `libthirdparty.a -lstdc++ -lc -lm`.

AlexanderDKazakov, 2024-11-28:
It looks like you're right. I dug into the third part and found the **unordered_map**. The linker seems to be doing something wrong. I'll investigate further. 

Thank you for your quick response and the helpful suggestion!

closed 2024-11-28 by AlexanderDKazakov