espressif/esp-idf #14946: ble client example, gatt_rsp_timeout and disconnect after a few seconds

Astdg9826, 2024-11-27:
Hello,

I run the gatt_client example with ESP32 and use it to connect a ble server running on an Android device.
The connection can be established and Ble notification can be received.
But after 20-30 sec, "BT_GATT: gatt_rsp_timeout" will occur and disconnection will be made.
I found that this may be related to the void gatt_rsp_timeout(TIMER_LIST_ENT *p_tle) function in gatt_utils.c.

What is the cause of the problem and how to fix it?
Thank you!

The program's log is shown below.
==================================
I (31693) GATTC_DEMO: 01 00 62 22 37 84
I (31873) GATTC_DEMO: ESP_GATTC_NOTIFY_EVT, receive notify value:
I (31873) GATTC_DEMO: 01 00 62 22 37 84
I (32063) GATTC_DEMO: ESP_GATTC_NOTIFY_EVT, receive notify value:
I (32063) GATTC_DEMO: 01 00 61 22 37 84
I (32243) GATTC_DEMO: ESP_GATTC_NOTIFY_EVT, receive notify value:
I (32243) GATTC_DEMO: 01 00 61 22 37 84
W (32263) BT_GATT: gatt_rsp_timeout disconnecting...
W (32263) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (32263) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
E (32263) GATTC_DEMO: write descr failed, error status = 85
W (32273) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (32283) GATTC_DEMO: ESP_GATTC_DISCONNECT_EVT, reason = 22

===========================

closed 2024-11-28 by Astdg9826

esp-zhp, 2024-11-29:
@Astdg9826 have you fixed this issue?

Astdg9826, 2024-11-29:
Thank you for asking! I found the problem, which is on the Android program side. It failed to respond a ble call from esp32.