espressif/esp-idf #14947: Camera Initialization Issue on ESP32-S3 Custom PCB

hardikofficial963, 2024-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We are currently developing a custom PCB based on the ESP32-S3 that integrates the following components:
    - OV5640 Camera
    - ES8311 Codec IC (for microphone and speaker functionality)
    - Servo Port
    - I2C Port
    - SD Card Port

While all other components are functioning as expected, we are facing a recurring issue with the camera.

Issue Description:
The PCB works fine initially after assembly, but after some time, the camera fails to initialize, showing errors such as:

    "Camera initialization error"
    "Camera SCCB failed"

Steps Taken:

1. Testing with Custom PCB:
The camera works immediately after assembly but stops functioning after some time.
Testing with ESP32-S3 Development Board:

2. We connected the OV5640 camera using an FFC breakout board and the same pins used in the custom PCB. The same initialization errors occur.

Attachments for Reference:
    - PCB Design Schematic  (1-4, 2-4)
    - PCB Layout   (3-4)
    - List of Pins Used  (4-4)

Could you please help us identify and resolve the issue? Any insights or suggestions would be greatly appreciated.

Looking forward to your guidance.
[image]
[image]
[image]
[image]

hardikofficial963, 2024-11-28:
what pull-up registers are used by the I2C bus?
Yes, we have tried 2k, 5k, 10k ohm pull up.

Are you using the esp32-camera library for driving the camera sensor?
Yes

Can you use the i2c driver to read/write OV5640 registers correctly (no other dependency, just using the i2c master driver in the esp-idf)
No, currently we are trying to test camera with Arduino Camera webserver example.

WangYuxin-esp, 2024-11-29:
This issue is not clear enough to determine whether it is a problem with the sensor or the development board.  
[Here](https://github.com/espressif/esp-who/blob/master/docs/zh_CN/get-started/ESP32-S3-EYE_Getting_Started_Guide.md) is a reference document.
Perhaps using an official development board and running the [examples](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera) here is a good way to check.

hardikofficial963, 2024-11-29:
> This issue is not clear enough to determine whether it is a problem with the sensor or the development board. [...]

@WangYuxin-esp 

The OV5640 sensor works perfectly on the ESP32-S3-Korvo-2 V3.0, as I’ve tested it thoroughly. However, we’re encountering reliability issues with our custom PCB.

Initially, when the PCB is assembled, the camera functions as expected, but after some time, it begins to throw a "Camera Initialization Error", "Camera SCCB Error", "Camera Not Found".
 In this state, if I move the same camera module to the ESP32-S3-Korvo-2, it continues to work flawlessly.

This leads me to suspect a hardware-related issue with our custom PCB. Could you please provide guidance on potential causes or troubleshooting steps to resolve this problem?

WangYuxin-esp, 2024-11-29:
Firstly, double check that the pin configuration in the software configuration is consistent with the hardware.
Then we can use an oscilloscope or logic analyzer to compare the relevant circuits on the official development board and custom development board. Including power pins such as XCLK Pin and DVDD Pins.

hardikofficial963, 2024-11-30:
> Firstly, double check that the pin configuration in the software configuration is consistent with the hardware. Then we can use an oscilloscope or logic analyzer to compare the relevant circuits on [...]

@WangYuxin-esp
Yes, I have checked the pin configuration, it is accurate.

WangYuxin-esp, 2024-12-02:
Perhaps submitting a [hardware review report](https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review) is a better option.

hardikofficial963, 2024-12-02:
> Perhaps submitting a [hardware review report](https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review) is a better option.

@WangYuxin-esp 
Yes, I have submitted Schematic and PCB Layout for Hardware Design Review.

hardikofficial963, 2024-12-02:
@WangYuxin-esp 
I am curious, if ESP provides a Ticket number or some ID so that I track this issue.

WangYuxin-esp, 2024-12-02:
This seems more like a hardware issue, you can submit a review request first. Then there will be relevant emails to reply to you. The issue channel here is mainly used to provide feedback on software issues, So as far as I know, there won't be an ID associated with these two things.

hardikofficial963, 2024-12-02:
> This seems more like a hardware issue, you can submit a review request first. Then there will be relevant emails to reply to you. The issue channel here is mainly used to provide feedback on software issues, So as far as I know, there won't be an ID associated with these two things.

I understand, Thanks @WangYuxin-esp for helping us  : )

closed 2024-12-03 by Alvin1Zhang