espressif/esp-idf #14949: BLE Observer Role required for scan response

PironexJRunge, 2024-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I would expect to receive scan responses while scanning when only BLE Central is enabled and BLE Observer is not.

### What is the actual behavior?

When running a slighlty modified version of the nimble example blecent, such that active scanning is used, discovery events for the scan response are only received if "Enable BLE Observer role" is checked in menuconfig. If only Enable BLE Central role is checked, scan responses are not received.

### Steps to reproduce.

1. Create Project based on nimble/blecent example
2. Modify `ble_cts_cent_gap_event`  as follows (Added specific output on scan response, remove other output, avoid connects): 
```
ble_cts_cent_gap_event(struct ble_gap_event *event, void *arg)
{
    struct ble_gap_conn_desc desc;
    struct ble_hs_adv_fields fields;
    int rc;

    switch (event->type) {
    case BLE_GAP_EVENT_DISC:
		if(event->disc.event_type == BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP){
			ESP_LOGI(tag, "Scan response event received");
		}
        rc = ble_hs_adv_parse_fields(&fields, event->disc.data,
                                     event->disc.length_data);
        if (rc != 0) {
            return 0;
        }

        /* An advertisment report was received during GAP discovery. */
        // print_adv_fields(&fields);

        /* Try to connect to the advertiser if it looks interesting. */
        // ble_cts_cent_connect_if_interesting(&event->disc);
        return 0;

...
}
```
3. In `ble_cts_cent_scan` change the following line to enable active scanning:
```
    /**
     * Perform a passive scan.  I.e., don't send follow-up scan requests to
     * each advertiser.
     */
    disc_params.passive = 0;
```
4. Leave menuconfig as is and build and run the example. after starting the scan, you should see "Scan response event received" outputs in the console
5. In menuconfig, disable all roles except central role. (Bluetooth->NimBLE Options)
6. Rebuild and run the example. No "Scan response event received" outputs are shown since no scan response is received
7. (opt.) In menuconfig reenable Observer role (only Central and Observer enabled now)
8. (opt.) Rebuild an run the example. Scan response output is available again

rahult-github, 2024-11-28:
Hi @PironexJRunge , 

If observer role is enabled, nimble will post all adv reports regardless of flags AD types .  

If observer role is not enabled, there are stringent checks for Flags AD type ( General / Limited flag ) . If these are not set, then the adv packet is dropped by nimble and hence you don't observe them since the reports are not passing for the checks ( i.e. they are not formatted correctly with all exact information).  The code can be checked [here](https://github.com/espressif/esp-nimble/blob/ca6c6173fca923cfa1966891fc6623fd9f9cb323/nimble/host/src/ble_gap.c#L1795).

PironexJRunge, 2024-11-28:
Hi @rahult-github,
thanks for your fast reply. I checked the code and compared what the device that I am scanning is providing and the device is only sending its AD flags in the advertising data, not in its scan response. Is this an issue with the device`s BT protocol? 
Or is there any way to combine  the advertising and scan response data for the sanity check? Currently the advertising and the scan response are given separately to the sanity check, which makes it fail for the scan response.

This issue occured to me while upgrading from 5.0.2 to 5.3.1. Looking into 5.0.2 no sanity check was done at all when the limited flag was set to 0.

Kind regards
Jan

rahult-github, 2024-11-28:
> Is this an issue with the device`s BT protocol? 

For general discoverable mode, adv reports should have some AD flag. 

[image]

And for non-discoverable mode: 

[image]

By not setting the AD type, the advertising events are received but being treated as non-discoverable and hence being dropped. 

By not setting observer role, the above conditions get checked and hence the observation. 

> This issue occured to me while upgrading from 5.0.2 to 5.3.1. Looking into 5.0.2 no sanity check was done at all when the limited flag was set to 0.

True, we migrate nimble versions and also keep adding fixes. So code difference observed .

PironexJRunge, 2024-11-28:
I think the main issue for me here is, that the advertising data does contain AD flags, but the scan response does not. 
Adding two log outputs to `ble_gap_rx_adv_report`:
```
void
ble_gap_rx_adv_report(struct ble_gap_disc_desc *desc)
{
#if NIMBLE_BLE_SCAN
    ESP_LOGI("SCN", "Event type: %d with data:", desc->event_type);
    ESP_LOG_BUFFER_HEX("SCN", desc->data, desc->length_data);
    if (ble_gap_rx_adv_report_sanity_check(desc->data, desc->length_data)) {
        return;
    }

    ble_gap_disc_report(desc);
#endif
}
```
provides the following output for the BT device I am interested in:
```
I (2985) SCN: Event type: 0 with data:
I (2985) SCN: 02 01 06 06 09 65 6c 6c 69 6f 0d ff c0 01 11 01 
I (2985) SCN: 01 06 90 38 0c fb 5d 62 
I (3025) SCN: Event type: 4 with data:
I (3025) SCN: 09 09 65 6c 6c 69 6f 54 61 67
```
We can see that the first three bytes of the advertising contain the AD flag field, so sanity check runs fine while the scan response only contains the device name and is thereby discarded by the sanity check. Is this an intended behaviour? Or would you expect that the scan response contains an AD flag field as well?
From my point of view the scan response is an addition to the advertising data already received and therefore must not contain the AD field again but I might be wrong on this.

rahult-github, 2024-11-29:
Hi @PironexJRunge ,

> From my point of view the scan response is an addition to the advertising data already received and therefore must not contain the AD field again  

Vol 3, Part C, Section 11 has the Advertising and **Scan Response data format**.  So it is safe to say that AD information is part of scan response .

[image]

PironexJRunge, 2024-11-29:
Alright, thanks for your efforts, I will contact the device manufacturer then and ask them how to get their device to provide a proper scan response.
Kind regards
Jan

closed 2024-11-29 by PironexJRunge

DrSegatron, 2025-03-29:
I'm not convinced by your reasoning that the Scan Response PSU should contain the Flags AD Type containing the General Discoverabillity flag to be considered for General Discoverability:

Section 9.2.6 on General Discoverability differentiates between "advertising data" and "scan response data".
[image]

In the conditions for General Discovery, only advertising data is specified:
[image]

**Therefore, I consider the new flags check in `ble_gap_rx_adv_report_sanity_check()` to be in error and therefore a bug.**

Luckily, the workaround is simple, as previously mentioned: Enable observer role to bypass the check.