espressif/esp-idf #14950: How can I send data of type int32 or larger from esp32-h2 module to another esp32-h2 module

mehmetbozhaydar, 2024-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am new to esp-idf. Which library should I use to send int32 type data with Zigbee?

suda-morris, 2024-11-29:
One of the data serialization/deserialization libraries could be used in esp-idf: [cbor](https://components.espressif.com/components/espressif/cbor/)

mehmetbozhaydar, 2024-11-29:
I'm unsure how to integrate CBOR with Zigbee and haven't been able to locate any relevant examples demonstrating their combined use. @suda-morris

xieqinan, 2024-12-10:
Hi @mehmetbozhaydar,

This issue is similar to [this one](https://github.com/espressif/esp-zigbee-sdk/issues/495), which has already been resolved. Let's close this issue, if you have other question, we can continue the discussion on the esp-zigbee-sdk repository.

closed 2024-12-11 by mehmetbozhaydar