espressif/esp-idf #14955: 在使用http请求的时候无法自定义头部字节数据

cannnnnnnnnnnn, 2024-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

使用软件版本V5.3.1，参考file_serving，在http请求中发送html文件，不知如何设置http请求中的响应标头字段数据
[image]

cannnnnnnnnnnn, 2024-11-29:
在使用**httpd_resp_send_chunk**函数时，查看函数**httpd_chunked_hdr_str**字段数据可以随意修改吗？

larryli, 2024-12-02:
https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/examples/protocols/http_server/file_serving/main/file_server.c#L398

cannnnnnnnnnnn, 2024-12-03:
> https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/examples/protocols/http_server/file_serving/main/file_server.c#L398

好的，感谢您的解答