espressif/esp-idf #14957: Crash in xTaskRemoveFromEventList

jimmyw, 2024-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect it not to crash

### What is the actual behavior?

In production with 1000X devices, i get 30 completely random crash reports every day, with similar pattern.

### Steps to reproduce.

No specific way to reproduce, in fact have never seen int reproduced my self.

### Build or installation Logs.

```plain
Example output from crash #7870

Stack trace from tiT task:

==================== CURRENT THREAD STACK =====================
#0  0x40381018 in xTaskRemoveFromEventList (pxEventList=<optimized out>) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/tasks.c:3869
#1  0x4037e906 in xQueueGenericSend (xQueue=0x3fcd976c, pvItemToQueue=0x0, xTicksToWait=<optimized out>, xCopyPosition=0) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/queue.c:993
#2  0x420ab7dd in sys_sem_signal (sem=0x3c2cfc1c) at /opt/esp/idf/components/lwip/port/freertos/sys_arch.c:134
#3  0x420accbe in lwip_netconn_do_dns_found (name=0x3c2b8438 <dns_table+36> "", ipaddr=0x3c2b8418 <dns_table+4>, arg=0x3fcb772c <s_mqtt_client_stack+5172>) at /opt/esp/idf/components/lwip/lwip/src/api/api_msg.c:2215
#4  0x42096a64 in dns_call_found (idx=0 '\000', addr=0x3c2b8418 <dns_table+4>) at /opt/esp/idf/components/lwip/lwip/src/core/dns.c:1026
#5  0x42096b48 in dns_correct_response (idx=0 '\000', ttl=60) at /opt/esp/idf/components/lwip/lwip/src/core/dns.c:1224
#6  0x42097784 in dns_recv (arg=<optimized out>, pcb=<optimized out>, p=0x3c2de9d4, addr=<optimized out>, port=<optimized out>) at /opt/esp/idf/components/lwip/lwip/src/core/dns.c:1371
#7  0x4209e531 in udp_input (p=0x3c2de9d4, inp=0x3fcf7558) at /opt/esp/idf/components/lwip/lwip/src/core/udp.c:404
#8  0x420a1d49 in ip4_input (p=0x3c2de9d4, inp=0x3fcf7558) at /opt/esp/idf/components/lwip/lwip/src/core/ipv4/ip4.c:746
#9  0x420a7e99 in ethernet_input (p=0x3c2de9d4, netif=0x3fcf7558) at /opt/esp/idf/components/lwip/lwip/src/netif/ethernet.c:186
#10 0x42096318 in tcpip_thread_handle_msg (msg=0x3c2d08fc) at /opt/esp/idf/components/lwip/lwip/src/api/tcpip.c:174
#11 0x42096378 in tcpip_thread (arg=0x0) at /opt/esp/idf/components/lwip/lwip/src/api/tcpip.c:148
#12 0x4037ee08 in vPortTaskWrapper (pxCode=0x4209634c <tcpip_thread>, pvParameters=0x0) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

Stack from task that does dns request:
```
==================== THREAD 18 (TCB: 0x3fcb7af8, name: 'mqtt_task') =====================
#0  0x400559e0 in ?? ()
#1  0x4037f0b5 in vPortClearInterruptMaskFromISR (prev_level=<optimized out>) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:564
#2  vPortExitCritical (mux=0x3fcd97c0) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504
#3  0x4037ec50 in xQueueSemaphoreTake (xQueue=0x3fcd976c, xTicksToWait=<optimized out>) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/queue.c:1853
#4  0x420ab80c in sys_arch_sem_wait (sem=0x3c2cfc1c, timeout=0) at /opt/esp/idf/components/lwip/port/freertos/sys_arch.c:165
#5  0x42096464 in tcpip_send_msg_wait_sem (fn=0x420ae05c <lwip_netconn_do_gethostbyname>, apimsg=0x3fcb772c <s_mqtt_client_stack+5172>, sem=0x3c2cfc1c) at /opt/esp/idf/components/lwip/lwip/src/api/tcpip.c:456
#6  0x420acc50 in netconn_gethostbyname_addrtype (name=0x3c2cfa5c <error: Cannot access memory at address 0x3c2cfa5c>, addr=0x3fcb7778 <s_mqtt_client_stack+5248>, dns_addrtype=2 '\002') at /opt/esp/idf/components/lwip/lwip/src/api/api_lib.c:1325
#7  0x420936e4 in lwip_getaddrinfo (nodename=0x3c2cfa5c <error: Cannot access memory at address 0x3c2cfa5c>, servname=0x0, hints=0x3fcb77d0 <s_mqtt_client_stack+5336>, res=0x3fcb77cc <s_mqtt_client_stack+5332>) at /opt/esp/idf/components/lwip/lwip/src/api/netdb.c:345
#8  0x420be50c in getaddrinfo (res=0x3fcb77cc <s_mqtt_client_stack+5332>, hints=0x3fcb77d0 <s_mqtt_client_stack+5336>, servname=0x0, nodename=0x3c2cfa5c <error: Cannot access memory at address 0x3c2cfa5c>) at /opt/esp/idf/components/lwip/include/lwip/netdb.h:23
#9  esp_tls_hostname_to_fd (host=0x3c2cfbb4 <error: Cannot access memory at address 0x3c2cfbb4>, hostlen=45, port=8883, addr_family=<optimized out>, address=0x3fcb7854 <s_mqtt_client_stack+5468>, fd=0x3fcb782c <s_mqtt_client_stack+5428>) at /opt/esp/idf/components/esp-tls/esp_tls.c:203
[... 70 lines cut ...]
```
#define listREMOVE_ITEM( pxItemToRemove ) \
    {                                     \
        List_t * const pxList = ( pxItemToRemove )->pxContainer;    <-- NULL     \
                                                                                 \
        ( pxItemToRemove )->pxNext->pxPrevious = ( pxItemToRemove )->pxPrevious; \
        ( pxItemToRemove )->pxPrevious->pxNext = ( pxItemToRemove )->pxNext;     \
CRASH-> if( pxList->pxIndex == ( pxItemToRemove ) )                              \
        {                                                                        \
            pxList->pxIndex = ( pxItemToRemove )->pxPrevious;                    \
        }                                                                        \
                                                                                 \
        ( pxItemToRemove )->pxContainer = NULL;                                  \
        ( pxList->uxNumberOfItems )--;                                           \
    }
```

Its clearly some kind of race condition, where pxContainer is NULL, and derefering the value in pxList->pxIndex Causes a read of NULL + 0x4, that causes a LoadProhibitedCause

This stack trace is not unique, this happens in other places when calling xTaskRemoveFromEventList.

Is there anyone that recognizes this, or can give me guidance on where to go from here?

The issue here is that this is so rare that i need 1000 devices to trigger the error. All my attempts to trigger it locally have failed, so i just cant try master version, or randomly change stuff.
```

KonstantinKondrashov, 2024-11-29:
Hi @jimmyw!
Thanks for the report. Please add sdkconfig and specify which chip you use. I guess it is a two-core chip.

jimmyw, 2024-11-29:
ESP32S3, Using External SPI ram.  Will add sdkconfig shortly.

KonstantinKondrashov, 2024-12-02:
It seems like the item has already been removed from the list `( pxItemToRemove )->pxContainer = NULL;` so the next call of `listREMOVE_ITEM()` leads to this issue, I guess. Yes, it's probably a race condition issue.
We will try to investigate it further.

jimmyw, 2024-12-02:
So i managed to bitsect and deploy with various features enabled/disabled that we did over the last releases.

The thing that made this crash, was that we changed from xTaskCreate to xTaskCreateStatic to static allocate tasks. We have noticed for example the mqtt client when we disconnect and connect, that memory might be so fragmented that we dont have a chunk of 6k internal ram available, and chip and to reset.

Rolling back to the version we did heap allocation of the stack, this rare crash completely stopped.

This could be an coincident, with us having a bug that managed to write  to random ram, and it accidentally was in the TCL of the mqtt task, even if i find that unlikely. Everything pointed to a race.

I have no idea what could have happened, if it was a memory alignment issue or something like that.

We also noticed that the crash was when there was a high memory pressure, and the system malloc started to hand out PSRAM instead of internal ram that was emptied.

So i guess we can close this one, i got no way to reproduce this locally, and it was rarely happening in production with low ram and static tasks. Unless you have any idea on where to look.

jimmyw, 2024-12-02:
This was my patch that made it faile:

```
Jimmys-MacBook-Pro:esp-mqtt jimmy$ git diff aa6f889fb4f6f743b3a550aa587713aabbdca1fc 
diff --git c/Kconfig w/Kconfig
index 9eb6c0b..c128e9d 100644
--- c/Kconfig
+++ w/Kconfig
@@ -109,6 +109,13 @@ menu "ESP-MQTT Configurations"
         help
             MQTT task stack size
 
+    config MQTT_TASK_STACK_STATIC
+        bool "Use static stack for MQTT task"
+        default n
+        depends on MQTT_USE_CUSTOM_CONFIG
+        help
+            Use static stack for MQTT task
+
     config MQTT_DISABLE_API_LOCKS
         bool "Disable API locks"
         default n
diff --git c/mqtt_client.c w/mqtt_client.c
index cece4fd..99ada46 100644
--- c/mqtt_client.c
+++ w/mqtt_client.c
@@ -43,6 +43,11 @@ static int mqtt_message_receive(esp_mqtt_client_handle_t client, int read_poll_t
 static void esp_mqtt_client_dispatch_transport_error(esp_mqtt_client_handle_t client);
 static esp_err_t send_disconnect_msg(esp_mqtt_client_handle_t client);
 
+#if CONFIG_MQTT_TASK_STACK_STATIC
+static StaticTask_t s_mqtt_client_task;
:...skipping...
diff --git c/Kconfig w/Kconfig
index 9eb6c0b..c128e9d 100644
--- c/Kconfig
+++ w/Kconfig
@@ -109,6 +109,13 @@ menu "ESP-MQTT Configurations"
         help
             MQTT task stack size
 
+    config MQTT_TASK_STACK_STATIC
+        bool "Use static stack for MQTT task"
+        default n
+        depends on MQTT_USE_CUSTOM_CONFIG
+        help
+            Use static stack for MQTT task
+
     config MQTT_DISABLE_API_LOCKS
         bool "Disable API locks"
         default n
diff --git c/mqtt_client.c w/mqtt_client.c
index cece4fd..99ada46 100644
--- c/mqtt_client.c
+++ w/mqtt_client.c
@@ -43,6 +43,11 @@ static int mqtt_message_receive(esp_mqtt_client_handle_t client, int read_poll_t
 static void esp_mqtt_client_dispatch_transport_error(esp_mqtt_client_handle_t client);
 static esp_err_t send_disconnect_msg(esp_mqtt_client_handle_t client);
 
+#if CONFIG_MQTT_TASK_STACK_STATIC
+static StaticTask_t s_mqtt_client_task;
+static StackType_t s_mqtt_client_stack[MQTT_TASK_STACK];
+#endif
+
 /**
  * @brief Processes error reported from transport layer (considering the message read status)
  *
@@ -1748,6 +1753,26 @@ esp_err_t esp_mqtt_client_start(esp_mqtt_client_handle_t client)
         return ESP_FAIL;
     }
     esp_err_t err = ESP_OK;
+#if CONFIG_MQTT_TASK_STACK_STATIC
+    if (client->config->task_stack != MQTT_TASK_STACK) {
+        ESP_LOGE(TAG, "MQTT task stack size must be %d", MQTT_TASK_STACK);
+        return ESP_FAIL;
+    }
+
+#if MQTT_CORE_SELECTION_ENABLED
+    ESP_LOGD(TAG, "Core selection enabled on %u", MQTT_TASK_CORE);
+    if (!(client->task_handle=xTaskCreateStaticPinnedToCore(esp_mqtt_task, "mqtt_task", MQTT_TASK_STACK, client, client->config->task_prio, s_mqtt_client_stack, &s_mqtt_client_task, MQTT_TASK_CORE))) {
+        ESP_LOGE(TAG, "Error create mqtt task");
+        err = ESP_FAIL;
+    }
+#else
+    ESP_LOGD(TAG, "Core selection disabled");
ent->config->task_prio, s_mqtt_client_stack, &s_mqtt_client_task))) {
+        ESP_LOGE(TAG, "Error create mqtt task");
+        err = ESP_FAIL;
+    }
+#endif
+#else
 #if MQTT_CORE_SELECTION_ENABLED
     ESP_LOGD(TAG, "Core selection enabled on %u", MQTT_TASK_CORE);
config->task_prio, &client->task_handle, MQTT_TASK_CORE) != pdTRUE) {
@@ -1760,6 +1785,7 @@ esp_err_t esp_mqtt_client_start(esp_mqtt_client_handle_t client)
         ESP_LOGE(TAG, "Error create mqtt task");
         err = ESP_FAIL;
     }
+#endif
 #endif
     MQTT_API_UNLOCK(client);
     return err;
```

euripedesrocha, 2024-12-04:
@jimmyw could you please share the mqtt event handler that you are using? In particular, I'm interested to see the handling of the disconnected event. 

From the first stack trace it looks like this happens just after a disconnection, am I correct?

Unrelated (but important for me :) ), could you share your reasoning for the mqtt task to be static, which problem you were targeting to solve with that change?

jimmyw, 2024-12-04:
Now i`m not personally responsible for this code, as it might not be the most pretty. But i cant see any wrong with it.

```
void mqtt_change_bit_and_fire_event(bool set, EventBits_t bit)
{
    mqtt_event_data_t event_data = {.bit = bit, .set = set};

    if (set) {
        xEventGroupSetBits(mqtt_event_group, bit);
    } else {
        xEventGroupClearBits(mqtt_event_group, bit);
    }

    ESP_ERROR_CHECK(esp_event_post(TIBBER_MQTT_EVENT, TIBBER_MQTT_EVENT_STATUS_CHANGE, &event_data, sizeof(mqtt_event_data_t),
                                   MS_TO_TICKS(CONFIG_TIBBER_MAX_POST_EVENT_BLOCK_MS)));
}

static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    (void)base;
    esp_mqtt_event_handle_t event = event_data;

    switch ((esp_mqtt_event_id_t)event->event_id) {
    case MQTT_EVENT_CONNECTED:
        ESP_LOGD(TAG, "MQTT_EVENT_CONNECTED");
        memset(&last_mqtt_error, 0, sizeof(last_mqtt_error));
        mqtt_change_bit_and_fire_event(BIT_CLEAR, MQTT_DISCONNECTED_BIT);
        mqtt_change_bit_and_fire_event(BIT_CLEAR, MQTT_CONNECTING_BIT);
        mqtt_change_bit_and_fire_event(BIT_SET, MQTT_CONNECTED_BIT);
        break;
    case MQTT_EVENT_DISCONNECTED:
        ESP_LOGW(TAG, "MQTT_EVENT_DISCONNECTED");
        mqtt_change_bit_and_fire_event(BIT_CLEAR, MQTT_CONNECTING_BIT);
        mqtt_change_bit_and_fire_event(BIT_CLEAR, MQTT_CONNECTED_BIT);
        mqtt_change_bit_and_fire_event(BIT_CLEAR, MQTT_PUBLISHED);
        mqtt_change_bit_and_fire_event(BIT_CLEAR, MQTT_SUBSCRIBED_BIT);
        mqtt_change_bit_and_fire_event(BIT_SET, MQTT_DISCONNECTED_BIT);
        break;
...
```

jimmyw, 2024-12-04:
So the reason why we tried static tasks, is that we have high memory pressure, and when changing mqtt server or on a network issues, we will reconnect mqtt again.

Doing esp_mqtt_client_start we will create a mqtt task, allocated on the heap. 
```
esp_err_t esp_mqtt_client_start(esp_mqtt_client_handle_t client)
...
    if (xTaskCreate(esp_mqtt_task, "mqtt_task", client->config->task_stack, client, client->config->task_prio, &client->task_handle) != pdTRUE) {
        ESP_LOGE(TAG, "Error create mqtt task");
        err = ESP_FAIL;
    }

```

This requires 6kb of continuous ram, and in production we have clients with months of runtime causing memory fragmentation to that extent that we did not have 6kb free to allocate. This made the xTaskCreate fail, and the system to restart.

Moving the task to static memory, did solve this issue, as we used the same memory address all the time. It also helped debugging crashes, as the memory offsets where the same in similar crashes, so we could learn the pattern so i kind of liked this approach. 

We since then solved the worst contenders in memory fragmentation so its not as great of a problem right now, that why we could safely revert that change.

jimmyw, 2024-12-04:
Here is another good one that kind of helped, i combining two heap allocations into one in the lwip stack and it seemed to help, but also boosted performance avoiding contention in the heap allocator.

The main issue here was really that the small 4 byte heap allocation that was done just as a 4 byte pointer to a queue, was done in external ram if you had the CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP option enabled.  This is problematic trying to access internal ram possibly in interrupts, by accessing a 4 byte pointer stored in external ram.

This patch allocated the struct and the semaphore in the same go, always in internal ram and helped a lot with stability/performance. Also let me know if you like a PR on this.

```
diff --git a/components/esp_netif/lwip/esp_netif_lwip.c b/components/esp_netif/lwip/esp_netif_lwip.c
index b534fcbfe7..04532455d6 100644
--- a/components/esp_netif/lwip/esp_netif_lwip.c
+++ b/components/esp_netif/lwip/esp_netif_lwip.c
@@ -203,8 +203,8 @@ static void netif_unset_garp_flag(struct netif *netif)
 #endif  // CONFIG_LWIP_GARP_TMR_INTERVAL
 
 #if !LWIP_TCPIP_CORE_LOCKING
-static sys_sem_t api_sync_sem = NULL;
-static sys_sem_t api_lock_sem = NULL;
+static sys_sem_t api_sync_sem = {0};
+static sys_sem_t api_lock_sem = {0};
 #endif
 
 /**
@@ -563,14 +563,14 @@ esp_err_t esp_netif_init(void)
     }
 
 #if !LWIP_TCPIP_CORE_LOCKING
-    if (!api_sync_sem) {
+    if (!sys_sem_valid(&api_sync_sem)) {
         if (ERR_OK != sys_sem_new(&api_sync_sem, 0)) {
             ESP_LOGE(TAG, "esp netif api sync sem init fail");
             return ESP_FAIL;
         }
     }
 
-    if (!api_lock_sem) {
+    if (!sys_sem_valid(&api_lock_sem)) {
         if (ERR_OK != sys_sem_new(&api_lock_sem, 1)) {
             ESP_LOGE(TAG, "esp netif api lock sem init fail");
             return ESP_FAIL;
diff --git a/components/lwip/port/freertos/include/arch/sys_arch.h b/components/lwip/port/freertos/include/arch/sys_arch.h
index ba85471397..b2d53764a1 100644
--- a/components/lwip/port/freertos/include/arch/sys_arch.h
+++ b/components/lwip/port/freertos/include/arch/sys_arch.h
@@ -18,13 +18,14 @@ extern "C" {
 #endif
 
 
-typedef SemaphoreHandle_t sys_sem_t;
+typedef StaticSemaphore_t sys_sem_t;
 typedef SemaphoreHandle_t sys_mutex_t;
 typedef TaskHandle_t sys_thread_t;
 
 typedef struct sys_mbox_s {
-  QueueHandle_t os_mbox;
+  StaticQueue_t os_mbox;
   void *owner;
+  uint8_t buffer[0];
 }* sys_mbox_t;
 
 /** This is returned by _fromisr() sys functions to tell the outermost function
@@ -42,7 +43,6 @@ void sys_delay_ms(uint32_t ms);
 #define sys_mutex_set_invalid( x ) ( ( *x ) = NULL )
 #endif
 
-#define sys_mbox_valid( x ) ( ( ( *x ) == NULL) ? pdFALSE : pdTRUE )
 
 /* Define the sys_mbox_set_invalid() to empty to support lock-free mbox in ESP LWIP.
  *
@@ -61,10 +61,11 @@ void sys_delay_ms(uint32_t ms);
  * However, if the sys_mbox_set_invalid() is not called after sys_mbox_free(), e.g. in netconn_alloc(),
[... 196 lines cut ...]
 sys_thread_sem_init(void)
 {
-  sys_sem_t *sem = (sys_sem_t*)mem_malloc(sizeof(sys_sem_t*));
+  // Always allocate semaphores from internal memory
+  sys_sem_t *sem = (sys_sem_t*)heap_caps_malloc(sizeof(StaticSemaphore_t), ( MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT ));
 
   if (!sem){
     ESP_LOGE(TAG, "thread_sem_init: out of memory");
     return 0;
   }
 
-  *sem = xSemaphoreCreateBinary();
-  if (!(*sem)){
-    free(sem);
-    ESP_LOGE(TAG, "thread_sem_init: out of memory");
+  // If successful, create a binary semaphore the pointer to the semaphore is returned
+  QueueHandle_t res = xSemaphoreCreateBinaryStatic(sem);
+
+  if (res != (QueueHandle_t)sem){
+    heap_caps_free(sem);
+    ESP_LOGE(TAG, "thread_sem_init: unable to create semaphore %p != %p", res, sem);
     return 0;
   }
 
```

euripedesrocha, 2024-12-05:
> Now i`m not personally responsible for this code, as it might not be the most pretty. But i cant see any wrong with it.
>
> ```
> void mqtt_change_bit_and_fire_event(bool set, EventBits_t bit)
> { [...]

Thank you. My question about the disconnected event was because there is a possibility of interaction from the event handler that might be dispatched from the user task. So depending on what you do, it could be problematic.

euripedesrocha, 2024-12-05:
Regarding your changes and PR creation, the PRs are welcomed, and you can create them to direct the discussions to the correct teams. I suggest that you split the changes in 2 different PRs to speed up the reviews.

sudeep-mohanty, 2025-02-18:
Analysing the code at `xTaskRemoveFromEventList()` and other parts of the FreeRTOS kernel, it does not seem likely that a race condition causes this crash. Since, the issue is resolved already by using dynamic memory for the mqtt task, I will close this ticket for now. In case, you observe the crash in `xTaskRemoveFromEventList()` again, please feel free to open a new ticket for us to investigate. Thanks.

closed 2025-02-18 by sudeep-mohanty