espressif/esp-idf #14958: Do not disable GPIO interrupt while regestering MCPWM

dizcza, 2024-11-29:
### Is your feature request related to a problem?

I'm always frustrated when an MCPWM capture disables my GPIO interrupts enabled earlier on the same pin.

This is more like a bug than a feature request really.

### Describe the solution you'd like.

This line should be modified https://github.com/espressif/esp-idf/blob/024c6dee5f9d4d99878e71f47e42da2114ad4137/components/driver/mcpwm/mcpwm_cap.c#L293

ESP-IDF v5.1

### Describe alternatives you've considered.

I see the master branch code is different:

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/esp_driver_mcpwm/src/mcpwm_cap.c#L303

Is this issue fixed in v5.2 and so on?

### Additional context.

For now, I re-enable my GPIO interrupt after I enable and start MCPWM capture. This hack requires knowledge of MCPWM's intrinsic realization.

For more info on why I need both MCPWM and GPIO ISR running on the same pin refer to https://github.com/espressif/esp-idf/issues/10327.

suda-morris, 2024-12-02:
Hi @dizcza Master branch and release/v5.4 (also the incoming v5.4 tag release) should have resolved this issue. Please have a try.

This is a common issue that exist in many peripheral drivers, where the GPIO configurations may be overridden by the peripheral drivers. Since v5.4, we have refactored them to only touch the "input/output" enable, and signal connection.

dizcza, 2024-12-02:
Great!

closed 2024-12-02 by dizcza