espressif/esp-idf #14959: ESP-IDF 5.3.1 Bluetooth LE (NimBLE) Pairing problem with Google Pixel 8 pro

andreyvoronko, 2024-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Espressif SoC revision.

ESP32-PICO-MINI-U2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Hello.

When trying to pair via Bluetooth BLE, on the Pixel 8 pro (android 15) phone, a window appears with a password entry and immediately disappears. I do not have time to enter the password. The log shows the connection error code 1028 = 0x404 BLE_SM_ERR_CONFIRM_MISMATCH

On Pixel 6a (android 15), everything works correctly. A window appears with a password entry and it is shown until I enter the password. After that, the connection occurs.

My guess. When pairing, Android 15 starts to sort through known proles, such as 00000 and 123456. But the answer from the ESP does not suit it. Since the android versions are the same, I think the reason is in the driver of the bluetooth chip in the phone.

Best regards

### What is the actual behavior?

-------

### Steps to reproduce.

```
	ble_hs_cfg.sm_io_cap = BLE_SM_IO_CAP_DISP_ONLY;
	ble_hs_cfg.sm_mitm = 1;
	ble_hs_cfg.sm_sc = 1;
	ble_hs_cfg.sm_oob_data_flag = 0; // disable QR authorization
	
	ble_hs_cfg.sm_bonding = 1;
	ble_hs_cfg.sm_our_key_dist |= BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;
	ble_hs_cfg.sm_their_key_dist |= BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;
	// Stores the IRK
	ble_hs_cfg.sm_our_key_dist |= BLE_SM_PAIR_KEY_DIST_ID;
        ble_hs_cfg.sm_their_key_dist |= BLE_SM_PAIR_KEY_DIST_ID;
```

```
case BLE_GAP_EVENT_PASSKEY_ACTION:
	ESP_LOGI(TAG_BLE, "PASSKEY_ACTION_EVENT started");
	struct ble_sm_io pkey = {0};
	int key = 0;

	if (event->passkey.params.action == BLE_SM_IOACT_DISP) {
		pkey.action = event->passkey.params.action;
		pkey.passkey = sd_cnf.ble_psw; // This is the passkey to be entered on peer
		ESP_LOGI(TAG_BLE, "Enter passkey %" PRIu32 " on the peer side", pkey.passkey);
		rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
		ESP_LOGI(TAG_BLE, "ble_sm_inject_io result: %d", rc);
	} else if (event->passkey.params.action == BLE_SM_IOACT_NUMCMP) {
		ESP_LOGI(TAG_BLE, "Accepting passkey..");
		pkey.action = event->passkey.params.action;
		pkey.numcmp_accept = key;
		rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
		ESP_LOGI(TAG_BLE, "ble_sm_inject_io result: %d", rc);
	} else if (event->passkey.params.action == BLE_SM_IOACT_OOB) {
		static uint8_t tem_oob[16] = {0};
		pkey.action = event->passkey.params.action;
		for (int i = 0; i < 16; i++) {
			pkey.oob[i] = tem_oob[i];
		}
		rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
		ESP_LOGI(TAG_BLE, "ble_sm_inject_io result: %d", rc);
	} else if (event->passkey.params.action == BLE_SM_IOACT_INPUT) {
		ESP_LOGI(TAG_BLE, "Input not supported passing -> 123456");
		pkey.action = event->passkey.params.action;
		pkey.passkey = sd_cnf.ble_psw;
		rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
		ESP_LOGI(TAG_BLE, "ble_sm_inject_io result: %d", rc);
	}
	return 0;
```
 
I tried the same thing

```
	ESP_LOGI(TAG_BLE, "PASSKEY_ACTION_EVENT started");
	if (event->passkey.params.action != BLE_SM_IOACT_DISP) {
		return BLE_ATT_ERR_REQ_NOT_SUPPORTED;
	}
	struct ble_sm_io pkey = {
		.action = BLE_SM_IOACT_DISP,
		.passkey = sd_cnf.ble_psw,
	};
	ble_sm_inject_io(event->passkey.conn_handle, &pkey);
```

```
case BLE_GAP_EVENT_ENC_CHANGE:
	/* Encryption has been enabled or disabled for this connection. */
	ESP_LOGI(TAG_BLE, "encryption change event; status:%d ", event->enc_change.status);
	rc = ble_gap_conn_find(event->connect.conn_handle, &desc);
	assert(rc == 0);
	ble_spp_server_print_conn_desc(&desc);
	if(desc.sec_state.authenticated == 0 || desc.sec_state.encrypted == 0 || desc.sec_state.bonded == 0) {
              ESP_LOGI(TAG_BLE, "failed pairing");
              ble_store_util_delete_peer(&desc.peer_id_addr);
              return ble_gap_terminate(event->connect.conn_handle, BLE_ERR_REM_USER_CONN_TERM);
	}
	return 0;
```

I tried the same thing

```
case BLE_GAP_EVENT_ENC_CHANGE:
	/* Encryption has been enabled or disabled for this connection. */
	ESP_LOGI(TAG_BLE, "encryption change event; status:%d ", event->enc_change.status);
	rc = ble_gap_conn_find(event->connect.conn_handle, &desc);
	assert(rc == 0);
	ble_spp_server_print_conn_desc(&desc);
	return 0;
```

### Debug Logs.

```plain
Pixel 6a - coupling is Ok

I (16947) ble: connection established; status=0 
I (16947) ble: conn_itvl:36 conn_latency:0 supervision_timeout:500 encrypted:0 authenticated:0 bonded:0
I (16951) ble: BLE_GAP_EVENT_CONNECT: conn_handle=0
I (16958) ble: Bond found for peer ba:27:a9:91:f2:63, allowing connection
I (16965) NimBLE: GAP procedure initiated: advertise; 
I (16970) NimBLE: disc_mode=2
I (16974) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (16984) NimBLE: 

I (17354) ble: PASSKEY_ACTION_EVENT started
I (17354) ble: Enter passkey 123456 on the peer side
I (17355) ble: ble_sm_inject_io result: 0
I (27879) ble: encryption change event; status:0 
I (27879) ble: conn_itvl:36 conn_latency:0 supervision_timeout:500 encrypted:1 authenticated:1 bonded:1
I (28733) ble: connection updated; status:0 
I (28733) ble: conn_itvl:6 conn_latency:0 supervision_timeout:500 encrypted:1 authenticated:1 bonded:1
I (28965) ble: subscribe event; conn_handle:0 attr_handle:10 reason:1 prevn:0 curn:0 previ:0 curi:1 | a 1 e 1
I (29175) ble: connection updated; status:0 
I (29175) ble: conn_itvl:36 conn_latency:0 supervision_timeout:500 encrypted:1 authenticated:1 bonded:1
I (40297) ble: subscribe event; conn_handle:0 attr_handle:10 reason:2 prevn:0 curn:0 previ:1 curi:0 | a 1 e 1
I (40298) ble: disconnect; reason:531 
I (40302) ble: conn_itvl:36 conn_latency:0 supervision_timeout:500 encrypted:1 authenticated:1 bonded:1
E (40313) ble: error enabling advertisement; rc=2

--------------------

Pixel 8a - Bluetooth doesn't pair

I (45604) ble: connection established; status=0 
I (45605) ble: conn_itvl:39 conn_latency:0 supervision_timeout:500 encrypted:0 authenticated:0 bonded:0
I (45609) ble: BLE_GAP_EVENT_CONNECT: conn_handle=0
I (45615) ble: Bond found for peer e5:ed:38:2d:1b:56, allowing connection
I (45623) NimBLE: GAP procedure initiated: advertise; 
I (45628) NimBLE: disc_mode=2
I (45631) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (45641) NimBLE: 

I (45977) ble: PASSKEY_ACTION_EVENT started
I (45977) ble: Enter passkey 123456 on the peer side
I (45978) ble: ble_sm_inject_io result: 0
I (46853) ble: encryption change event; status:1028 
I (46853) ble: conn_itvl:39 conn_latency:0 supervision_timeout:500 encrypted:0 authenticated:0 bonded:0
I (46858) ble: failed pairing
I (46864) NimBLE: GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19

I (46951) ble: disconnect; reason:534 
I (46951) ble: conn_itvl:39 conn_latency:0 supervision_timeout:500 encrypted:0 authenticated:0 bonded:0
E (46955) ble: error enabling advertisement; rc=2
```

andreyvoronko, 2024-11-29:
Additional information. If I put the phone in Safe mode, the password entry window does not disappear. And you can enter the password. In safe mode, additional services via Bluetooth are disabled on the phone.

ESPAbhinav, 2025-02-19:
Hi,
I followed the provided steps on a Google Pixel 7 running Android 15. However, the issue is not reproducible on this device. Unfortunately, I do not have the exact phone model, Google Pixel 8 Pro, to test further. It seems that other pixel versions with the same Android 15 are working fine . Below is the log for your reference:

[image]

To check further can you please provide detailed debug logs of this issue? Maybe that can help. To enable logs please follow the procedure below in menu config options. Below is the sample screenshot for detailed logs.
1. Component Config  --> Log Output --> Default Log Verbosity --> Debug
2. Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs 

[image]

rahult-github, 2025-03-27:
Hi @andreyvoronko  , any updates ?

ESPAbhinav, 2025-05-08:
Hi @andreyvoronko, I am closing this issue for now. Please feel free to reopen in case of further questions.

closed 2025-05-08 by ESPAbhinav