espressif/esp-idf #14960: "espcoredump.py" not working on v5.3.1 corefile

gamename, 2024-11-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

Some background: I'm dynamically copying core files from my esp32's and saving them externally. Since they are saved as ELF files, I should be able to run `espcoredump.py` on them to get a summary. Via menuconfig, I've set my core files output to be ELF.

But, after I download a corefile to my local host, the `espcoredump.py` utility fails with the error: parsing expected b'\x7fELF' but parsed b'dT\x00\x00'

The value b'\x7fELF' always appears 24 bytes into the file, not at the beginning. I verified the 24-byte padding is also present on the esp32 file before downloading. The `espcoredump.py` utility expects the file to start with b'\x7fELF'.

Is there a work-around, or some other utility I should be using?

igrr, 2024-12-01:
What you are trying to do should work, the utility should account for this additional header before the ELF header. Could you please add an example, how exactly are you invoking espcoredump.py?

gamename, 2024-12-01:
> What you are trying to do should work, the utility should account for this additional header before the ELF header. Could you please add an example, how exactly are you invoking espcoredump.py?

@igrr 

Certainly!  Here is an invocation from directly inside the component dir for 5.3.1:
```bash
./espcoredump.py info_corefile ~/Downloads/core_dump.bin
WARNING: /Users/tennis/Downloads/project_description.json does not exist. Please build the app with "idf.py build"
Traceback (most recent call last):
  File "/Users/tennis/esp/esp-idf-v5.3.1/components/espcoredump/./espcoredump.py", line 77, in <module>
    main()
  File "/Users/tennis/esp/esp-idf-v5.3.1/components/espcoredump/./espcoredump.py", line 62, in main
    temp_core_files = espcoredump.info_corefile()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/esp_coredump/coredump.py", line 540, in info_corefile
    self.exe_elf = ESPCoreDumpElfFile(self.prog)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/elf.py", line 316, in __init__
    super(ESPCoreDumpElfFile, self).__init__(elf_path, _e_type, _e_machine)
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/elf.py", line 119, in __init__
    self.read_elf(elf_path)
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/elf.py", line 130, in read_elf
    header_tables = ElfHeaderTables.parse(elf_bytes)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 404, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 416, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 428, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 2236, in _parse
    subobj = sc._parsereport(stream, context, path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 428, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 2770, in _parse
    return self.subcon._parsereport(stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 428, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 2236, in _parse
    subobj = sc._parsereport(stream, context, path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 428, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 2770, in _parse
    return self.subcon._parsereport(stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 428, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 2236, in _parse
    subobj = sc._parsereport(stream, context, path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 428, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 2770, in _parse
    return self.subcon._parsereport(stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 428, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/tennis/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/construct/core.py", line 2845, in _parse
[... 10 lines cut ...]
==================== ESP32 CORE DUMP START ====================
[... cut ...]
Crashed task handle: 0x3fcc47c0, name: 'process_coredum', GDB name: 'process 1070352320'
Crashed task is not in the interrupt context
Panic reason: abort() was called at PC 0x4037c76f on core 0

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x4037ad4b
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
...etc
```

HTH,
-T

fhrbata, 2024-12-01:
Hello @gamename ,

IIUC the usage of `espcoredump.py` is
```
./espcoredump.py info_corefile --core ~/Downloads/core_dump.bin <app elf>
```
It looks like you are passing the `core file` instead of the application ELF. Please see `espcoredump.py info_corefile --help`. I think this is very nicely explained [here](https://github.com/espressif/esp-idf/issues/7283#issuecomment-882534954).

gamename, 2024-12-01:
> Hello @gamename ,
>
> IIUC the usage of `espcoredump.py` is
>
> ```
> ./espcoredump.py info_corefile --core ~/Downloads/core_dump.bin <app elf>
> ```
>
> It looks like you are passing the `core file` instead of [...]

Hi @fhrbata 

Thank you, but the core file is in ELF format already so it can be interpreted without any additional files.

fhrbata, 2024-12-01:
Hello @gamename ,

> Hi @fhrbata
> 
> Thank you, but the core file is in ELF format already so it can be interpreted without any additional files.

If I understand correctly, the ELF format for a coredump is used merely as a container, meaning it's one of the ways a corefile can be stored. Typically, the corefile records the memory snapshot of a crashed system along with additional information(registers, stack, threads info).  The program ELF is required to decode the coredump.

This is explained in the link I mentioned above:
> 3. Core dump binary is produced by the program running in the chip when a crash happens. This file happens to use ELF as a container format, but it is different from your program ELF file. The core dump binary can be read from Flash and then decoded into a human-readable form by espcoredump.py. In addition to the core file to perform the decoding the application ELF file is also necessary.

HTH

gamename, 2024-12-01:
@fhrbata @igrr 

...snip...
>

If that's the case, then why can one save a decoded summary to the UART at crash time? In other words, there isn't an external ELF file needed in that case. 

Confused.

fhrbata, 2024-12-01:
Perhaps an additional note: The `idf.py coredump-info` command wraps the `espcoredump` tool and automatically adds the program ELF file using info from the `project_description.json` file, which is created during the build process. When using `espcoredump` directly, you must specify it manually. This is why `idf coredump-info --core ~/Downloads/core_dump.bin` functions correctly for you, even if only the core file is provided as an argument.

gamename, 2024-12-01:
> Perhaps an additional note: The `idf.py coredump-info` command wraps the `espcoredump` tool and automatically adds the program ELF file using info from the `project_description.json` file, which is [...]

Yes, I understand that. :) Calling `idf coredump-info` will automatically pull in the needed file.

But again, on the chip at crash time a summary is generated.  No ELF is involved.  How can that happen? 

Replicating that summary generation is my goal.

fhrbata, 2024-12-01:
> Yes, I understand that. :) Calling `idf coredump-info` will automatically pull in the needed file.
>
> But again, on the chip at crash time a summary is generated. No ELF is involved. How can that [...]

Honestly, I'm not very familiar with all the functions and features related to coredump. You can find more information about it at this link: [here](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-guides/core_dump.html).

I think the coredump sent to UART is simply BASE64 encoded content, which is decoded by the esp-idf-monitor and not on the target. https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-guides/core_dump.html#automatic-decoding

gamename, 2024-12-01:
...snip...
> Honestly, I'm not very familiar with all the functions and features related to coredump. You can find more information about it at this link: [...]

Clearly, I'm not very familiar either. ;)

If `esp-idf-monitor` does the crash interpretation, that is a problem.  What happens when a crash happens and you cannot use monitor and there is just a core file? 

It would mean one always has to keep the build environment to interpret any crashes.  That does not scale.  There could be hundreds of esp32 endpoints with many different software releases.  When one of them crashes, it will be necessary to visit lots of build environments to find the one ELF file which allows the crash to be interpreted.  

Or, am I missing something (I hope so)?

fhrbata, 2024-12-01:
> ...snip...
>
>
> Clearly, I'm not very familiar either. ;)
>
> If `esp-idf-monitor` does the crash interpretation, that is a problem. What happens when a crash happens and you cannot use monitor and there [...]

I don't want to mislead you, so I hope @igrr or someone more knowledgeable on this topic will correct me if needed. From what I understand, the app running on the target doesn't have any knowledge about the app's ELF due to size constraints. This means it can't access the app's ELF debug/dwarf information and, therefore, can't decode the coredump. The coredump must be downloaded from the target and analyzed alongside with the app's ELF. The coredump stored in the ELF format includes the app's ELF sha256 or its part, which I believe is intended to be used to identify the correct app's ELF file for decoding the coredump. In other words, keeping the app ELF files around is IMHO necessary.

gamename, 2024-12-01:
> I don't want to mislead you, so I hope @igrr or someone more knowledgeable on this topic will correct me if needed. From what I understand, the app running on the target doesn't have any knowledge [...]

@fhrbata I think you may be right.  It is beginning to look like there is no way to interpret a core file unless you keep the original build ELF around. 

If that is true, how can one determine which build/ELF is correct for a core file? If there are hundreds of endpoints in the environment, that could be a challenge...

igrr, 2024-12-01:
@gamename Typically, the application uploading the core file to the crash analysis service will also provide the information about itself. This may include the application version, which the crash analysis service will use to determine the correct ELF file for the given core file.

You can also find the hash of the application ELF file by calling [esp_app_get_description](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv423esp_app_get_descriptionv) and getting [app_elf_sha256](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv4N14esp_app_desc_t14app_elf_sha256E) member of the resulting structure. If you pass that information to the crash analysis service along with the core file, the service can find the correct ELF file.

Of course, this all requires that you upload the ELF file to the service for every version of the application you deply.

In case you are interested in a complete solution for application diagnostics (including crash analysis), Espressif offers [ESP Insights](https://insights.espressif.com/) service.

cdollar393, 2024-12-01:
@gamename I implemented something similar to what you're going for in another project. As you've seen, you need to save a number of the artifacts from each of your builds in order to decode a core dump after the fact.

You can use esptool.py to get the details of a built firmware binary with the [image_info](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#output-bin-image-details-image-info) command.

```
$ esptool.py image_info --version 2 build/gshub.bin
esptool.py v4.8.1
File size: 1128256 (bytes)
Detected image type: ESP32

ESP32 image header
==================
Image version: 1
Entry point: 0x40081438
Segments: 5
Flash size: 4MB
Flash freq: 40m
Flash mode: DIO

ESP32 extended image header
===========================
WP pin: 0xee (disabled)
Flash pins drive settings: clk_drv: 0x0, q_drv: 0x0, d_drv: 0x0, cs0_drv: 0x0, hd_drv: 0x0, wp_drv: 0x0
Chip ID: 0 (ESP32)
Minimal chip revision: v3.0, (legacy min_rev = 3)
Maximal chip revision: v3.99

Segments information
====================
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      0  0x2f404  0x3f400020  0x00000018  DROM
      1  0x00bec  0x3ffbdb60  0x0002f424  BYTE_ACCESSIBLE, DRAM
      2  0xbfcd8  0x400d0020  0x00030018  IROM
      3  0x054d8  0x3ffbe74c  0x000efcf8  BYTE_ACCESSIBLE, DRAM
      4  0x1e530  0x40080000  0x000f51d8  IRAM

ESP32 image footer
==================
Checksum: 0x16 (valid)
Validation hash: 73b781cbfbd4fcb5ce1ac89d96cfb666527c76f6286db1fcf7fd60f4e11bb603 (valid)

Application information
=======================
Project name: gshub
App version: c6ecf45-dirty
Compile time: Nov 26 2024 14:20:35
ELF file SHA256: c4c9592b0f0ca31a96d40156ebe26ba5359bb29c7344813ec280c8793f3c59e4
ESP-IDF: v5.3.1
Secure version: 0
```

Note that the above command is run against the firmware .bin file, not the .elf file which you'll ultimately use to decode the core dump.

In my particular project I wrote a shell script that ran a few commands to get build info for artifacts, and then copied the artifacts to some storage location along with the extracted info like the ELF SHA256. I have CMake run the script for me on each build with a snippet like this is the main CMakeLists.txt just after the `project()` section:

```
# copy built files to local dist directory
# see: https://github.com/espressif/esp-idf/issues/9885#issuecomment-1266791709
add_custom_command(TARGET app POST_BUILD
    COMMAND ${CMAKE_COMMAND} -E echo "Running copy-local-build.sh"
    COMMAND ${CMAKE_CURRENT_LIST_DIR}/copy-local-build.sh ${CMAKE_BINARY_DIR}
)
```

Then on the ESP32 I used the following method to check for and grab the core dump from flash, and then upload it via MQTT (no base64 conversion - just binary data). As part of the upload process I do similar to what @igrr described, and read the crashed app SHA256 from the core dump. Then I use that as part of the MQTT topic path so I know which application ELF to use when I go to decode the core dump.

```
esp_err_t clIotUploadCoreDump(const char *deviceId, const char *runningFwVersion) {
#if CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH
[... 59 lines cut ...]
    // now that it is published we need to erase the old core dump
    if(esp_core_dump_image_erase() != ESP_OK) {
      ESP_LOGE(TAG, "Failed to erase core dump partition");
    }
    else {
      ESP_LOGI(TAG, "Erased core dump partition");
    }

    // post message about core dump upload done
    if(esp_event_post(CL_EVENT, CL_IOT_COREDUMP_DONE_EVENT, NULL, 0, pdMS_TO_TICKS(CL_EVENT_POST_TIMEOUT_MS)) != ESP_OK) {
      ESP_LOGW(TAG, "Failed to post core dump done event");
    }
  }

freeSummary:
  if(summary)
    free(summary);
  return ret;
#else
  return ESP_OK;
#endif
}
```

Hopefully this helps a bit if you go with a homemade solution.

gamename, 2024-12-01:
@igrr Thanks! That's great. I'll go ahead and close this.

igrr, 2024-12-02:
@gamename I think you meant to tag @cdollar393 :)

gamename, 2024-12-02:
> @gamename I think you meant to tag @cdollar393 :)

lol.  Yes, my mistake.  Mr @cdollar393 could you share the content of `copy-local-build.sh`?

gamename, 2024-12-03:
Reopened due to a confusing point in the description(s) above concerning the sha256 values.

Here is the partial output of `esptool.py image_info --version 2 build/firmware.bin`:
```
...snip...
ELF file SHA256: 5984ad7a56624d3e8833ec47c7be3a0828271cc9ec46f98777e8c713ccf72301
                 ^^^^^^^^^
```
Here is the code used to print the summary of the crash file and get the sha256 value:
```c
    esp_core_dump_summary_t *summary = malloc(sizeof(esp_core_dump_summary_t));
    if (summary) {
        if (esp_core_dump_get_summary(summary) == ESP_OK) {
            ESP_LOGW(TAG, "Core dump SHA-256: %s", summary->app_elf_sha256);
        }
    }
```

And this is the output:
```
W (08:24:14.036) CRASH_FILE_MANAGER: Core dump SHA-256: 5984ad7a5
                                                        ^^^^^^^^^
```

Why is the core dump sha256 value only the first 9 chars of the full checksum?  Is that still usable?

KonstantinKondrashov, 2024-12-03:
Hi @gamename!
The len of the sha256 value is defined by CONFIG_APP_RETRIEVE_LEN_ELF_SHA (by default is 9). The 9 characters are enough to identify the elf file which is used. You can increase it in the menuconfig if wish.
IIRK, the sha256 of elf is read somewhere at the startup and stored in an array and prints it when a crash happens.
Thanks.

gamename, 2024-12-03:
> Hi @gamename! The len of the sha256 value is defined by CONFIG_APP_RETRIEVE_LEN_ELF_SHA (by default is 9). The 9 characters are enough to identify the elf file which is used. You can increase it in [...]

Ah, ok. Thanks!

closed 2024-12-03 by gamename