espressif/esp-idf #14962: MDNS problems

pavel808, 2024-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to get mdns working for my project. I am referencing the following documentation : 

https://docs.espressif.com/projects/esp-protocols/mdns/docs/latest/en/index.html

About my project. It is an application which runs on an ESP32-S3 on an ESP Thread Borer Router board. 
The application has a HTTP server running which allows clients like CURL to communicate with it over Wi-Fi. 

Obviously having a hostname assigned would be the best so that the HTTP client doesn't have to remember the local IP address every time. I am trying to set the hostname as "server".

Here is my below code in my` app_main.cpp`and also some of the terminal output when my application starts. 

I start the mdns service before the HTTP server. 
From the logs it appears to set the hostname ok. Regardless, it is not possible to ping the hostname from anywhere on the same local network. 

I also notice that on my router, I can see that hostname "espressif" gets assigned, via the sdkconfig file : `CONFIG_LWIP_LOCAL_HOSTNAME="espressif"`

Any help with this would be greatly appreciated. Thanks.

```
#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <esp_matter.h>
#include <esp_matter_console.h>
#include <esp_matter_controller_client.h>
#include <esp_matter_controller_console.h>
#include <esp_matter_controller_utils.h>
#include <esp_matter_ota.h>
#include <esp_http_server.h>
#if CONFIG_OPENTHREAD_BORDER_ROUTER
#include <esp_openthread_border_router.h>
#include <esp_openthread_lock.h>
#include <esp_ot_config.h>
#include <esp_spiffs.h>
#include <platform/ESP32/OpenthreadLauncher.h>
#endif // CONFIG_OPENTHREAD_BORDER_ROUTER
#include <app_reset.h>
#include <common_macros.h>

#include <app/server/Server.h>
#include <credentials/FabricTable.h>

#include "protocol_examples_common.h"

#include "lwip/err.h"
#include "lwip/sys.h"

#include "mdns.h"

static const char *TAG = "app_main";
uint16_t switch_endpoint_id = 0;

using namespace esp_matter;
using namespace esp_matter::attribute;
using namespace esp_matter::endpoint;

extern void initialize_device_controller();

httpd_handle_t start_webserver();

static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
[... 139 lines cut ...]
I (2400) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=0 cp=0
I (2400) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=0000/00 cm=0 cp=0
I (2400) app_main: MDNS Init OK
I (2400) app_main: MDNS mdns_hostname_set ok
I (2400) app_main: MDNS instance name set ok
I (2400) app_main: Calling start_webserver()
I (2400) web_server: Starting server on port: '80'
I (2410) web_server: Registering URI handlers
W (2410) httpd_uri: httpd_register_uri_handler: no slots left for registering handler
W (2410) httpd_uri: httpd_register_uri_handler: no slots left for registering handler
W (2410) httpd_uri: httpd_register_uri_handler: no slots left for registering handler
W (2410) httpd_uri: httpd_register_uri_handler: no slots left for registering handler
I (3300) chip[DL]: Posting ESPSystemEvent: Wifi Event with eventId : 5
I (3310) web_server: Stopping webserver
E (3310) httpd: Failed to send shutdown signal err=-1
E (3310) web_server: Failed to stop http server
I(27270) OPENTHREAD:[N] Mle-----------: RLOC16 2c00 -> fffe
I(27630) OPENTHREAD:[N] Mle-----------: Attach attempt 1, AnyPartition reattaching with Active Dataset
I(34220) OPENTHREAD:[N] RouterTable---: Allocate router id 11
I(34240) OPENTHREAD:[N] Mle-----------: RLOC16 fffe -> 2c00
I(34250) OPENTHREAD:[N] Mle-----------: Role detached -> leader
I(34260) OPENTHREAD:[N] Mle-----------: Partition ID 0x262e9573
I (34300) OPENTHREAD: Platform UDP bound to port 49153
I (121610) chip[DL]: Posting ESPSystemEvent: IP Event with eventId : 1
```

nopnop2002, 2024-12-11:
You can check the mDNS status with the command below.
```
$ avahi-browse --all
+ enp2s0 IPv6 VOYAGER                                       Microsoft Windows Network local
+ enp2s0 IPv4 VOYAGER                                       Microsoft Windows Network local
+ enp2s0 IPv4 landisk-HDL-AA1(smb)                          Microsoft Windows Network local
+ enp2s0 IPv6 VOYAGER                                       _device-info._tcp    local
+ enp2s0 IPv4 VOYAGER                                       _device-info._tcp    local
+ enp2s0 IPv4 esp32-server                                  Web Site             local
+ enp2s0 IPv4 landisk-HDL-AA1                               Web Site             local
+ enp2s0 IPv4 landisk-HDL-AA1 [34:76:c5:71:e0:6d]           Workstation          local

$ avahi-resolve --name esp32-server.local
esp32-server.local      192.168.10.155

```

pavel808, 2024-12-11:
@nopnop2002 Thanks for your response. Unfortunately with that command I still don't  see my ESP device. Only local stuff from the Ubuntu computer where the command is issued from. 

```
avahi-browse --all
+ wlp0s20f3 IPv6 Latitude-3420                                 SSH Remote Terminal  ubuntu-server.local
+ wlp0s20f3 IPv4 Latitude-3420                                 SSH Remote Terminal  ubuntu-server.local
+ wlp0s20f3 IPv6 Pi-hole DNS on Latitude-3420                  DNS Server           ubuntu-server.local
+ wlp0s20f3 IPv4 Pi-hole DNS on Latitude-3420                  DNS Server           ubuntu-server.local
+ wlp0s20f3 IPv6 Pi-hole DNS on Latitude-3420                  _domain._tcp         ubuntu-server.local
+ wlp0s20f3 IPv4 Pi-hole DNS on Latitude-3420                  _domain._tcp         ubuntu-server.local
+ wlp0s20f3 IPv6 Pi-hole DNS on Latitude-3420                  Web Site             ubuntu-server.local
+ wlp0s20f3 IPv4 Pi-hole DNS on Latitude-3420                  Web Site             ubuntu-server.local
+ wlp0s20f3 IPv6 Nextcloud on Latitude-3420                    Secure Web Site      ubuntu-server.local
+ wlp0s20f3 IPv4 Nextcloud on Latitude-3420                    Secure Web Site      ubuntu-server.local
+ wlp0s20f3 IPv6 Location of Latitude-3420                     Apple Home Sharing   ubuntu-server.local
+ wlp0s20f3 IPv4 Location of Latitude-3420                     Apple Home Sharing   ubuntu-server.local

```

nopnop2002, 2024-12-11:
Can you stop this local DNS server?
```
+ wlp0s20f3 IPv6 Pi-hole DNS on Latitude-3420                  DNS Server           ubuntu-server.local
+ wlp0s20f3 IPv4 Pi-hole DNS on Latitude-3420                  DNS Server           ubuntu-server.local
```

nopnop2002, 2024-12-12:
The following is not required.
```
    ESP_LOGI(TAG, "MDNS mdns_hostname_set ok");
    //set default instance
    mdns_instance_name_set("esp-server");
    if (err != ESP_OK)
    {
	 ESP_LOGE(TAG, "MDNS instance name set fail : %s", esp_err_to_name(err));
	 return;
    }

    ESP_LOGI(TAG, "MDNS instance name set ok");
```

Now I think you can ping server.local.

pavel808, 2024-12-12:
@nopnop2002 I have tried both of these but unfortunately I still get `Name or service not known `when trying to ping it.

nopnop2002, 2024-12-13:
A simple sample is published here.
https://github.com/espressif/esp-protocols/tree/master/components/mdns/examples/query_advertise

When you run it, it will respond to ping.
```
$ LANG=C ping esp32-mdns.local
PING esp32-mdns.local (192.168.10.111) 56(84) bytes of data.
64 bytes from 192.168.10.111 (192.168.10.111): icmp_seq=1 ttl=64 time=93.3 ms
64 bytes from 192.168.10.111 (192.168.10.111): icmp_seq=2 ttl=64 time=93.5 ms
64 bytes from 192.168.10.111 (192.168.10.111): icmp_seq=3 ttl=64 time=14.8 ms
64 bytes from 192.168.10.111 (192.168.10.111): icmp_seq=4 ttl=64 time=36.1 ms
64 bytes from 192.168.10.111 (192.168.10.111): icmp_seq=5 ttl=64 time=58.2 ms
```

pavel808, 2024-12-16:
@nopnop2002 I flashed that sample, and unfortunately I can't ping the mdns name still from my Ubuntu computer (Only the IP address directly). 

I even uninstalled and re-installed avahi daemon and utils. 

Checking `avahi-browse -all`: 

```
avahi-browse -all
+ wlp0s20f3 IPv6 192-168-178-1                                 Microsoft Windows Network local
+ wlp0s20f3 IPv6 fritz-box                                     Microsoft Windows Network local
+ wlp0s20f3 IPv4 192-168-178-1                                 Microsoft Windows Network local
+ wlp0s20f3 IPv4 fritz-box                                     Microsoft Windows Network local
+ wlp0s20f3 IPv6 ESP32-WebServer1                              Web Site             local
+ wlp0s20f3 IPv6 ESP32-WebServer                               Web Site             local
+ wlp0s20f3 IPv4 ESP32-WebServer1                              Web Site             local
+ wlp0s20f3 IPv4 ESP32-WebServer                               Web Site             local
```

Now `avahi-resolve --name esp32-mdns.local `: 

```
avahi-resolve --name esp32-mdns.local
esp32-mdns.local	192.168.178.71
```

So it does resolve, but still can't ping it:) Frustrating. 

Do I need to modify anything in the `/etc/avahi/avahi-daemon.conf` file ? I have a copy of it attached here. Thanks. 

[avahi-daemon.conf-copy.txt](https://github.com/user-attachments/files/18156110/avahi-daemon.conf-copy.txt)

nopnop2002, 2024-12-16:
>Do I need to modify anything in the /etc/avahi/avahi-daemon.conf file ?

No

nopnop2002, 2024-12-17:
From my experience, if you ping using the IP address once and then pinging using the mDNS host name, it may work correctly.

I think it's probably a cache issue with WiFi router.

```
ping 192.168.178.71
ping esp32-mdns.local
```

david-cermak, 2025-01-10:
Closing as it as resolved and unrelated to IDF code. Thanks to @pavel808 and @nopnop2002 for the investigation, testing,  and providing guidance.

closed 2025-01-10 by david-cermak