espressif/esp-idf #14963: "phy_init: saving new calibration data because of checksum failure, mode(0)" on almost every start

kriegste, 2024-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Under what circumstances does this message appear at startup?

> phy_init: saving new calibration data because of checksum failure, mode(0)

I am asking because I get this on almost every start. Also software restart or hardware restart. I have several devices, with very different power supplies. All are affected. This may be since IDF 5.3.1. But I could be mistaken and it only caught my eye now. Experiments show that "mode(n)" refer to whether "full" or "partial" calibration is selected. However, this does not make a difference in the frequency of this message appearing.
I cannot believe this is really a "checksum failure". That would mean reading the NVS fails. But my WiFi credentials (SSID and password) which are also in the NVS, are never misread. The WiFi connection succeeds every time regardless if the message appears or not.
Any insights or ways to debug?

AxelLin, 2024-12-02:
I also reported the same symptom with v5.2 branch: https://github.com/espressif/esp-idf/issues/13251#issuecomment-2443023985, but got no response.

@mhdong

AxelLin, 2024-12-04:
@mhdong 
Is it ok writing the calibration data so frequently? https://github.com/espressif/esp-idf/issues/14963#issuecomment-2512793421

mhdong, 2024-12-04:
hi @MaxwellAlan @kriegste 
Considering the frequency of software restarts, hardware restarts, the impact on the chip's flash wear is acceptable.

closed 2024-12-20 by espressif-bot

AxelLin, 2024-12-20:
@mhdong 

I don't get it, this commit does not fix anything.
See the original report: "I am asking because I get this on almost every start."
Which means it's unlikely to be "the calibration data is outdated" case.
So the question is: Is it normal that hitting "data checksum failure" so frequently?

Also, you should have different return value for "data checksum failure" and "the calibration data is outdated" so the users can figure out the root cause. (we don't need to **guess** the root cause.)

AxelLin, 2024-12-20:
> hi @kriegste Current output "phy_init: saving new calibration data because of checksum failure, mode(0)" is not entirely accurate. It should differentiate between checksum errors and calibration updates. In the future, we plan to revise the description here or use more precise return values to distinguish between checksum errors and updates.

But you even didn't figure out if it's "checksum errors" or "calibration updates" for the reported case. 
What is the time period for calibration updates?

kriegste, 2024-12-20:
If there is indeed a checksum failure it should still be logged as a warning. Don't you agree, Espressif?

mhdong, 2024-12-23:
For checksum errors or calibration data being outdated, these simply indicate that we need to update calibration data. We don't need to handle them differently based on the specific return value, so there's no need for further differentiation.

AxelLin, 2024-12-23:
> For checksum errors or calibration data being outdated, these simply indicate that we need to update calibration data. We don't need to handle them differently based on the specific return value, so there's no need for further differentiation.

If the user cannot differentate it when reprot the issue, how could you
know if it could be a bug (checksum error) or expected behavior (expired)  from a user reported log?
What is the time interval for calibration data expiration?

mhdong, 2024-12-23:
hi @AxelLin 
"Calibration data being outdated" has nothing to do with time. It simply means that the stored calibration information no longer matches the current environment.

AxelLin, 2024-12-24:
> hi @AxelLin "Calibration data being outdated" has nothing to do with time. It simply means that the stored calibration information no longer matches the current environment.

Which means ther is no way to differentiate "Calibration data being outdated"
and "Checksum errors" from a user's report.
The point is not about if you need to handle them differently.
The point is there is no way to check if "Checksum errors" happens (which should not happen).

zggcd157, 2025-04-15:
复位就好了，因为你之前擦除了NVS里保存天线(RF)的校准数据或者缩小了NVS导致RRF校验数据消失导致的