espressif/esp-idf #14964: ESP32 as BTDM controller over HCI: occasional hardware error

danergo, 2024-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Latest

### Espressif SoC revision.

NodeMCU-ESP-32S

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

NodeMCU-ESP-32S

### Power Supply used.

External 5V

### What is the expected behavior?

Stable operation

### What is the actual behavior?

During our stress test on a RPi which uses ESP32 as a Bluetooth co-processor (attached via HCI), we saw some occasional "Hardware Error" in the HCI logs.

HCI logs are already shared.

### Steps to reproduce.

BLE Ad scanning with filtering OFF, and occasionally GATT connect, also occasionally BT Classic connect.

esp-zhp, 2024-12-02:
I’m not quite clear about your question. At **line 423920** and **line 443764** in the HCI log, those are ADV reports. What is their direct connection to the "Hardware Error" message in the logs? Could you please describe your question in more detail to help us understand and investigate further?
[image]
[image]

Additionally, I recommend turning off the HCI log during your tests. Excessive logging might introduce unexpected performance issues. HCI logs are typically used only for debugging purposes.

Does the issue still occur after disabling the HCI log? If it does, please share more details, such as the frequency of the issue or any specific test scenarios where it happens.

danergo, 2024-12-02:
Dear @esp-zhp:

Our long-time stress test process is still running, hence we didn't (yet) turned off any logging on the ESP side. We wish to keep this running for a couple of weeks, to make it extra sure about the solution for the lost bytes.

Now back to this Hardware issue, in btmon_v612.pcap, we see this line: "Hardware Error". Previous messages were LE Meta ads, and after this Hardware error, host issued a Reset:

[image]

Situation is very same with both 423920 and 443764.

Please note we saw this in "btmon_v612.pcap" log (local btmon log), and not the ESP log.

danergo, 2024-12-06:
Dear @esp-zhp: do you have any updates on this topic?

We started a new test, and now disabled all HCI logs on the ESP side, to see how far it will reach. Also we did turn on the hardware ad filtering based on peer address and ad content.

Will let you know our results.

esp-zhp, 2024-12-09:
Thank you for the update. I’m waiting to see the results of your tests after disabling the HCI logs. Please let me know if the same issue still occurs under these new conditions.  
Looking forward to your findings!

danergo, 2025-01-06:
Dear @esp-zhp: it seems no issues so far.

We will do another test for 7 days, and in case we won't notice any errors, will report back here to close this one.

Thank you.
Happy New Year!

danergo, 2025-01-14:
No issues so far! Closing this now.

Many thanks!

closed 2025-01-14 by danergo