espressif/esp-idf #14968: Does esp-idf supports usb over ip to flash or monitor?

TroyMitchell911, 2024-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a serve and laptop. The esp32  is plugged into my laptop. And I share the usb port via ip to the server. I can use `minicom` to check esp32 log.

But when I try to use `idf.py flash -p /dev/ttyACM0` I got error. `idf.py monitor -p /dev/ttyACM0` same.

```bash
❯ idf.py monitor -p /dev/ttyACM0                                                                     with troy@troy-server at  17:19:11
Usage: idf.py monitor [OPTIONS]
Try 'idf.py monitor --help' for help.

Error: Invalid value for '-p' / '--port': Path '/dev/ttyACM0' is not readable.
❯ idf.py flash -p /dev/ttyACM0                                                                       with troy@troy-server at  17:21:44
Usage: idf.py flash [OPTIONS]
Try 'idf.py flash --help' for help.

Error: Invalid value for '-p' / '--port': Path '/dev/ttyACM0' is not readable.
❯ lsusb | grep Esp                                                                                   with troy@troy-server at  17:21:49
Bus 005 Device 009: ID 303a:1001 Espressif USB JTAG/serial debug unit
❯ ls /dev/ttyACM0                                                                                    with troy@troy-server at  17:21:52
/dev/ttyACM0

```

tore-espressif, 2024-12-02:
Here's guide to access remote ports https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/remote-serial-ports.html

Would that solve your problem?

TroyMitchell911, 2024-12-06:
> Here's guide to access remote ports https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/remote-serial-ports.html
> 
> Would that solve your problem?

I haven't tried it. But why I can not use usbip to share the usb port?

TroyMitchell911, 2024-12-06:
I can now use `RFC2217` to flash and `usbip` to share the USB port, then use minicom to connect. 
However, this process feels cumbersome. 
Is there a way to streamline and combine all these functions into a single, more efficient solution.

igrr, 2024-12-06:
Can you try `idf.py monitor` with RFC2217 as well, does it still produce that "connection failed..." error?

Something along the lines of:

```shell
idf.py -p rfc2217://100.72.195.98:4000 flash monitor

# or
export ESPPORT=rfc2217://100.72.195.98:4000
idf.py flash monitor
```

TroyMitchell911, 2024-12-19:
> Can you try `idf.py monitor` with RFC2217 as well, does it still produce that "connection failed..." error?
> 
> Something along the lines of:
> 
> ```shell
> idf.py -p rfc2217://100.72.195.98:4000 flash monitor
> 
> # or
> export ESPPORT=rfc2217://100.72.195.98:4000
> idf.py flash monitor
> ```
still produce that "connection failed..." error

peterdragun, 2025-01-02:
Hi @TroyMitchell911, can you please try using the port with the `?ign_set_control` suffix as mentioned in the docs? I suggest wrapping the port in quotes to ensure it does not get interpreted in an unexpected way.
In your case, it would look something like this: `"rfc2217://100.72.195.98:4000?ign_set_control"`

I can reproduce the same issue when I skip the suffix, but it works without any problems if I include it. This option is needed to allow setting DTR and RTS which are used for resetting the chip.
Let me know if that helps, thank you!

dobairoland, 2025-01-13:
I'm closing this for inactivity. I think we have discussed this adequately. Please ping me if this is still an issue and we can re-open it.

closed 2025-01-13 by dobairoland