espressif/esp-idf #14971: Support dynamic low power mode

xobs, 2024-12-03:
### Is your feature request related to a problem?

I have a design that can run off a variable input power supply. It can take anywhere from 1.8 to 5V. When running at lower voltages, the current draw is higher. As a result, the system is more prone to browning out.

### Describe the solution you'd like.

I would like to be able to measure the voltage prior to turning on the PA and to adjust the transmit power depending on whether there is enough headroom.

### Describe alternatives you've considered.

Currently I run with a TX power of 10, because this is the worst case. This is not ideal because it means I can't have higher power when more voltage is available.

### Additional context.

There is a function that dials down the PHY power if a brownout occurs, but I would like to dial back power before a brownout reset occurs.

I can manually adjust the PHY data prior to starting wifi, but it seems to be a poor idea to snoop internal data structures that way.

BitsForPeople, 2024-12-03:
May [esp_wifi_set_max_tx_power()](https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t) be what you're looking for?

xobs, 2024-12-05:
It's unclear what that function does, and it looks like it's in `libnet80211.a` so I haven't investigated what it actually does.

The documentation for `ESP_PHY_MAX_WIFI_TX_POWER` says the minimum power is 10 dBm, but that function looks like it'll allow as low as 2 dBm.

That function also indicates that it won't run without wifi having been started.

It could be that I could set my device to transmit on the "lowest" power (10 dBm) and then adjust it lower or higher using that function, but the fact that it requires wifi to already be running is a problem.

BitsForPeople, 2024-12-06:
The actual problem may be to find out if and by how much you can dynamically  lower the WiFi TX power without losing connection.

esp-lis, 2024-12-30:
@xobs Refer to the following diagram for the power supply requirements of the ESP32-S3. An excessively low or high supply voltage may cause the chip to operate in an unexpected state.
[image]

xobs, 2024-12-30:
@esp-lis that's correct. The issue I'm running into is the power draw is too high, particularly during PA initialization, and so I need to back it down.

My power supply generates 3.3V, but during initialization the current can spike which causes the voltage to dip, particularly if the upstream power supply can't keep up. One solution is to add a lot more capacitance, but I don't have space on the PCB to do that.

Another solution is to reduce the transmit power. There is [esp_wifi_set_max_tx_power()](https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t) but it only takes effect after calibration, which would already cause the brownout.

I can configure the overall maximum transmit power to be lower as a compile time option, but it's not clear if there is a way to increase the power after starting it up. Particularly since it operates by overwriting the table in a private function.

xobs, 2024-12-30:
There may also be some confusion from the initial issue report.

When I say that it can run from 1.8V - 5V, I'm referring to the input of the whole project. There is a buck-boost that converts it to ~3.4V which is within the acceptable range of the target. The PA is beefy enough that when running at lower voltages the buck-boost can't keep up.

esp-lis, 2025-02-26:
> [@esp-lis](https://github.com/esp-lis) that's correct. The issue I'm running into is the power draw is too high, particularly during PA initialization, and so I need to back it down.
>
> My power [...]

My local test results align with your description. The maximum Tx power that can be configured at runtime via the [`esp_wifi_set_max_tx_power()`](https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t) API is constrained by the value of the **ESP_PHY_MAX_WIFI_TX_POWER** option. When **ESP_PHY_MAX_WIFI_TX_POWER** is set to 10 dBm, it means that the maximum Tx power that can be adjusted through [`esp_wifi_set_max_tx_power()`](https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t) during application runtime is limited to 10 dBm.

I have summarized your two requirements as follows:

- During PA initialization, you want the Tx power to be as low as possible to mitigate the voltage drop caused by PA power consumption.
- After PA initialization is complete, you want to increase the Tx power.

I am aware of another approach that could achieve a similar effect, which you may want to evaluate and test. Below is a brief description of this approach for your reference:
The current **esp-idf** supports using different Tx power levels for different Tx rates. I understand that the PA initialization process occurs during the WiFi station's connection to the AP, and during this process, the Tx rate is typically 1 Mbps. Would adjusting the Tx power for the 1 Mbps rate to a lower value while keeping the Tx power for other rates higher help address your issue?

[image]

xobs, 2025-02-27:
Interesting. I hadn't heard of differing power rates. In such a scenario, 1 Mbps would be permanently on the lowest power, while other rates would be free to increase? Is there any documentation on how to do that, or is it a new feature?

esp-lis, 2025-02-28:
> Interesting. I hadn't heard of differing power rates. In such a scenario, 1 Mbps would be permanently on the lowest power, while other rates would be free to increase? Is there any documentation on how to do that, or is it a new feature?

@xobs you can reference [RF_calibration.html#phy-initialization-data](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/RF_calibration.html#phy-initialization-data).

esp-lis, 2025-03-07:
@xobs If there are no further questions, I will close this task for now. You can reopen it if you have any questions later. Thank you!

closed 2025-03-07 by Alvin1Zhang