espressif/esp-idf #14972: Continuous ADC stops working while BLE Mesh is on

UncleBigLu, 2024-12-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-4375-gf420609c33

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

The `adc_continuous_evt_cbs_t::on_conv_done` events keeps triggering when data available.

### What is the actual behavior?

If I enable a task that periodically send BLE message via `esp_ble_mesh_client_model_send_msg`, the adc stops working after around 10 seconds.

### Steps to reproduce.

/* main.c - Application main entry point */

/*
 * SPDX-FileCopyrightText: 2017 Intel Corporation
 * SPDX-FileContributor: 2018-2021 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include "esp_err.h"
#include "freertos/idf_additions.h"
#include "mesh/access.h"
#include "portmacro.h"
#include <stdint.h>
#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "esp_log.h"
#include "nvs_flash.h"

#include "esp_ble_mesh_common_api.h"
#include "esp_ble_mesh_provisioning_api.h"
#include "esp_ble_mesh_networking_api.h"
#include "esp_ble_mesh_config_model_api.h"
#include "esp_ble_mesh_generic_model_api.h"
[... 645 lines cut ...]
    err = ble_mesh_nvs_open(&NVS_HANDLE);
    if (err) {
        return;
    }

    ble_mesh_get_dev_uuid(dev_uuid);

    /* Initialize the Bluetooth Mesh Subsystem */
    err = ble_mesh_init();
    if (err) {
        ESP_LOGE(TAG, "Bluetooth mesh init failed (err %d)", err);
        return;
    }

    if (store.app_idx != ESP_BLE_MESH_KEY_UNUSED) {
        ESP_ERROR_CHECK(esp_ble_mesh_model_subscribe_group_addr(
            esp_ble_mesh_get_primary_element_address(), CID_ESP,
            ESP_BLE_MESH_VND_MODEL_ID_SERVER, RX_GROUP_ADDR));
        ESP_LOGI(TAG, "Start broadcast task");
        xTaskCreate(broadcast_task, "broadcast_task", 4096, NULL, 3, NULL);
    }

    ESP_LOGI(TAG, "Start ADC sampling and perform FFT");
    xTaskCreate(adc_fft_task, "adc_fft_task", 8192, NULL, 4, NULL);
}

Haerteleric, 2024-12-04:
Faced the same problem. I Think you have to move the ADC Funcs into IRAM. There is a Menuconfig option for that:

```
CONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE=y
```

UncleBigLu, 2024-12-06:
> CONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE=y

Well It works! Thanks Haerteleric :)

closed 2024-12-06 by UncleBigLu