espressif/esp-idf #14973: Heap memory leak when using gpio_install_isr_service and gpio_uninstall_isr_service

MmdMi, 2024-12-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Title:** Heap memory leak when using `gpio_install_isr_service` and `gpio_uninstall_isr_service`

**Description:**  
I encountered a heap memory leak issue while testing the `gpio_install_isr_service` and `gpio_uninstall_isr_service` functions in my ESP-IDF project. When these functions are called, 16 bytes of heap memory remain unreleased, causing a memory leak.  

**Steps to reproduce:**  
1. Call `gpio_install_isr_service` with default flags.  
2. Call `gpio_uninstall_isr_service`.  
3. Compare the heap size before and after using the following code snippet:  

```c
uint32_t start_heap = esp_get_free_heap_size();
ESP_ERROR_CHECK(gpio_install_isr_service(0));
ESP_ERROR_CHECK(gpio_uninstall_isr_service());
uint32_t end_heap = esp_get_free_heap_size();
ESP_LOGI(TAG, "Diff[%ld]", (start_heap - end_heap));
```

**Observed behavior:**  
The output of the above code is:  

```
I (459) test_heap: Diff[16]
```

**Expected behavior:**  
After calling `gpio_uninstall_isr_service`, the heap memory should be fully released, and the difference should be `0`.  

**ESP-IDF version:**  
- ESP-IDF v5.3.1  

**Environment:**  
- ESP32-WROOM-32 module  
- Platform: Ubuntu 24.04 

**Additional notes:**  
This issue might accumulate over time if the `gpio_install_isr_service` and `gpio_uninstall_isr_service` functions are used repeatedly in a long-running application, leading to eventual heap exhaustion.

ESP-Marius, 2024-12-04:
The interrupt allocator will lazy allocate some descriptors, and these will be re-used when interrupts are freed.

So these 16 bytes you see here will be used again when someone allocates a new interrupt, repeatably calling  `gpio_install_isr_service` will not cause issues or heap exhaustion.

Alvin1Zhang, 2024-12-26:
@MmdMi Thanks for reporting, can you please help share if any further updates for the issue? Thanks.

closed 2024-12-27 by ESP-Marius