espressif/esp-idf #14974: ESP-IDF ESP32s3 Project head driver/rmt_tx.h not found

emofalling, 2024-12-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

The cmake can find head file `"driver/rmt_tx.h"`, then make my project successfully.

### What is the actual behavior?

The cmake can't find head file `"driver/rmt_tx.h"`, and throwing an error:
```
In file included from I:/Projects/esp32-s3zhukongyaokongche/fasheduan/main/drivers/ws2812.c:1:
I:/Projects/esp32-s3zhukongyaokongche/fasheduan/main/drivers/ws2812.h:7:10: fatal error: driver/rmt_tx.h: No such file or directory
    7 | #include "driver/rmt_tx.h"
      |          ^~~~~~~~~~~~~~~~~
```
but in the same project on esp32, cmake can find this head file.

### Steps to reproduce.

1. I copy the `drivers/ws2812.c` from other project(esp32), and copy to this project(esp32s3)
2. change CMakeLists.txt to:
```
idf_component_register(SRCS "main.c" "drivers/ws2812.c"
                       INCLUDE_DIRS "."
                       #REQUIRES 
                       PRIV_REQUIRES nvs_flash esp_wifi)

```
3. then click:
[image]
4. throw this error.

### Build or installation Logs.

```plain
[0/1] Re-running CMake...-- ccache will be used for faster recompilation
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file I:/Projects/esp32-s3zhukongyaokongche/fasheduan/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- App "fasheduan" version: 1
-- Adding linker script I:/Projects/esp32-s3zhukongyaokongche/fasheduan/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script I:/Projects/esp32-s3zhukongyaokongche/fasheduan/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script I:/ESP-IDF/v5.2.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script I:/ESP-IDF/v5.2.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script I:/ESP-IDF/v5.2.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script I:/ESP-IDF/v5.2.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script I:/ESP-IDF/v5.2.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script I:/ESP-IDF/v5.2.2/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
[... 20 lines cut ...]
ninja: build stopped: subcommand failed.
```

### More Information.

But the other lib on esp32s3 can find header file, such as:
```c
// Standard
#include <stdio.h>
#include <string.h>
// esp
#include "esp_task_wdt.h"
#include "esp_log.h"
// RGB light
//#include "drivers/ws2812.h"
// FreeRTOS
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
// NVS
#include "nvs.h"
#include "nvs_flash.h"
// WiFi
#include "esp_wifi.h"
#include "esp_mac.h"
```

Dazza0, 2024-12-03:
@emofalling 

In the `ws2812.c` component's `CMakeLists.txt`, does the following resolve the build error?

```cmake
idf_component_register(SRCS "main.c" "drivers/ws2812.c"
                       INCLUDE_DIRS "."
                       REQUIRES driver
                       PRIV_REQUIRES nvs_flash esp_wifi)
```

emofalling, 2024-12-04:
@Dazza0 
Thank your help. I successfully resolved the build error.

suda-morris, 2024-12-05:
Glad you resolved the issue. @emofalling 

BTW, WRT the ws2812 driver, Espressif officially support this component: https://components.espressif.com/components/espressif/led_strip/

Feel free to give it a try :)

closed 2024-12-05 by suda-morris

emofalling, 2024-12-06:
@suda-morris 
Thanks. I may be change the ws2812 driver to esp32s3's built-in.

wernerschmitz, 2025-03-19:
Can anyone explain me how to solve this in platformIO (framework arduino)