espressif/esp-idf #14976: Using idf.py build cant't build the example 'examples\peripherals\usb\device\tusb_msc'

yizi15, 2024-12-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

USE Clion IDE with `idf_cmd_init.bat` could build success.

### What is the actual behavior?

but use `Initialize-Idf.ps1` and run `idf.py build`  could get this:

### Steps to reproduce.

1. open powershell
2. run Initialize-Idf.ps1
3. run idf.py set-target esp32s3
4. idf.py build

### Build or installation Logs.

```plain
PS D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc> idf.py -p com17 flash
Executing action: flash
Running ninja in directory D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc\build
Executing "ninja flash"...
[1/7] Performing build step for 'bootloader'
CMake Error: The current CMakeCache.txt directory D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/build/bootloader/CMakeCache.txt is different than the directory d:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/cmake-build-debug-esp51_230913/bootl
oader where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
[1/1] cmd.exe /C "cd /D D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc\cmake-build-debug-esp51_230913\bootloader\esp-idf\esptool_py && python D:/Espressif5/frameworks/esp-idf-v5.1.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 D:/Espressif5/framework
s/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/cmake-build-debug-esp51_230913/bootloader/bootloader.bin"
Bootloader binary size 0x51c0 bytes. 0x2e40 bytes (36%) free.
[2/5] Linking CXX executable tusb_msc.elf
FAILED: tusb_msc.elf 
cmd.exe /C "cd . && D:\Espressif5\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/build/tusb_msc.map -Wl,--no-warn-rw
x-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\tusb_msc.elf.rsp -o tusb_msc.elf  && cd ."
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(tusb_msc_main.c.obj):(.literal.storage_init_spiflash+0x8): undefined reference to `wl_mount'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(tusb_msc_main.c.obj): in function `storage_init_spiflash':
D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/main/tusb_msc_main.c:249: undefined reference to `wl_mount'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/fatfs/libfatfs.a(diskio_wl.c.obj): in function `ff_wl_read':
D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:37: undefined reference to `wl_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:37: undefined reference to `wl_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:37: undefined reference to `wl_read'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/fatfs/libfatfs.a(diskio_wl.c.obj): in function `ff_wl_write':
D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:50: undefined reference to `wl_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:50: undefined reference to `wl_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:50: undefined reference to `wl_erase_range'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:55: undefined reference to `wl_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:55: undefined reference to `wl_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:55: undefined reference to `wl_write'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/fatfs/libfatfs.a(diskio_wl.c.obj): in function `ff_wl_ioctl':
D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:72: undefined reference to `wl_size'
[... 6 lines cut ...]
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a(tusb_msc_storage.c.obj):(.literal._read_sector_spiflash+0xc): undefined reference to `wl_re
ad'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a(tusb_msc_storage.c.obj):(.literal._get_sector_size_spiflash+0xc): undefined reference to `w
l_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a(tusb_msc_storage.c.obj):(.literal._get_sector_count_spiflash+0x8): undefined reference to `
wl_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a(tusb_msc_storage.c.obj): in function `_write_sector_spiflash':
D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/l_components/espressif__esp_tinyusb/tusb_msc_storage.c:112: undefined reference to `wl_erase_range'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/l_components/espressif__esp_tinyusb/tusb_msc_storage.c:114: undefined
 reference to `wl_write'
[... cut ...]
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/l_components/espressif__esp_tinyusb/tusb_msc_storage.c:81: undefined 
reference to `wl_size'
collect2.exe: error: ld returned 1 exit status
```
```

### More Information.

[compile_commands.json](https://github.com/user-attachments/files/18001604/compile_commands.json)

yizi15, 2024-12-04:
```
PS D:\Espressif5\frameworks\esp-idf-v5.1.2> cd D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc\
PS D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc> .\env.ps1                                                                           
Setting PYTHONNOUSERSITE, was not set
Using Python in D:\Espressif5\python_env\idf5.1_py3.11_env\Scripts
Python 3.11.2                                      
Using Git in D:\Espressif5\tools\idf-git\2.39.2\cmd
git version 2.39.2.windows.1
Setting IDF_PATH: D:\Espressif5\frameworks\esp-idf-v5.1.2
Checking Python compatibility                            
Adding ESP-IDF tools to PATH...

Name                           Value
----                           -----
OPENOCD_SCRIPTS                D:\Espressif5\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE              1
ESP_ROM_ELF_DIR                D:\Espressif5\tools\esp-rom-elfs\20230320\
IDF_PYTHON_ENV_PATH            D:\Espressif5\python_env\idf5.1_py3.11_env
ESP_IDF_VERSION                5.1
IDF_DEACTIVATE_FILE_PATH       C:\Users\FENGXI~1\AppData\Local\Temp\tmpvjvycfniidf_19988

Added to PATH
-------------
D:\Espressif5\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
D:\Espressif5\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
D:\Espressif5\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
D:\Espressif5\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
D:\Espressif5\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
D:\Espressif5\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
D:\Espressif5\tools\cmake\3.24.0\bin
D:\Espressif5\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin
D:\Espressif5\tools\ninja\1.10.2\
D:\Espressif5\tools\idf-exe\1.0.3\
D:\Espressif5\tools\ccache\4.8\ccache-4.8-windows-x86_64
D:\Espressif5\tools\dfu-util\0.11\dfu-util-0.11-win64
D:\Espressif5\frameworks\esp-idf-v5.1.2\tools
Checking if Python packages are up to date...
Constraint file: D:\Espressif5\espidf.constraints.v5.1.txt
Requirement files:
 - D:\Espressif5\frameworks\esp-idf-v5.1.2\tools\requirements\requirements.core.txt
Python being checked: D:\Espressif5\python_env\idf5.1_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of idf-driver, idf-python-wheels, xtensa-esp32-elf use command 'python.exe D:\Espressif5\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe D:\Espressif5\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

PS D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc> idf.py set-target esp32s3                                                           
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:\Espressif5\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 D:\Espressif5\frameworks\esp-idf-v5.1.2\examples\peripherals\usb\device\tusb_msc"...
-- Existing sdkconfig 'D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/sdkconfig' renamed to 'D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/sdkconfig.old'.
-- Found Git: D:/Espressif5/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 137 lines cut ...]
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/components/fatfs/diskio/diskio_wl.c:55: undefined reference to `wl_sector_size'
rase_range'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a(tusb_msc_storage.c.obj):(.literal._write_sector_spiflash+0x10): undefined reference to `wl_
write'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a(tusb_msc_storage.c.obj):(.literal._read_sector_spiflash+0xc): undefined refere [... cut ...]
D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/l_components/espressif__esp_tinyusb/tusb_msc_storage.c:76: undefined reference to `wl_sector_size'
d:/espressif5/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/peripherals/usb/device/tusb_msc/l_components/espressif__esp_tinyusb/tusb_msc_storage.c:81: undefined 
reference to `wl_size'
collect2.exe: error: ld returned 1 exit status

```

tore-espressif, 2024-12-04:
@yizi15 I tried building the example on both v5.1.2 and master, all work OK.

I can see that you modified the project

You log says
```
Processing 1 dependencies:
[1/1] idf (5.1.2)
```

But it should be
```
Processing 3 dependencies:
[1/3] espressif/esp_tinyusb (1.4.5)
[2/3] espressif/tinyusb (0.17.0~1)
[3/3] idf (5.1.2)
```

The path in your compiler error message is also modified, it should say _**managed_components**_
/examples/peripherals/usb/device/tusb_msc/***l_components***/espressif__esp_tinyusb/tusb_msc_storage.c:76: undefined reference to `wl_sector_size'

The symbols that the linker is failing to find are from `wear_leveling` component, which is a public dependency of `fatfs`, which is included here

https://github.com/espressif/esp-usb/blob/51bbacdc0ac1f5b12f9797817c90dc80146ce3da/device/esp_tinyusb/CMakeLists.txt#L40

have you modified the CMakeLists.txt ?

yizi15, 2024-12-04:
@tore-espressif, It seems to not about the dependencies. the `wear_leveling` src endswith .cpp, and the files in the compile_commands.json . I try add a file `test.cpp` to my CMakeLists.txt, and wirte a error in the cpp, but no error in build log.  
Clion IDE can report the error. Compare the two build directory, the cmake and ninja is not same exe. the obj file `build\esp-idf\wear_levelling\CMakeFiles\__idf_wear_levelling.dir\wear_levelling.cpp.obj` is 5 times more than the `cmake-build-debug-esp51_230913\esp-idf\wear_levelling\CMakeFiles\__idf_wear_levelling.dir\wear_levelling.cpp.obj`

the 'build\esp-idf\main\CMakeFiles\__idf_main.dir\test.cpp.obj' is 
[image]
even the file is 
[image]

tore-espressif, 2024-12-04:
@yizi15 I tried the same and got correct error

```
C:/esp_repos/esp-idf/release-v5.1/examples/peripherals/usb/device/tusb_msc/main/test.cpp:2:5: error: 'asd' was not declared in this scope
    2 |     asd
      |     ^~~
```   
      
I suggest you delete `build`, `managed_components`, `dependencies.lock` and `sdkconfig`. Revert all your changes and build from clean state.

I'll close the issue as there is no issue to big fixed on our side

closed 2024-12-04 by tore-espressif