espressif/esp-idf #14977: BLE ANCS Pairing Issue with ESP32 (Error Code: 0x66/102)

nitin-jethava, 2024-12-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Issue Description**:  

The ESP32 running the ESP-IDF BLE ANCS example frequently fails to pair with an iOS device after repeated connect/disconnect cycles. The pairing fails with the error code `0x66` (102).

---

**Steps to Reproduce**:  
1. Run the built-in ESP-IDF BLE ANCS example on the ESP32.  
2. Connect the ESP32 to an iOS device.  
3. Disconnect and reconnect multiple times or move the iOS device in and out of range.  
4. Observe pairing failure with error `0x66`.

---

**Observation**:  
- The issue occurs more frequently when the connection/disconnection cycle is fast.  
- It appears the device cannot proceed with pairing or retrieve ANCS services/characteristics due to timing issues.  
- Even if disconnections are delayed, the problem reoccurs after multiple cycles.

---

**Logs**:
````````````````````````````````````````````````````````````````````````
I (31) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (31) boot: compile time Dec  4 2024 12:49:33
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20a7ch (133756) map
I (150) esp_image: segment 1: paddr=00030aa4 vaddr=3ffbdb60 size=04bech ( 19436) load
I (158) esp_image: segment 2: paddr=00035698 vaddr=40080000 size=0a980h ( 43392) load
I (175) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=7bb34h (506676) map
I (349) esp_image: segment 4: paddr=000bbb5c vaddr=4008a980 size=0caf0h ( 51952) load
I (381) boot: Loaded app from partition at offset 0x10000
I (381) boot: Disabling RNG early entropy source...
I (393) cpu_start: Multicore app
I (402) cpu_start: Pro cpu start user code
I (402) cpu_start: cpu freq: 160000000 Hz
I (402) app_init: Application information:
I (405) app_init: Project name:     ble_ancs
I (410) app_init: App version:      1
I (414) app_init: Compile time:     Dec  4 2024 12:48:21
I (420) app_init: ELF file SHA256:  516f9382e...
I (425) app_init: ESP-IDF:          v5.3.1
I (430) efuse_init: Min chip rev:     v0.0
I (435) efuse_init: Max chip rev:     v3.99 
I (440) efuse_init: Chip rev:         v3.1
I (445) heap_init: Initializing. RAM available for dynamic allocation:
I (452) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (458) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (464) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (470) heap_init: At 3FFC75F0 len 00018A10 (98 KiB): DRAM
I (476) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[... 117 lines cut ...]
I (140053) BLE_ANCS: pair status = fail
I (140053) BLE_ANCS: fail reason = 0x66
I (140053) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (140083) BLE_ANCS: advertising start success
I (144123) BLE_ANCS: ESP_GATTC_OPEN_EVT
W (144403) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x13
E (144413) BT_BTM: Device not found

W (144413) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
W (144413) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 73:6e:05:0f:9e:24
I (144423) addr: 73 6e 05 0f 9e 24 
I (144433) BLE_ANCS: pair status = fail
I (144433) BLE_ANCS: fail reason = 0x66
I (144433) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (144463) BLE_ANCS: advertising start success
````````````````````````````````````````````````````````````````````````
  
Key observations from logs:  
1. Initial pairing succeeds (`pair status = success`).    
2. On reconnection attempts, pairing fails (`pair status = fail`, `fail reason = 0x66`).

---

**Potential Cause**:  
The BLE stack might not handle rapid reconnections or repeated disconnection scenarios effectively, leading to pairing and service discovery failure.

esp-zhp, 2024-12-09:
---

Based on your log, the reason for the SMP failure is 0x66, which indicates an **SMP timeout**. This means that during the SMP process, no packets were received from the peer device.  

```
ESP_AUTH_SMP_CONN_TOUT, /*!< Pairing process failed due to connection timeout */
```

Additionally, I noticed the following in your log:  
`I (144433) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13`.  
The peer device appears to be repeatedly disconnecting.  

I suspect that during the peer device’s disconnection process, the SMP procedure has not been completed, which ultimately results in the SMP failure.  

I recommend ensuring the connection is maintained until the SMP process is successfully completed before disconnecting. This should help resolve your issue.  

---

nitin-jethava, 2024-12-09:
Thank you for the insights, @esp-zhp 

It seems the issue arises because the user is frequently moving in and out of the device's Bluetooth range. This causes the Bluetooth connection to toggle repeatedly, which may be leading to the SMP timeout and disconnection issues.

Given this behavior, could you suggest steps to prevent or mitigate this problem? Should we consider making adjustments in the firmware to handle frequent reconnections more effectively, or are there other solutions you would recommend?

esp-zhp, 2024-12-10:
@nitin-jethava 
It doesn’t seem like the disconnection was caused by exceeding the connection range. If it were, an error code of 0x08 would have been reported instead of 0x13. The 0x13 code indicates that the peer device actively terminated the connection, meaning the iPhone device disconnected intentionally.  

There are two possibilities: one is that the Apple Bluetooth stack actively disconnected, and the other is that the user disconnected at the application layer.  

I suspect the frequent disconnections might be caused by user actions. Do you have any other perspectives? do you know the scenario in which this issue occurred?

esp-zhp, 2024-12-10:
---

Based on your recent description, the issue appears to have changed and seems to be related to SMP (Security Manager Protocol). It looks like the ESP device may have lost the bond information.

To verify this assumption, you can try printing the bond list using the following APIs:  
- `int esp_ble_get_bond_device_num(void)`  
- `esp_err_t esp_ble_get_bond_device_list(int *dev_num, esp_ble_bond_dev_t *dev_list)`

If it is confirmed to be an SMP-related issue, you can test the behavior without bonding to see if the problem still occurs.  

Please let us know if further assistance is needed!  

---

nitin-jethava, 2024-12-10:
You are correct, @esp-zhp. I had tested this, and the bond information indeed appears to be missing when the issue occurs.

Under normal circumstances, the `int esp_ble_get_bond_device_num(void)` API returns `1`, indicating the presence of bond information. However, when the error (`0x66`) occurs, the same API returns `0`, suggesting that the bond information is lost or in an idle state.

Since bonding is closely tied to the ANCS functionality in our application, this loss of bond information directly impacts the ability to receive notifications seamlessly.

Given this scenario, could you please advise on the best approach to handle such cases? Specifically:

1. What is the ideal way to manage situations where bond information is lost?
2. Are there recommended steps or fallback mechanisms we can implement in the firmware to gracefully handle this situation and re-establish the bond without requiring user intervention?

Your guidance on this matter would be greatly appreciated as it will help us optimize our handling of these disconnections.

esp-zhp, 2024-12-11:
In fact, we have never encountered a situation where the ESP32 loses bond information.  
Could you add more debug logs to trace when exactly the bond information gets lost?

nitin-jethava, 2024-12-11:
Thank you for your input, @esp-zhp.  

After reviewing your suggestions, I’ve added debug logs at key events to monitor the bond list and collected the following observations:  

1. **Before Connection**:  
   The bond list size is logged after advertising starts, showing an initial value:  
   ```
   I (48502) BLE_ANCS: advertising start success  
   I (48502) BONDING_LIST: Bond list size after restart adverties: 3  
   ```

2. **After Connection (ESP_GATTC_CONNECT_EVT)**:  
   The bond list size remains consistent, confirming successful pairing:  
   ```
   I (50842) BLE_ANCS: ESP_GATTC_CONNECT_EVT  
   I (50842) BONDING_LIST: Bond list size after connection: 3  
   I (50852) BLE_ANCS: ESP_GATTC_OPEN_EVT  
   I (47092) BLE_ANCS: pair status = success  
   ```

3. **After Disconnection (ESP_GATTC_DISCONNECT_EVT)**:  
   The bond list size decreases when an error occurs. Specifically, after the `0x66` error:  
   ```
   I (51832) BLE_ANCS: pair status = fail  
   I (51832) BLE_ANCS: fail reason = 0x66  
   I (51832) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x8  
   I (51842) BONDING_LIST: Bond list size after disconnection: 2  
   ```

4. **Reconnection Attempts**:  
   Subsequent attempts fail with bond list inconsistencies persisting:  
   ```
   I (52902) BLE_ANCS: fail reason = 0x66  
   I (52902) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13  
   I (52912) BONDING_LIST: Bond list size after disconnection: 2  
   ```

From these logs, it’s clear that bond information is lost after certain disconnection scenarios, particularly when errors `0x66` and `0x13` are encountered. This affects the ANCS functionality, as bonding is critical for receiving notifications seamlessly.  

I’ve attached the full log file (`ble_ancs_logs_11_dec.txt`) for your review.  

Could you please advise on the following:

1. The most effective approach to recover from bond information loss.
2. Any additional configurations or changes in the pairing process that could help prevent this issue.

If you require additional logs, specific test scenarios, or further debugging, please feel free to let me know. I’ll be glad to provide any necessary details or support to help resolve this issue.

[ble_ancs_logs_11_dec.txt](https://github.com/user-attachments/files/18096306/ble_ancs_logs_11_dec.txt)

esp-zhp, 2024-12-12:
Based on the additional logs you provided, we further confirmed the bond loss scenario. Could you check whether the example actively clears bond information?  

If there’s no active clearing, we suspect this might be a bug in the protocol stack. I’ve added some HCI logs for further debugging. 
[0003-fix-bt-Display-hci-data-including-up-and-down.zip](https://github.com/user-attachments/files/18105103/0003-fix-bt-Display-hci-data-including-up-and-down.zip)
Please apply the patch and share the complete logs with me.  

Additionally, have you tried using the latest branch?

nitin-jethava, 2024-12-12:
Thank you for the clarification, @esp-zhp.  

1. **No Active Clearing of Bond Information**:  
   After reviewing the example code, I can confirm that the bond information is not actively cleared in the example.  

2. **Patch Application and Log Analysis**:  
   I applied the provided HCI log patch and rebuilt the project. Below are the observations from the debug logs:  

   - **Before Connection**:  
     ```
     I (322733) BLE_ANCS: advertising start success  
     I (322733) BONDING_LIST: Bond list size after restart adverties: 3
     ```
   - **After Connection**:  
     ```
     I (324143) BLE_ANCS: ESP_GATTC_CONNECT_EVT  
     I (324153) BONDING_LIST: Bond list size after connection: 3
     ```
   - **After Disconnection (Error 0x66)**:  
     ```
     I (324543) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13  
     I (324543) BONDING_LIST: Bond list size after disconnection: 2
     ```
   These logs indicate that the bond list size decreases after the disconnection, suggesting a bond loss scenario.  

3. **Attached Full Logs**:  
   I have attached the full log file for your reference, which includes the HCI logs and additional debug outputs.
   
[ble_ancs_logs_12_dec.txt](https://github.com/user-attachments/files/18107064/ble_ancs_logs_12_dec.txt)

4. **ESP-IDF Version**:  
   I am currently using the stable release `v5.3.2`.  

Could you please advise further based on these findings? Do you suspect this could be a protocol stack issue, and are there additional steps or tests you would recommend?  

If you require additional logs, specific test scenarios, or further debugging, please feel free to let me know. I’ll be glad to provide any necessary details or support to help resolve this issue.

esp-zhp, 2024-12-13:
@nitin-jethava 
I fixed this issue for you,please test it and give me a feedback.thanks.
[0001-fix-ble-bluedroid-Fixed-BLE-bond-lost-if-pair-fail.zip](https://github.com/user-attachments/files/18124084/0001-fix-ble-bluedroid-Fixed-BLE-bond-lost-if-pair-fail.zip)

nitin-jethava, 2024-12-13:
Thank you so much for your solution, @esp-zhp!

The issue is now resolved, and everything is working as expected. I appreciate your detailed explanation and the suggested changes regarding bond information retention. The fix has worked for me, and I can confirm the solution is effective.

I'll also be starting a new thread for a similar issue in the Arduino environment, as I was using the `BLEDevice` library for the development. I'll need to make similar changes in that library to address this issue, and I'll make sure to update the discussion accordingly.

Thanks again for your valuable support and for guiding me through this!

esp-zhp, 2024-12-16:
@nitin-jethava 
thanks for your updating.
Actually, I don’t know much about Arduino, and this seems to be a new issue. If your BLE problem has already been resolved, you can create a separate issue specifically about Arduino. More knowledgeable experts will be able to help you with that.

nitin-jethava, 2024-12-16:
Thank you so much for your guidance, @esp-zhp!

I truly appreciate your help in addressing this issue, and for your suggestion regarding the Arduino environment. I’ve created a new issue specifically for the `BLEDevice` library on the Arduino platform, and you can find it here: [[Issue #1201](https://github.com/nkolban/esp32-snippets/issues/1201)](https://github.com/nkolban/esp32-snippets/issues/1201).

Since I’ve now raised the issue in the appropriate place, please feel free to close this one. 

Thanks again for all your support! I’ll keep you updated on any progress and let you know if I need further assistance.

Best regards,  
Nitin

esp-zhp, 2024-12-16:
@nitin-jethava yeah,please close this issue.

closed 2024-12-16 by nitin-jethava