espressif/esp-idf #14979: esp32-p4 Stuck on psram initialization

SSEHX, 2024-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-4666-g030c9957db

### Espressif SoC revision.

esp32-p4 v0.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-p4 v1.4

### Power Supply used.

USB

### What is the expected behavior?

正常运行

### What is the actual behavior?

ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x10 (CHIP_LP_WDT_RESET),boot:0x10f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x1254
load:0x4ff2abd0,len:0xc4c
--- 0x4ff2abd0: call_start_cpu0 at /Users/logicdance/esp/master/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

load:0x4ff2cbd0,len:0x3174
--- 0x4ff2cbd0: esp_flash_encryption_enabled at /Users/logicdance/esp/master/esp-idf/components/bootloader_support/src/flash_encrypt.c:86

entry 0x4ff2abd0
--- 0x4ff2abd0: call_start_cpu0 at /Users/logicdance/esp/master/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

W (28) boot.esp32p4: CPU has been reset by WDT.
I (262) hex_psram: vendor id    : 0x0d (AP)
I (262) hex_psram: Latency      : 0x01 (Fixed)
I (262) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (263) hex_psram: dev id       : 0x03 (generation 4)
I (268) hex_psram: density      : 0x07 (256 Mbit)
I (272) hex_psram: good-die     : 0x06 (Pass)
I (276) hex_psram: SRF          : 0x02 (Slow Refresh)
I (281) hex_psram: BurstType    : 0x00 ( Wrap)
I (285) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (290) hex_psram: BitMode      : 0x01 (X16 Mode)
I (294) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (299) hex_psram: DriveStrength: 0x00 (1/1)
I (303) MSPI DQS: tuning success, best phase id is 2
I (491) MSPI DQS: tuning success, best delayline id is 12
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 200MHz
I (504) mmu_psram: .rodata xip on psram
I (551) mmu_psram: .text xip on psram

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x10 (CHIP_LP_WDT_RESET),boot:0x10f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x1254
load:0x4ff2abd0,len:0xc4c
--- 0x4ff2abd0: call_start_cpu0 at /Users/logicdance/esp/master/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

load:0x4ff2cbd0,len:0x3174
--- 0x4ff2cbd0: esp_flash_encryption_enabled at /Users/logicdance/esp/master/esp-idf/components/bootloader_support/src/flash_encrypt.c:86

entry 0x4ff2abd0
--- 0x4ff2abd0: call_start_cpu0 at /Users/logicdance/esp/master/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

W (28) boot.esp32p4: CPU has been reset by WDT.
I (262) hex_psram: vendor id    : 0x0d (AP)
I (262) hex_psram: Latency      : 0x01 (Fixed)
I (262) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (263) hex_psram: dev id       : 0x03 (generation 4)
I (268) hex_psram: density      : 0x07 (256 Mbit)
I (272) hex_psram: good-die     : 0x06 (Pass)
I (276) hex_psram: SRF          : 0x02 (Slow Refresh)
I (281) hex_psram: BurstType    : 0x00 ( Wrap)
I (285) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (290) hex_psram: BitMode      : 0x01 (X16 Mode)
I (294) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (299) hex_psram: DriveStrength: 0x00 (1/1)
I (303) MSPI DQS: tuning success, best phase id is 2
I (491) MSPI DQS: tuning success, best delayline id is 12
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 200MHz
I (504) mmu_psram: .rodata xip on psram
I (551) mmu_psram: .text xip on psram
```

Icarus113, 2024-12-06:
could you help post your sdkconfig, we can try if this can be reproduced.
@SSEHX

SSEHX, 2024-12-06:
[sdkconfig.zip](https://github.com/user-attachments/files/18037589/sdkconfig.zip)
@Icarus113 Hello, this is the SDK config file with the problem

cfriedt, 2024-12-07:
I hit exactly the same issue. It's a boot-loop triggered by the watchdog timer.

Something is hanging firmware.

```shell
--- esp-idf-monitor 1.5.0 on /dev/cu.usbserial-10 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I (255) esp_image: segment�ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x17a8
load:0x4ff2abd0,len:0xf34
--- 0x4ff2abd0: esp_bootloader_get_description at /Users/cfriedt/esp/esp-idf/components/esp_bootloader_format/esp_bootloader_desc.c:39 (discriminator 1)

load:0x4ff2cbd0,len:0x3404
--- 0x4ff2cbd0: esp_flash_encryption_enabled at /Users/cfriedt/esp/esp-idf/components/bootloader_support/src/flash_encrypt.c:88

entry 0x4ff2abda
--- 0x4ff2abda: call_start_cpu0 at /Users/cfriedt/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

I (25) boot: ESP-IDF v5.5-dev-811-ga2f420a36e 2nd stage bootloader
I (26) boot: compile time Dec  7 2024 16:45:49
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v1.0
I (30) boot: efuse block revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32p4: SPI Speed      : 80MHz
I (42) boot.esp32p4: SPI Mode       : QIO
I (46) boot.esp32p4: SPI Flash Size : 16MB
I (50) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00800000
I (82) boot:  3 storage          Unknown data     01 82 00810000 00700000
I (90) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=48060020 size=74e08h (478728) map
I (174) esp_image: segment 1: paddr=00084e30 vaddr=30100000 size=0002ch (    44) load
I (176) esp_image: segment 2: paddr=00084e64 vaddr=3010002c size=0003ch (    60) load
I (179) esp_image: segment 3: paddr=00084ea8 vaddr=4ff00000 size=0b170h ( 45424) load
I (195) esp_image: segment 4: paddr=00090020 vaddr=48000020 size=5c5f4h (378356) map
I (255) esp_image: segment 5: paddr=000ec61c vaddr=4ff0b170 size=0cd34h ( 52532) load
I (266) esp_image: segment 6: paddr=000f9358 vaddr=4ff17f00 size=03498h ( 13464) load
I (270) esp_image: segment 7: paddr=000fc7f8 vaddr=50108080 size=0001ch (    28) load
I (277) boot: Loaded app from partition at offset 0x10000
I (277) boot: Disabling RNG early entropy source...
I (291) hex_psram: vendor id    : 0x0d (AP)
I (291) hex_psram: Latency      : 0x01 (Fixed)
I (292) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (292) hex_psram: dev id       : 0x03 (generation 4)
I (297) hex_psram: density      : 0x07 (256 Mbit)
I (301) hex_psram: good-die     : 0x06 (Pass)
I (305) hex_psram: SRF          : 0x02 (Slow Refresh)
I (310) hex_psram: BurstType    : 0x00 ( Wrap)
I (314) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (319) hex_psram: BitMode      : 0x01 (X16 Mode)
I (323) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (328) hex_psram: DriveStrength: 0x00 (1/1)
I (333) MSPI DQS: tuning success, best phase id is 2
I (510) MSPI DQS: tuning success, best delayline id is 11
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 200MHz
I (536) mmu_psram: .rodata xip on psram
I (554) mmu_psram: .text xip on psram
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x10 (CHIP_LP_WDT_RESET),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x17a8
load:0x4ff2abd0,len:0xf34
--- 0x4ff2abd0: esp_bootloader_get_description at /Users/cfriedt/esp/esp-idf/components/esp_bootloader_format/esp_bootloader_desc.c:39 (discriminator 1)

load:0x4ff2cbd0,len:0x3404
--- 0x4ff2cbd0: esp_flash_encryption_enabled at /Users/cfriedt/esp/esp-idf/components/bootloader_support/src/flash_encrypt.c:88

entry 0x4ff2abda
--- 0x4ff2abda: call_start_cpu0 at /Users/cfriedt/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

I (26) boot: ESP-IDF v5.5-dev-811-ga2f420a36e 2nd stage bootloader
I (27) boot: compile time Dec  7 2024 16:45:49
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v1.0
```

Icarus113, 2024-12-09:
Have reproduced the issue @SSEHX , we'll check it and give you feedback later.

Icarus113, 2024-12-10:
The fix will be updated here soon. @SSEHX @cfriedt

closed 2024-12-20 by espressif-bot