espressif/esp-idf #14980: ESP32S3 - (MTCK)GPIO39 No longer configures to GPIO on 5.3.1

tivetdb, 2024-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I upgraded from 5.2.2 to 5.3.1 following these [instructions](https://github.com/espressif/esp-idf/releases/tag/v5.3.1) and can no longer set GPIO39 as a GPIO with this code:

Nothing else in the project changed. The pin is always low when configured as an Output:

```
        gpio_reset_pin(39);
        esp_rom_gpio_pad_select_gpio(39);
        gpio_set_level(39, 1);
        gpio_set_direction(39, GPIO_MODE_OUTPUT);

      //Second attempt

        io_config = 0;
        io_config |= 1ULL << 39;
        gpio_config_t io_conf;
        io_conf.intr_type = GPIO_INTR_DISABLE;
        io_conf.mode = GPIO_MODE_OUTPUT;
        io_conf.pin_bit_mask = io_config;
        io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
        io_conf.pull_up_en = GPIO_PULLUP_ENABLE;
        gpio_config(&io_conf);
        gpio_dump_io_configuration(stdout, io_config);
        
        
  while (true) {
        gpio_set_level(39,0);
        vTaskDelay(pdMS_TO_TICKS(1000));
        gpio_set_level(39,1);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
        
```

The dump reports that it is set up, but it does not toggle on a WROOM with nothing connected to it:

IO[39] -
  Pullup: 1, Pulldown: 0, DriveCap: 2
  InputEn: 0, OutputEn: 1, OpenDrain: 0
  FuncSel: 1 (GPIO)
  GPIO Matrix SigOut ID: 256 (simple GPIO output)
  SleepSelEn: 1

Kainarx, 2024-12-06:
Hi, @tivetdb 
Do other gpio's have this problem as well, or just gpio39? This can not be reproduced on my side. Could you help post your sdkconfig?

tivetdb, 2024-12-06:
@Kainarx 

My SDKCONFIG was outdated after moving the project to 5.3.1, causing these issues.

After running idf.py menuconfig and saving, a recompile solved my issue.

Thank you

closed 2024-12-06 by tivetdb