espressif/esp-idf #14981: ESP32H2 Openthread SED with LEDC not working

AAJAY5, 2024-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## Issue
- I have took sleepy end device as template project and added LEDC driver to generate PWM in sleep mode. Aim is device poll every 5 second if any pending message it will receive it and process again go to sleep.
- LEDC driver working with timer wakeup and light sleep example. But in this case when node connect to router it changes it state to Child as required and start sleep.
- Till now LEDC is working.
- Once it connected to network and change it's state to child LEDC stop working. It not working in sleep mode.

```c
/*
 * SPDX-FileCopyrightText: 2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * OpenThread Command Line Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdint.h>
#include <stdio.h>
#include <unistd.h>
#include <string.h>
#include "esp_err.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_openthread.h"
#include "esp_openthread_cli.h"
#include "esp_openthread_lock.h"
#include "esp_openthread_netif_glue.h"
#include "esp_ot_sleepy_device_config.h"
#include "esp_vfs_eventfd.h"
#include "driver/uart.h"
#include "nvs_flash.h"
#include "openthread/logging.h"
#include "openthread/thread.h"
#if CONFIG_ESP_SLEEP_DEBUG
#include "esp_timer.h"
#include "esp_sleep.h"
#include "esp_private/esp_pmu.h"
#include "esp_private/esp_sleep_internal.h"
#endif

#include <openthread/ip6.h>
#include <openthread/message.h>
#include <openthread/udp.h>
#include "lights.h"

#ifdef CONFIG_PM_ENABLE
#include "esp_pm.h"
#endif

#if !SOC_IEEE802154_SUPPORTED
#error "Openthread sleepy device is only supported for the SoCs which have IEEE 802.15.4 module"
#endif

#if CONFIG_IDF_TARGET_ESP32H2
/** 
 * ESP32-H2 SleepMode Power issue LEDC
 * https://github.com/espressif/esp-idf/issues/13698#issuecomment-2119607575
[... 286 lines cut ...]
        .timer_num        = LEDC_TIMER_0,
        .duty_resolution  = LEDC_TIMER_8_BIT,
        .freq_hz          = 1000,  // 1000 Hz frequency
        .clk_cfg          = LEDC_USE_RC_FAST_CLK
    };
    ESP_ERROR_CHECK(ledc_timer_config(&ledc_timer));

    ledc_channel_config_t ledc_channel = {
        .gpio_num       = LEDC_OUTPUT_IO,
        .speed_mode     = LEDC_LOW_SPEED_MODE,
        .channel        = LEDC_CHANNEL_0,
        .timer_sel      = LEDC_TIMER_0,
        .duty           = 0,  // Default 10% brightness
        .hpoint         = 0
    };
    ESP_ERROR_CHECK(ledc_channel_config(&ledc_channel));
}

// Function to set PWM brightness
void pwm_set_brightness(uint8_t brightness) {
    ESP_LOGI(TAG, "Setting brightness to %d", brightness);
    ESP_ERROR_CHECK(ledc_set_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0, brightness));
    ESP_ERROR_CHECK(ledc_update_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0));
}
```

zwx1995esp, 2024-12-10:
Hi, @AAJAY5  Could you please set the `LEDC_SLEEP_MODE_NO_ALIVE_ALLOW_PD` for the para `ledc_channel_config_t` in function `pwm_init()` and then have a try? For more details, please refer to [ledc_sleep_mode_t](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_ledc/include/driver/ledc.h#L37)

chshu, 2025-02-26:
@AAJAY5 Please reopen if any follow up questions.

closed 2025-02-26 by chshu